MichaIng/DietPi #542: Documentation update for Samba

RajatNair, 2016-10-02:
Was following instructions documented here - http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=10#p56 for Samba. After making changes to conf file and trying to start the service using - 
`service samba start`
following error was throw -

> Failed to restart samba.service: Unit samba.service is masked.

` service smbd start` or `systemctl restart smbd` works.

Guess we should update Samba docs.

Fourdee, 2016-10-02:
@RajatNair 

Great find, you've also uncovered a bug 👍 

```smbd``` and ```nmbd``` services were not being applied correctly by ```dietpi-services```

closed 2016-10-02 by Fourdee

RajatNair, 2016-10-02:
I just came back to add that dietpi-services was throwing an error cause of Samba service -

>  DietPi-Services
> ─────────────────────────────────────────────────────
>  Mode: restart
>  Please wait...
>  [Ok] cron
>  [Failed] samba
>  [Ok] lighttpd

While @Fourdee already fixed it. That was quick! Nice work 👍