MichaIng/DietPi #544: PINE64  inital install loops

tmose, 2016-10-04:
When installing the latest version of DietPI on pine 64 2 gig the installer goes into a never ending loop.  This only seems to happen with the newest installer and not a previous version.

Fourdee, 2016-10-04:
@tmose 

Thanks for the report. 
Is this in regards to the automatic "first run setup" of DietPi, or, installing software in ```dietpi-software```?

tmose, 2016-10-04:
I have not been able to get past it so it would be on the "first run setup".

Thomas M. Ose
Ose Micro Solutions, Inc.

Fourdee, 2016-10-05:
@tmose 

I was unable to replicate the endless loop on in my tests on v127 Pine image.

Is there any chance you could provide more information on the endless loop, and/or, take a screenshot/picture if possible?

tmose, 2016-10-05:
Dan,

I will try again and see what I can capture.  Is there a log file created
that I can send you.

The install worked on one board but loops on a 2 gig and 1 gig board.

Thomas M. Ose
Ose Micro Solutions, Inc.

tmose, 2016-10-05:
Dan ,

One more bit of info,  I am using a ScanDiskUltra 32gb class10 and no wifi
or bluetooth.

Thomas M. Ose
Ose Micro Solutions, Inc.

tmose, 2016-10-05:
Dan,

Just finished a clean install of v127.  After agreeing to the software
license, the install goes back to the command line and loops.  I then
entered ^C to interrupt the loop and then restarted DietPi-Software.  Did
not select any options and started the install.  At the command prompt I
got the following error "Failing Apt-get simulation".  I then did a
"apt-get install -f" followed by a "apt-get upgrade".  restarted
DietPi-Software and this time everything installed.  After reboot was able
to upgrade to v132 and install all the software I selected.

Hope this helps.

Thomas M. Ose
Ose Micro Solutions, Inc.

tmose, 2016-10-05:
Dan,

Thought things were good but tried to install mate and it failed badly now
not recognizing keyboard after reboot.   I will wait until you look at this
before trying again.

Thomas M. Ose
Ose Micro Solutions, Inc.

Fourdee, 2016-10-05:
@tmose 

Hi Thomas, thanks for the updated information.

It looks like a network related issue. Possibly dropping connection from time to time.

> install mate and it failed badly now
not recognizing keyboard after reboot. 

This could indicate a hardware failure or insufficient power supply.

Thomas:
- Which make/model of power supply are you using in these tests? (eg: 5v 2amp Samsung phone charger)
- It might be worth trying other power supplies (if you have one available). Make sure to start from a fresh image write each time.

> The install worked on one board but loops on a 2 gig and 1 gig board.

Again, this could be power related, if the board it works on is 512MB due to less power draw. I'll run a test on my 1GB board tomorrow, my 2GB is in use at the moment (makeshift tablet lol).

Fourdee, 2016-10-06:
@tmose 

> I'll run a test on my 1GB board tomorrow, my 2GB is in use at the moment (makeshift tablet lol).

1GB installed fine aswell for me. Unable to replicate the loop. 
If you can get back to me regarding the following questions: https://github.com/Fourdee/DietPi/issues/544#issuecomment-251767824, when you can, i'd be grateful.

tmose, 2016-10-06:
> (comment)

Dan,

I tried again and this time did not use the ethernet connection but
connected the wifi card I had.  With wifi everything worked as expected. I
now think that my problem may be related to the faulty network chip issue
others were seeing, and may explain why under other distros I was only
getting 100 connection.

Any way to switch the network connection down to 100m, I can install with
Wifi but would rather run on a cabled connection.  I also noticed that my
3rd card installed fine but hangs every now and then on network traffic.
This card is running DietPi just fine except for the network issue.

I already contacted PINE64 but not sure where that will go. Hope I do not
have 3 boat anchors.

Thanks for all the help

Thomas M. Ose
Ose Micro Solutions, Inc.

Fourdee, 2016-10-07:
@tmose 

>  I
now think that my problem may be related to the faulty network chip issue
others were seeing, and may explain why under other distros I was only
getting 100 connection.

Ah yes, remember seeing a few forum posts on network issues. its also been confirmed by Pine officially as an issue.

http://forum.pine64.org/showthread.php?tid=835&pid=19704#pid19704

So your boards can still be used, i'll make a option in ```dietpi.txt``` that will force 100Mbit when set. Give me a day or two and i'll let you know when its ready for testing.

Fourdee, 2016-10-07:
@tmose 

Ok, new image ready for testing where we can force 100mbit, please do the following:
- Download updated image: http://dietpi.com/downloads/testing/DietPi_PineA64-arm64-(Jessie).7z
- Extract and write to a SD card
- Before powering on. Open up ```/boot``` and then open the file ```dietpi.txt```
- Find this line and change the value to  ```Ethernet_Force_Speed=100```
- Save changes, plug in sd card, power on.

tmose, 2016-10-07:
>    testing/DietPi_PineA64-arm64-(Jessie).7z

Dan,

Will try to get to this over the weekend and get back to you with results.

Thanks for all the help.

Thomas M. Ose
Ose Micro Solutions, Inc.

tmose, 2016-10-07:
Dan,

Actually got some time to install.  Install works great with setting to
100.  Did notice Firefox did not start correctly but will work on that and
getting my application to run later.

Thanks again for all the help.

Thomas M. Ose
Ose Micro Solutions, Inc.

Fourdee, 2016-10-07:
@tmose 

> Actually got some time to install.  Install works great with setting to
100.

Excellent, thanks for letting us know 👍 

> Did notice Firefox did not start correctly but will work on that and
getting my application to run later.

Firefox binary on ARM64 is broken (segmentation fault if i remember correctly), same issues on Odroid C2. As a alternative, try installing Chromium instead:
```
dietpi-software install 133
```

Fourdee, 2016-10-08:
Marking this as closed. Please reopen if required.

closed 2016-10-08 by Fourdee