MichaIng/DietPi #546: DietPi-Software | nginx maximum upload size

pfeerick, 2016-10-04:
After upgrading to v132, I tried out Pydio with a nginx webserver backend, and I encountered an error ([413 request entity too large](http://www.cyberciti.biz/faq/linux-unix-bsd-nginx-413-request-entity-too-large/)) when uploading files larger than 2MB. It appears to be a bug (or a feature if you look at it that way) in the nginx configuration. Editing /etc/nginx/nginx.conf to have "client_max_body_size 8796093022207M;" in the http section fixed the problem. I was running this on the pine64, hence the size, copied from the php.ini upload max! ;-)

Fourdee, 2016-10-04:
@pfeerick 

Great find, thanks for letting us know 👍 I'll add that into the nginx conf by default.

Fourdee, 2016-10-04:
Ok, done.

```client_max_body_size ``` will be set to same value as ```php_max_upload_size="$(( $(php -r 'print(PHP_INT_MAX);') / 1024 / 1024))M"```

Thanks again @pfeerick , great stuff 👍

closed 2016-10-04 by Fourdee

pfeerick, 2016-10-05:
No worries... thanks for the quick patch! :D I was hoping you'd use the PHP max upload size value for that also!  👍