MichaIng/DietPi #548: Personal notes | My little file

Fourdee, 2016-10-05:
===

Wheezy:
```
[ o.k. ] Ending debootstrap process and preparing cache [ wheezy ]
wheezy-ng-armhf.382...293.tgz:  200MB [ 157MB/s] [============================================] 109%
[ o.k. ] Applying distribution specific tweaks for [ wheezy ]
sed: can't read /root/output/cache/sdcard/etc/default/ramlog: No such file or directory
[ o.k. ] Applying common tweaks
sed: can't read /root/output/cache/sdcard/etc/default/console-setup: No such file or directory
[ o.k. ] Installing kernel [ linux-image-sun8i ]
[ o.k. ] Installing u-boot [ linux-u-boot-bananapim2plus ]
[ o.k. ] Installing headers [ linux-headers-sun8i ]
[ o.k. ] Installing board support package [ bananapim2plus ]
sed: can't read /root/output/cache/sdcard/etc/lirc/hardware.conf: No such file or directory
sed: can't read /root/output/cache/sdcard/etc/lirc/hardware.conf: No such file or directory
sed: can't read /root/output/cache/sdcard/etc/lirc/hardware.conf: No such file or directory
cp: cannot create regular file ‘/root/output/cache/sdcard/etc/lirc/lircd.conf’: No such file or directory
[ o.k. ] Installing extra applications and drivers
[ o.k. ] Merging and packaging linux firmware [ @host ]
[ o.k. ] Installing linux firmware [ 5.21 ]
[ o.k. ] Installing additional driver [ MT7601U ]
[ o.k. ] Checking git sources [ mt7601 old ]
[ .... ] Creating local copy
[ .... ] Fetching updates
remote: Counting objects: 264, done.
remote: Compressing objects: 100% (207/207), done.
remote: Total 264 (delta 54), reused 255 (delta 53), pack-reused 0
Receiving objects: 100% (264/264), 1.29 MiB | 649.00 KiB/s, done.
Resolving deltas: 100% (54/54), done.
From https://github.com/igorpecovnik/mt7601
 * branch            old        -> FETCH_HEAD
 * [new branch]      old        -> origin/old
[ .... ] Checking out
[ o.k. ] Installing additional driver [ RT8192 ]
[ o.k. ] Checking git sources [ rt8192cu master ]
[ .... ] Creating local copy
[ .... ] Fetching updates
remote: Counting objects: 201, done.
remote: Compressing objects: 100% (177/177), done.
remote: Total 201 (delta 43), reused 71 (delta 21), pack-reused 0
Receiving objects: 100% (201/201), 1.17 MiB | 1.25 MiB/s, done.
Resolving deltas: 100% (43/43), done.
From https://github.com/dz0ny/rt8192cu
 * branch            master     -> FETCH_HEAD
 * [new branch]      master     -> origin/master
[ .... ] Checking out
[ o.k. ] Building deb [ armbian-tools ]
[ o.k. ] ... downloading sources [ temper ]
[ o.k. ] ... downloading sources [ BT utils ]
[ o.k. ] ... compiling [ temper ]
[ error ] ERROR in function compiling [ tools.sh:70 ]
[ error ] Error building [ temper ]
[ o.k. ] Process terminated
[ error ] ERROR in function unmount_on_exit [ debootstrap-ng.sh:576 ]
[ error ] debootstrap-ng was interrupted
[ o.k. ] Process terminated
root@vm-crosscomp:~#
```

Fourdee, 2016-11-04:
384khz

speaker-test -c2 -r 384000 -F S32_LE

cat /proc/asound/card*/pcm0p/sub0/hw_params

Fourdee, 2016-12-28:
Sparky:
https://github.com/volumio/platform-sparky
https://github.com/sparkysbc/Linux

Fourdee, 2017-01-04:
Pair bluetooth speaker (rpi 3 ):
- Turn on the Bluetooth speaker and enter discovery mode
- ```bluetoothctl ```
- ```power on ```
- ```agent on```
- ```scan on```
- Pair with your Bluetooth speaker pair MAC Address, at this point there should be no pass code to enter
- ```trust MAC Address```
- ```connect MAC Address```

Fourdee, 2017-01-14:
Pine case:
- 28'c idle
- 80'c @ 5mins
- 18'c room

Fourdee, 2017-01-28:
H3 audio commits 4.10
https://github.com/torvalds/linux/search?p=1&q=h3+audio&type=Commits&utf8=%E2%9C%93

Fourdee, 2017-01-29:
JustBoom IR remote install:
```
wget http://dietpi.com/downloads/misc/community/install_justboom_ir_remote.sh
sudo chmod +x install_justboom_ir_remote.sh
sudo ./install_justboom_ir_remote.sh remoteir justboom_ir_remote
```

Fourdee, 2017-02-06:
```
pcm.mmap0 {
    type mmap_emul;
    slave {
      pcm "hw:1,0";
    }
}

pcm.!default {
  type plug;
  slave {
    pcm mmap0;
  }
}

ctl.!default {
    type hw
    card 1
}
```

Fourdee, 2017-03-27:
Clear all comments in a script, that start with whitespace and are 1st entry on line (excludes ^#):
```
sed -i '/^\s*[[:space:]]#/d' /DietPi/dietpi/boot
```

Fourdee, 2018-02-04:
@k-plan @MichaIng 

Reference:
https://twitter.com/armbian/status/960225496918306817

MichaIng, 2018-02-04:
@Fourdee 
Just ignore. In the end users decide what kind of value something is.
The value of open source is, that using, changing and in case "rebranding" is possible, so you have a rich offer of solutions where users can decide what they need.

k-plan, 2018-02-04:
@Fourdee 
Who is armbian on twitter? Igor? Mikhail? Think not, sound for me like mister tk .... :rofl: 
If so ... ignore. 💩  😃 

It's only indicate, you are on on the right path. 
Grudge will only be generated if somebody do thinks well and meet with success.
😉 

@MichaIng 
> Just ignore.
Hmm .... this is not as easy as it sound, to take it not personally.
Revile like this are for quite some time and be seated deep-rooted.

Fourdee, 2018-02-04:
@k-plan 

> tk

I'd expect no less 😉 

> Just ignore. In the end users decide what kind of value something is.

I'd love to, but ARMbian's method of dealing with "competition", is to use the users inexperience and a method of fear control, to spread negative non-sense:
https://forum.armbian.com/search/?&q=dietpi&author=tkaiser

I on the other hand, will only reduce myself to that level, if i'am provoked and feel the need to defend what i'am passionate about and believe in. In most cases, I have the facts to back up any claims I make (unlike ARMbian/TK's).

Just winds me up, I have better things to do deal with one man's lack of control, whos only self worth is to generate fear and non-sense using ignorance and disrespect.

Fourdee, 2018-02-05:
@k-plan @MichaIng
Fed up with this non-sense, felt the need: 
http://dietpi.com/phpbb/viewtopic.php?f=9&t=2794&p=10691#p10691

Fourdee, 2018-02-16:
N1:
- Case was fun, although, needs a step by step guide for all users
- Molex power fitting is extremely tight, may bend/break board for some users if done incorrectly.
- Onboard SATA slot lacks the clip system with SATA2.0+, means the cable is likely to come out unexpectedly.

Fourdee, 2018-02-17:
@MichaIng @k-plan 

We featured in Linux Format magazine (9/10 score) 💘 :
http://dietpi.com/phpbb/viewtopic.php?f=9&t=2857#p10989

k-plan, 2018-02-17:
> Linux Format magazine have a review in Issue 233 for DietPi (page 51):

https://www.pressreader.com/australia/linux-format/20180116/281681140273193

> DietPi server...........................
> In his quest for more Pi, Les Pounder is told to go on a diet. We think he may have misunderstood the request...
> 
> Linux Format 16 Jan 2018
> 
> DietPi may not look pretty, but under the hood we have a powerful yet easy-touse server that’s suitable for all levels of user.
>
> Les Pounder loves his pies too much, but didn’t realise this was made to make his Pi lose weight.
>
> When someone says Raspberry Pi, then of course you think Raspbian, the default distribution for the Pi. However, Raspbian is a big distribution that requires at least an 8GB microSD card for all its great software. What if you want to build your own Pi server? Do you really need all that bloat? Sounds like we need to go on a diet!
>
> DietPi is a lightweight distribution that uses the base Raspbian (Debian) distribution, DietPi kernel 4.9.52 from 2 October 2017, and puts it on the cabbage soup diet. Gone is the desktop, Sonic Pi and other popular applications and in their place we find a bare bones server distribution offering stability and low-power consumption.
>
> But DietPi has a neat trick up its sleeve: its ease of use. Typically, setting up a server takes time and skill, but what DietPi provides is an easy-to-use workflow for those wishing to set up a simple server in their home or development environment.
>
> DietPi is best enjoyed on a 4GB or larger microSD card and it’s compatible with every model of Raspberry Pi, even the very first model of Pi back in 2012! Installing DietPi on an SD is identical to Raspbian, requiring the image to be written to a blank card. On boot DietPi doesn’t have a flash boot sequence; rather, it gets to work straight away. Via an interactive installer that updates the distribution to the latest software, very handy as this ensures the system is up to date and ready to be used.
>
> Once installed, DietPi uses a whiptail menu system, which can be called at any time from the command line, enabling anyone to administrate their system. We used the Ethernet configuration tool to copy our interface settings (from DHCP) and set them as a static IP, all by pressing one key!
>
> Simple stuff
>
> Installing software is also ridiculously easy. DietPi has its own software menu that includes software which has been optimised for use with the distribution. To test installing software with DietPi we went to the Software Optimised menu and found ourselves in wonderland!
>
> The list of optimised software isn’t exhaustive, but it covers many popular applications for a Raspberry Pi. NFS, Samba and Cloudprint servers can be easily installed, and we even found an entry to install OctoPrint, a web interface for 3D printers. We chose to install PiHole, an ad-blocking DNS server that sits on your home network and stops ads from being served to devices using the DNS server. The install for PiHole took around ten minutes, largely due to a number of package installs. But once installed all we needed to do was point our devices to the DietPi/ PiHole DNS server and everything worked.
>
> The simplicity is the beauty of DietPi. From the menu we can set up our Ethernet/WiFi to use static IP addresses. We can configure Bluetooth and I2C for devices and sensors. We can stress-test the RAM, CPU and file system to ensure that our platform is fit for purpose. DietPi can also be used in the same manner as a normal Raspbian or Debian server.
>
> We’re impressed with DietPi. It’s a lightweight, practical distribution for those that just want to quickly deploy a server. It may not be for hardcore sysadmins, as servers typically require “tweaking” for the best results (and there’s nothing stopping a sysadmin from doing that), but for the Pi user wishing to try something new, DietPi is an ideal choice. It may even bring new life to older models of Raspberry Pi.

MichaIng, 2018-02-18:
@Fourdee 
Btw. great step and words you found in dialog with ARMbian, also from their side it sounds promising. I really hope this brings the unnecessary conflict to an end.

I don't have an ARMbian device here to see, but generally we could pronounce credits for kernel and base image providers more, to counteract the feeling that some people from ARMbian seemed to have? Liking or disliking the final product is everybody's freedom, so nothing about that.

k-plan, 2018-02-18:
@MichaIng 
> Btw. great step and words you found in dialog with ARMbian

Yes, agree 💯  👍 

> also from their side it sounds promising.

Hmm ... no. It's only Mikhail as far as I noticed. He is okay, have help out alot in past.
It is a little bit more difficult and problems are more deeper. Igor is the head, the founder of ARMbian.
Please have a look: http://www.armbian.com/authors/
First row is the importent.
Mr. tk is since a long time go to the barricades:  [Example](https://forum.armbian.com/topic/1108-security-alert-for-allwinner-sun8i-h3a83th8/?tab=comments#comment-17460)

The hole thing is similar to the problems with some members of Pihole projekt.
Head/founder: `jacobsalmela` , similar to `Igor`
Project member how will help out: `DL6ER` , similar to `Mikhail`
Heater: Mr. tk  , similar to Pro-mo-Faux

From my point of view, the main coder have no problem with Dietpi.
But the others don't grant any appreciation to other projects based on their work.

That's why I have put in to cancel all cheap  Chinese boards.
No support for Orange-. Banana-, Nano-Pi rubish  stuff  => no need for armbian any more ...
Who cares, if we support thirty or fifteen SBC. Most problems and support goes into this 10 $ garbage.

> I don't have an ARMbian device here to see ...

You can get one form me, but you don't want it ... 😃 

Igor, Mikhail and others did a great job. This boards will not work without them.
The manufacturer did nothing, beside take your money and throw every month a new board on the market.

MichaIng, 2018-02-18:
@k-plan 
Okay, let's see how things develop, we/Fourdee can not do much than trying to defuse/calm from our side and do best to pronounce and give credits enough to the work we are based on.

From PiHole there was/is something similar? There I can absolutely not understand, as we it is the total other way round: Not we are based on their work, but they do benefit from every Distro/image that offers their work! So they should be thankful for everyone, who provide PiHole, or not? 🤔 Okay at least of it is not implemented so badly, that claims finally arrive them, instead of Distro maintainer for bad implementation 😉.  At least during my tests PiHole worked fine always. We just need to remove the `pihole-FPL` service on uninstall, always forget to add PR about that after recognizing again and again 😆.

> The manufacturer did nothing, beside to take your money and throw every month a new board on the market.

No idea who much they support kernel/hardware developers with samples, API, help etc., but yeah, I am always wondering how often there would be exactly no ready-to-run image for a board, without work of ARMbian and others. One should think, that board and well maintained image belongs strictly together. But yeah, that is also open source: Many people providing much for nothing but interest, joy and believe in doing something helpful 😏.

k-plan, 2018-02-18:
@MichaIng 

> From PiHole there was/is something similar?

You can make your own experience. Simple ask here, you have DietPi with PiHole installed and now have a problem to bring your PiHole version up-to-date:

https://www.reddit.com/r/pihole/

https://discourse.pi-hole.net/

https://twitter.com/The_Pi_Hole

k-plan, 2018-02-18:
@MichaIng 

> At least during my tests PiHole worked fine always. We just need to remove the pihole-FPL service on uninstall, ... 

Yes, tested to uninstall. It's correct, `pihole-FTL` have to be removed manually. 

> always forget to add PR about that after recognizing again and again 😆.

Yes, please do so. 👍  

And if you can, a working fix for `pihole -up` will be very nice.

As well as a installation fix and freeze to `Pi-hole Version v3.2.1` | `Web Interface Version v3.2.1` |  `FTL Version  v2.13.2 ` for all DietPi v6.x images based on Jessie (Odroid and RPi)

Nice feature will be to have now the updated block page.
[image]

😃  ... just need this to get a always fine working PiHole installation.

> instead of Distro maintainer for bad implementation

> .. maybe due to new version (of PiHole server) there is something broken? I also broke my VM now by uninstalling PiHole and one "automated installed" dependency too much. Their uninstaller is a pain: Why not just auto-marking instead of purging, including all possibly important core system packages automatically...

I have to say nothing ....  🙊  😄 

```
DietPi-Software
─────────────────────────────────────────────────────
 Mode: Uninstall

[  OK  ] Uninstalling Pi-hole: block adverts for any device on your network
  [?] Are you sure you would like to remove Pi-hole? [y/N] y
  [✓] Root user check
  [i] All dependencies are safe to remove on Raspbian
  [i] The following dependencies may have been added by the Pi-hole install:
    apt-utils dialog debconf dhcpcd5 git iproute2 whiptail bc cron curl dnsmasq dnsutils iputils-ping lsof netcat sudo unzip wget idn2 lighttpd php-common php-cgi php-sqlite3
  [?] Do you wish to go through each dependency for removal? (Choosing No will leave all dependencies installed) [Y/n] n

```
Please NEVER use suggested uninstall defaults. 😃 
They have until now only RPi and Raspbian on their focus! 🙈

k-plan, 2018-02-18:
@Fourdee 
@MichaIng 
@  project evangelists 😄 

After reading igors statment, all should be clear form now on. What a shame.

[image]

Will stop reading over there, think now spitfire and shitstorm will take place. :popcorn: 

My individual route of march is certain.
Unfortunately, I have not be wrong.

Sorry.

MichaIng, 2018-02-18:
I don't get it, does he actually want to have a "solution" (in means of communication in how to solve, what ever he thinks is unfair or something)? I hardly believe that any of our users ask for any support on ARMbian, so that can't be it...

If he would be a bid concrete in what should be actually the problem to bring a down sized image (thereby totally ignoring all the other benefits of DietPi, easy configuration software installation, etc.), it could be considered. But just saying there ARE (definitely!) issues or something important is "lost", but not showing any evidence about what, at least one! example, does really not make sense at all. That small/cleaned images mean you might have to install more packages back for certain software is clear, and that we have to take care/test to not remove anything essential for stable running system is clear as well, but me and I guess many others like the idea, to have it that way, have first just what is really needed and install stuff on demand instead of speculation. I am tired or overloaded software (bundles) filling drives, file system or tasks overviews, even if in some cases that might not take "too" much system resources. Have enough of this from Windows and it's totally overloaded bloaded grown standard programs: Adobe Reader, Windows Media Player, Winzip (okay is this actually still in use?) and all the **it like this, that by times even cost money but does it's job way less fast and good than tiny open source alternatives, being only in use as it's installed by default and people think it is the only solution, at least the only, known to them... This is what I love about Linux, you have much more control and possibilities in what and how, so I like to go until the edge of what is possible. But what I am arguing here 😆.

So at least if it's about Igor, do not mention ARMbian any more at all within DietPi, to not risk any of our users address any issues to them? And I thought the opposite might have been wanted 😆 ...
We don't "need" to care, besides respecting `GPLv2`.

And yeah, evangelists, another topic.. He likes so mix thinks together the way it fits him right now... What was the age? 🙄 Looks like there is not much chance to do anything "right" as such a person will find always something "wrong" then. So yes @k-plan looks you were right 🙈

k-plan, 2018-02-18:
Psst .. Micha

> But just saying there ARE (definitely!) issues or something important is "lost", but not showing any evidence about what, at least one! example, does really not make sense at all.

That is accurate the way or point, to get Fourdee in rage and he will do/say silly things.  
 I think, he does know this precisely from former arguing between Mr. tk .
Deliberate provocation, individual fitted. Really clever, but bad style. 👎 

The point is, user think and say loud, DietPi is the better ARMbian.
So it is a personal harassment for igor, Mr. tk and other for things they are doing with great personal passion and conviction.

And who is now the project evangelists ?  😃 

With PiHole it is the same.

And yes, we do as well your thing with personal passion and conviction.
Perhaps out of different motivation.
And at this point, it becomes irrational ....

MichaIng, 2018-02-18:
> Psst .. Micha
? Oh I think that could be misunderstood: No I mean that Igor (and Mr. TK via Twitter as well) seem so sure that there are issues in how DietPi does. I do absolutely not agree, that's the reason I am supporting especially the removal of even more packages: dbus (+logind), nignx -> nginx-light, and the whole apt-mark auto everything first thing 😄. That is how I like things do: Start with as much nothing as possible and add only on actually demand.

k-plan, 2018-02-18:
Micha, ich glaube, ich habe dich schon verstanden.

Wenn es dafür belegbare technische Gründe geben sollte, was ich sicherlich nicht beurteilen kann,  warum legt man diese nicht offen?
Besonders, wenn man direkt darauf angesprochen wird? So ein riesiges Geheimnis? Oder gibt es gar keine? Alles eine Glaubensfrage?
Oder wie Igor selber schreibt: `- what exactly is mystified -` 😄 
Ich habe bisher nur Polemik lesen dürfen.

Ist doch alles Kinderkacke, aber Dan geht jedes Mal direkt steil dabei.

MichaIng, 2018-02-18:
@k-plan 
Ah lol, okay I was not reading precisely enough, a bid jumping between thesis, wife, DietPi and spontaneously baking cake in the kitchen 😆. To switch to better topic: Apple poppy (=Mohn) vanilla remains, without recipe, and became better than the last one, strictly made by recipe from www.chefkoch.de with promising hight rating 🤔.

k-plan, 2018-02-18:
Was sicher richtig ist, ist die Sache mit dem `root` , aktivierten SSH Zugriff und vorinstallierten SSH Zertifikat.

Aber:
- ich finde es praktischer/bequemer als das dauernde `sudo` Zeug. Vanilla Debian geht mir da echt auf den Sack ohne root user, Zwangspasswortänderung, vorgeschiebene Zeichenauswahl und Länge.
- A la Raspian vorher per 'magic' Dateierzeugen SSH zu aktivieren ist doch auch nicht das Wahre. Wem soll das nutzen?

DietPi ist für Anfänger, Einsteiger , Endanwender zur privaten Nutzung.
Sollte somit nur im LAN laufen und nicht direkt aus dem Internet erreichbar ein.
Und der Einstieg und das Nutzen sollte simpel und einfach sein.
Wenn ich schon Sicherheitsbedenken im privaten LAN habe, liegen meine Probleme schon ganz wo anders.

Ja, um DietPi als öffentlich erreichbaren Server im Internet zu betreiben, ist es weniger geeignet, wenn man Null Ahnung hat und sich auch nicht damit beschäftigen will.

Sicherheit kostet immer Komfort und Einfachheit.
Und absolute Sicherheit wird es bei vernetzten Computern nie geben, es sei denn, wir lassen den Stromanschluss aus der Dose.

Wie immer gilt, Hirn und Verstand ist durch keine Technik und Software zu ersetzen.

MichaIng, 2018-02-18:
@k-plan 
Ja naja, bevor es mit SSH richtig gefährlich werden kann, muss der port ja auch erst mal freigegeben/weitergeleitet werden. Und wer braucht schon von überall Zugriff auf sein Serverterminal... Dann ist ein sicheres Passwort die nächste Hürde und im Fall bieten wir fail2ban mit strengerem (als Standard) vorinstalliertem SSH Schutz.
Ich selbst nutze immer key authentication + pass phrase, habe schon überlegt die Einrichtung mal in dietpi-config>security options einzubauen. Bin nur gerade nicht sicher, ob sich die Keys vom Server aus gut im PuTTY compatiblen Format ausspucken lassen. Erzeuge die von PuTTYgen aus.

Und ja, habe es auch in Windows vor über nem Jahrzehnt mal mit eingeschränktem Standardnutzer versucht, wie überall empfohlen. Aber schon da hat es mich genervt, andauernd wechseln zu müssen, da ich eigentlich immer mal zwischendurch am basteln, einstellen, installieren etc. bin. Insofern schnell aufgegeben und auf single admin user gewechselt. In Linux dasselbe...

Für den Anfänger DietPi Nutzer wäre ein standard non-root Account aber wohl wirklich ganz gut, allerdings bitte mit optionalem Wechsel zurück zu root als Standard 😉. Ist ja auch ein offenes Thema, aber definitiv nicht mal eben gemacht. Schritt für Schritt gehen wir ja in die Richtung, Nutzer besteht und bekommt langsam immer mehr Aufgaben. MySQL Software hat inzwischen ja auch eigene mysql user, der Wechsel der login scripts (alias / globals / Banner) von ~/.bashrc zu /etc/profile.d ist geplant, was ja genau aus einem Userwunsch entstand, der DietPi als non-root betreibt. Scheint also schon das meiste zu gehen.

k-plan, 2018-02-18:
@MichaIng 

War keine Kritik, ich finde es voll in Ordnung so wie es im Moment ist. Kann gerne so bleiben.
Wem es nicht gefällt, der kann sich ja was anderes aussuchen. Gibt ja genug Distributionen.

Habe es nur angeführt, weil es die mir zur Zeit einzigen schlüssigen Ansatzpunkte sind, die ich so zwischen den Zeilen heraus lesen konnte. Abgesehen von Polemik.

Bin schon der Überzeugung, mit den vorgegebenen Projekt Prämissen, machen wir zur Zeit noch fast alles richtig. Besonders haben viele Sachen, seit du unterstützt, neue Fahrt aufgenommen. Bin fest überzeugt, Dan ist mega glücklich, dass er jetzt adäquate Unterstützung hat. Mich freut es jedenfalls sehr.

Und lass dir den Kuchen schmecken, wenn er denn fertig ist.

Fourdee, 2018-02-18:
@k-plan @MichaIng 

If i'am honest, I expected better from Igor.

> But just saying "there are definitely issues" or something important is "lost", but not showing any evidence about what, at least one! 

This is exactly ARMbian's problem. They never can provide any proof and/or facts to back up their "claims" when slagging us off (or even them declaring something in general). Where as, I, am able (hence the reason for http://dietpi.com/phpbb/viewtopic.php?f=9&t=2794#p10691, and having to defend TK's nonsense claims).

ARMbian is very one sided. 
What they say is always "correct", everyone else is "wrong", even if there is no evidence to back it up their side (99% of the time). 
Its a tight nit comminuty (Igor/TK) of one sided people who get satisfaction controlling their users and enforcing "blind correctness" on outdated Linux ways.

I think its clear, Igor + TK are stuck in their ways. They follow the Linux comminuty of devs (also stuck in their ways, remember SystemD being slagged off, yet, its more efficient with multi-threading services and so simple to create a service). They dislike, someone coming along with a fresh approach to Linux, that clearly works for us and our users. At the end of the day, we cut the bullshit out of Linux, and essentially that comminuty which Igor + TK belong to, and, want to "control".

I'll leave the reply for now, or i'll show my anger. Drives me up the fucking wall. 

Notes:

> You are obviously not communicating your work properly if your users poke us with such funny statements: 
"I think @armbian is good too, in cases where there is no @DietPi_ "
"Thus, there is room for improvement and that is clearly what @DietPi_ successful does, ~81 Mb as a downloaded zip."
"Dietpi is faster meaner and leaner than Armbian.

Works both ways, we receive complaints regarding ARMbian, but we dont encourage it or slag them off in our reply.

What we can do: "Do not contact ARMbian for any support, please contact us" on website.

Fourdee, 2018-02-18:
@MichaIng @k-plan 
And I know for a fact, TK slags us off on various forums (Odroid/NanoPi), its his day job to troll. I think regardless of what we do, its always going to continue. They simply do not like improved changed, and a new way of thinking (which they cant control).

Shame we all can't be adults.

Fourdee, 2018-02-18:
@MichaIng @k-plan 
And that not to say we are "perfect" by any means, but we are evolving and constantly improving.

To sum it up, the way I see it:
- ARMbian = The world is flat
- DietPi = The world is round

Regardless of common sense, or if we provide a 100 page proof, we are always the abnormality and target for their childish enjoyment.

Right, off to bed, calm down, sleep on it 👿

k-plan, 2018-02-18:
@Fourdee 

> If i'am honest, I expected better from Igor.

Yes 💯  correct. Me too.

> I'll leave the reply for now ...

Please do so. Good decision.  And please stick on this. 👍 

> or i'll show my anger. Drives me up the fucking wall.

> That is accurate the way or point, to get Fourdee in rage and he will do/say silly things.
> I think, igor does know this precisely from former arguing between Mr. tk vs. Dan.
> Deliberate provocation, individual fitted. Really clever, but bad style. 👎

> I think regardless of what we do, its always going to continue.

Yes, I think the same.

> Shame we all can't be adults.

😃  Really? 

> slags us off on various forums (Odroid/NanoPi)

Yes, have noticed this. See that Igor and TK are now on HardKernel "developer sample list".
They follow you up.

> What we can do: "Do not contact ARMbian for any support, please contact us" on website.

or:

> That's why I have put in to cancel all cheap Chinese boards.
> No support for Orange-. Banana-, Nano-Pi rubish stuff => no need for armbian any more ...
> Who cares, if we support thirty or fifteen SBC. Most problems and support goes into this 10 $ garbage.

 To get out of the way from this childish affectation.
But we have to think about this.

> Right, off to bed, calm down, sleep on it 👿

Yes, 👍  good idea. Good night Dan.

Fourdee, 2018-02-18:
@k-plan 

> That's why I have put in to cancel all cheap Chinese boards.

I've been considering this for a long time, dropping ARMbian images. More hassle than its worth. But I feel it will not change their actions, it will always continue regardless.

I mean for christ sakes. All this recent stuff all started because we got a 9/10 review in Linux Format Magazine and it pissed them off. Regardless, we must be doing something right.

Yep, i'am off. Vodka + Coke doing justice and maybe watch some Death in Paradise (really chills you out) http://www.bbc.co.uk/programmes/b01pvmf6

Just another happy day, experiencing the joys of open source love, spread around the world 😉

Fourdee, 2018-02-18:
@MichaIng 
Apologies for the language. I can "usually" keep a lid on it 😉 

---

@k-plan 

Night Frank 😃

k-plan, 2018-02-18:
@Fourdee 

> But I feel it will not change their actions, it will always continue regardless.

You can be right at this point, but I'm unsure.

> Regardless, we must be doing something right.

Yep, 💯  👍   😄

MichaIng, 2018-02-18:
> 😃 Really?

👍 Stay child where you can, but become adult, if it hurts other people, or something like that 😄.

> Who cares, if we support thirty or fifteen SBC. Most problems and support goes into this 10 $ garbage.

Sadly some of our users would be the ones, that finally pay for this then. And I guess most didn't know before buying, about how good/bad the producers image offer/maintenance and what 3rd party images/distros exist.

But yeah, sleep over it, forget all bad, remember all good, I with out both a good night and great start into the new week 😃 @Fourdee @k-plan

k-plan, 2018-02-18:
_k-plan locked as too heated and limited conversation to collaborators just now_
 _but keep a eye on it, every one can read it_

Fourdee, 2018-02-19:
@MichaIng @k-plan 

[image]
https://github.com/Fourdee/DietPi/issues/1537

k-plan, 2018-02-19:
------------------

**My little personal notes for pihole problems:**

------------

[Pi-hole v3.3 Released: It’s “Extra” Special](https://pi-hole.net/2018/02/14/pi-hole-v3-3-released-its-extra-special/)

Problem:

> If you’re running Jessie and you updated Pi-hole to v3.3, you likely ran into issues.  
> This is because the version of `dnsmasq` that ships with it does not support the ` log-queries=extra ` option, which we use in v3.3.

_Update 2018-02-14 18:43 (version issues and not working after update)_

> If you’re running Jessie, your version of `dnsmasq` will not work with this release, 
> so you’ll need to revert to the previous versions:
> 
> ```
> cd /etc/.pihole
> sudo git fetch --tags
> sudo git checkout v3.2.1
> cd /var/www/html/admin
> sudo git fetch --tags
> sudo git checkout v3.2.1
> cd "$HOME"
> pihole -r
> pihole checkout ftl v2.13.2
> pihole -v
> ```
> 

Option two  **ONLY RPi and BEFORE any Update is started**: 
- install the version of `dnsmasq` that supports the extra flag (v2.76)

> Please note, you should only try this on **Rasbpian Jessie** and do so at your own risk (but in our opinion the risk is low)
> 
> First step: Download more recent version of `dnsmasq` compiled for **Raspbian Jessie** from the official sources
> 

> ```
> wget https://archive.raspberrypi.org/debian/pool/main/d/dnsmasq/dnsmasq-base_2.76-5+rpi1_armhf.deb
> wget https://archive.raspberrypi.org/debian/pool/main/d/dnsmasq/dnsmasq_2.76-5+rpi1_all.deb
> 
> ## Ensure requirements are fulfilled
> sudo apt-get install libnetfilter-conntrack3 libmnl0
> 
> ## Install downloaded packages
> sudo dpkg -i dnsmasq-base_2.76-5+rpi1_armhf.deb
> sudo dpkg -i dnsmasq_2.76-5+rpi1_all.deb
> 
> ## Verify it worked
> dnsmasq -v
> 
> Dnsmasq version 2.76  Copyright (c) 2000-2016 Simon Kelley
> Compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify
> 
> pihole -up
> 
> ```

-------------

 **DietPi pihole update problem:**
(tested with Jessie and Stretch, DietPi v15x and v6.x, Pi-hole v3.1 => v3.2 => v3.3 )

- if it is ONLY a FTL version update, 
```
:~# pihole -v
  Pi-hole version is v3.2 (Latest: v3.2)
  AdminLTE version is v3.2 (Latest: v3.2)
  FTL version is vDev-437af07 (Latest: v2.13.1)
```
you can simple start `pihole -up` without doing anything else. (tested - work)

- if it is a pi-hole core and/or webinterface update
```
:~# pihole -v
  Pi-hole version is v3.2 (Latest: v3.2.1)
  AdminLTE version is v3.2 (Latest: v3.2.1)
  FTL version is vDev-437af07 (Latest: v2.13.1)
```

you have to do  (tested - work for me) :

```
#Exit path for non-root logins.
	if (( $UID != 0 )); then

		echo -e 'Error: Root privileges required. Please run the command with "sudo"\n'
		exit 1

	fi

##  have a look, what is to  update
 pihole -v

## if only FTL version have to be updated // can be omitted
# pihole -up
## recheck version (if successful update)
# pihole -v
## finished

## if it is a core/AdminLTE version update or more - Backup lighttpd.conf
mv /etc/lighttpd/lighttpd.conf /etc/lighttpd/lighttpd.conf.bak

## https://discourse.pi-hole.net/t/why-does-the-pi-hole-installer-produce-the-error-unable-to-complete-update-contact-pi-hole/4159
cd /etc/.pihole;git reset --hard
cd /var/www/html/admin;git reset --hard
cd "$HOME"

## now update pi-hole
pihole -up

## recheck version (if successful update)
pihole -v

## Restore lighttpd.conf
cp /etc/lighttpd/lighttpd.conf.bak /etc/lighttpd/lighttpd.conf

## Restart all running services
dietpi-services restart

## finished
```

see:
`https://github.com/Fourdee/DietPi/issues/1375`

https://discourse.pi-hole.net/t/why-does-the-pi-hole-installer-produce-the-error-unable-to-complete-update-contact-pi-hole/4159

-------------------

**DietPi will not remove `pihole` completely**
( running service ` pihole-FTL`  after uninstall )

> @ MichaIng
> 
> At least during my tests PiHole worked fine always. We just need to remove the pihole-FPL service on uninstall, ...

Yes, tested to uninstall. It's correct, pihole-FTL have to be removed manually.

see:
`https://github.com/Fourdee/DietPi/issues/548#issuecomment-366512459`

---------------------------------

**DietPi missing pihole updated block page (since v3.x)**

see:
`https://github.com/Fourdee/DietPi/issues/1313`

- no solution so far for all all different web server stacks.

------------------------

**pihole uninstaller **ONLY** optimised for standard **Raspbian** installation**

```
DietPi-Software
─────────────────────────────────────────────────────
 Mode: Uninstall

[  OK  ] Uninstalling Pi-hole: block adverts for any device on your network
  [?] Are you sure you would like to remove Pi-hole? [y/N] y
  [✓] Root user check
  [i] All dependencies are safe to remove on Raspbian
  [i] The following dependencies may have been added by the Pi-hole install:
    apt-utils dialog debconf dhcpcd5 git iproute2 whiptail bc cron curl dnsmasq dnsutils iputils-ping lsof netcat sudo unzip wget idn2 lighttpd php-common php-cgi php-sqlite3
  [?] Do you wish to go through each dependency for removal? (Choosing No will leave all dependencies installed) [Y/n] n

```
Please **NEVER** use suggested **uninstall defaults.** 

see:
`https://github.com/Fourdee/DietPi/issues/548#issuecomment-366534242`

______________________________

Fourdee, 2018-02-19:
@MichaIng @k-plan 

I've cleaned up what was discussed in the heat of the moment. What was said, was needed to be said and read by the people who needed to read it, at that time. 

But I want to thank you both for your support throughout this. I appreciate it more than you know, you are both legends 👍

MichaIng, 2018-02-20:
❤️

k-plan, 2018-02-20:
:heart: 

:hear_no_evil:    :blush:

Done a little bit housekeeping at forum and here.

But now back to our day-to-day routine.

Fourdee, 2018-03-21:
hmm

```dietpi-update```

https://raw.githubusercontent.com/Fourdee/DietPi/testing/dietpi/server_version-6
```
Loop detected on cache-lcy19223-LCY: TuFMEtJVICCO9h0Zpde/N9L9b7tugV9xeSSEt9M0!LCY!cache-lcy19223-LCY
```

Fourdee, 2018-05-06:
@MichaIng @k-plan 
Webserver stats: https://dietpi.com/report.html

k-plan, 2018-05-07:
@Fourdee 

Wow, nice to see, you have reach the goal within 20 hours: :heart: 

https://www.gofundme.com/dietpi#updates

Impressive, the  two really big  donations by TLLim888 (Pine64) and  Simon Fell:

https://twitter.com/TLLim888/status/993340398054473728

https://www.gofundme.com/dietpi

@MichaIng 
- was mich persönlich richtig erfreut ist,  wenn ich denn die Namen richtig interpretiere, die Anzahl der Deutschen Spender für diese Aktion.  :+1: 

@Fourdee : can you please delete `Update 1` , it was out of competition .... :wink:

Fourdee, 2018-05-08:
@MichaIng 

Sent you an email, need your address. To send you the spare dev sample of NanoPi K1+. 😃

Fourdee, 2018-05-10:
Nextcloud install timed tests:
https://twitter.com/DietPi_/status/994654906630983681

Fourdee, 2018-05-18:
@MichaIng 

Dreamed about this last night 😃 https://twitter.com/DietPi_/status/997550976532930561

MichaIng, 2018-05-19:
@Fourdee 
🎉 👍 Fantastic idea!!

We need to test carefully where we implement, as it should also depend if the bottleneck is actually CPU or instead RAM or disk and in case of RAM and disk multi threading might make things slower.

Max threads of course should be based on CPU count minus amount of additional CPUs that the commands use by themselves. But found indeed most commands/processes just using a single CPU.

Fourdee, 2018-07-11:
@MichaIng 

Kids broke my glasses beyond repair 👎 . Can't see anything past 5cm in front of my face 🤣 

I'am going to be out of commission for at least a few days, until new glasses arrive.

v6.12 rolled forward to next week.
Feel free to apply commits as you need them, including PR's. I can check everything (if needed) when I return.

Fourdee, 2018-08-20:
@MichaIng 

GPU is the way to go 🤣 
[image]

Fourdee, 2018-08-29:
@MichaIng 

I've noticed our webserver has a delay during peak hours. Due to Apache2 hitting the worker limit:
[image]

I'll increase workers to 80, see how it goes. Might be an issue with hitting swapdisk though. ``` /etc/apache2/mods-available/mpm_event.conf```

Just need to restart apache2 service, at some point lol 😃

MichaIng, 2018-08-29:
@Fourdee 
Jep, check my one email in between the permissions issue spam (🤣), was thinking if cloudflare allows server restart (also whole machine), while still providing static content (image downloads and all that is related to software installs)? Not sure if just a static web page version is available (see that on other websites, when down, by times) or as well static download content.

This would allow some updates, DietPi to version 6.14 e.g. as well 😄. I think forum and ssh server (survey) down is no big issue, but would be nice if software installs would not fail due to dietpi.com downloads not available.

MichaIng, 2018-09-01:
@Fourdee 
Ah jep, sorry totally forgot to reply. Too busy the last days/weeks 😅. The mugs are great ☕️ 👏.
Do not yet find a task for the SBC. Maybe when study is finished and job is there, so I can afford some media gimmicks 😃.

Many many thanks, I really appreciate ❤️!

Fourdee, 2018-10-05:
@MichaIng 

Server upgraded to 4GB + 4cores. But performance sucks.

Server was pretty much idle, 35 seconds for CPU test.

Gonna upgrade the CPU power rating + £4/month

Fourdee, 2018-10-14:
@MichaIng 

Beta released to public:
https://twitter.com/DietPi_/status/1051493658971779072

MichaIng, 2018-10-14:
@Fourdee 
👍 🥂 

Btw.:
- For the "testing" => "dev" branch switch to have any effect, we need to merge v6.17 with new patch_file into testing, of course. At best we do this with v6.17 release. Just to not forget 😉.

MichaIng, 2019-01-31:
@Fourdee
I am out of town until Saturday night. Most likely without internet access.

[1 more reply]

closed 2019-04-22 by MichaIng