MichaIng/DietPi #550: Add support for generic S905 devices

CGarces, 2016-10-09:
Dietpi has support for Odroid C2 that use S905 soc.
Will be great add support for generic S905 and S905x devices shell as Android boxes.

Currently exist several builds from other projects that can be used as reference.

http://forum.armbian.com/index.php/topic/1143-armbian-for-amlogic-s905/
https://github.com/kszaq/LibreELEC.tv

The mainstream LibreELEC kernel 3.14 looks very good and is frequently updated.
https://github.com/LibreELEC/linux-amlogic/

My skills are very limited, but if someone start a WIP branch I can help with some testing

Fourdee, 2016-10-09:
@CGarces 
At 1st, i thought "We dont want to support this, its not worth it". Then I seen the price of one of these devices (£28).

Interesting, £28 for a Odroid C2 + Case + PSU + EMMC + WIFI: https://www.amazon.co.uk/Android-Amlogic-Cortex-Loaded-Player/dp/B01GPFEU8Q/ref=sr_1_1?s=electronics&ie=UTF8&qid=1476013925&sr=1-1&keywords=s905

So with DietPi currently, we setup our images on-top of existing images, then tweak, optimize and set ready for DietPi.

Therefore, if there is a way to read the EMMC on the device, after this is completed. We could make an image and support all those devices by simply changing the ```.dtb```

So this is an example of how to generate the DietPi image:
- Download/write 
Armbian_5.20_Vegas95_Debian_jessie_3.14.79_20160921.img.xz to EMMC
- Go through the setup script and run commands: https://github.com/Fourdee/DietPi/blob/master/PREP_SYSTEM_FOR_DIETPI.sh
- Under ```#??? Does this device have a unique HW ID index and file? check /DietPi/dietpi/dietpi-obtain_hw_model```, the ID would be ```12```.

In theory, it should work. But I dont have a device to test this.

CGarces, 2016-10-19:
- [x] Install & Boot from ARMBian
- [ ] Install Diet PI
- [ ] Configure WIFI
- [ ] Remote IR
- [ ] Get KODI working properlly

Ok, small progress I have executed  PREP_SYSTEM_FOR_DIETPI.sh with some errors.

See [full log](http://pastebin.com/V7dgg1Kn)

http://mirror.ox.ac.uk give me a 404 so I have used http://archive.raspbian.org/raspbian. BTW not is valid for aarch64.

I get some errors related to missing packages

```
E: Unable to locate package ethtool
E: Unable to locate package p7zip-full
E: Unable to locate package hfsplus
E: Package 'debconf-utils' has no installation candidate
E: Unable to locate package ifmetric
E: Package 'resolvconf' has no installation candidate
E: Package 'zip' has no installation candidate
E: Unable to locate package wput
E: Unable to locate package dphys-swapfile
E: Unable to locate package firmware-realtek
E: Unable to locate package firmware-ralink
E: Package 'firmware-brcm80211' has no installation candidate
E: Unable to locate package firmware-atheros
```
A lot of errors related to paths....

```
./PREP_SYSTEM_FOR_DIETPI.sh: line 153: /boot/dietpi/dietpi-logclear: No such file or directory
./PREP_SYSTEM_FOR_DIETPI.sh: line 159: /boot/dietpi/.install_stage: No such file or directory
./PREP_SYSTEM_FOR_DIETPI.sh: line 386: /DietPi/dietpi/finalise: No such file or directory

```

This are real errors? Should I open separate issues?

Fourdee, 2016-10-19:
@CGarces 

> ./PREP_SYSTEM_FOR_DIETPI.sh

This can't currently be run as a script, manual copy and paste. I'll make a note in the file to prevent other users doing the same.

I'd suggest starting again from the ARMbian image. Manually go through PREP_SYSTEM_FOR_DIETPI.sh.

CGarces, 2016-10-19:
This can't currently be run as a script? so how should be run? I don't get you.

Fourdee, 2016-10-19:
@CGarces 
Read through the file, line by line. Copy and paste the commands.

When you reach sections with the following, copy and paste the whole code:
```
cat << _EOF_ > /etc/apt/sources.list
..........
....something.......
_EOF_
```

CGarces, 2016-10-19:
Ok `understood.

I haven't read the head of the file that said.
**copy the following files and folders to /boot/ https://github.com/Fourdee/DietPi**

And.. ` cat << _EOF_ > /etc/apt/sources.list` is overwriting the   /etc/apt/sources.list so I have executed the script with  http://archive.raspbian.org/raspbian, that I don't need in any case.

Created #571 for people that not read the documentation.

CGarces, 2016-10-19:
I'm really confused with PREP_SYSTEM_FOR_DIETPI.sh
Even executing it Line by line looks like files at `/boot/dietpi/` should be at `/DietPi/dietpi/` but I not see any cp or mv or symlink

Fourdee, 2016-10-19:
> Even executing it Line by line looks like files at /boot/dietpi/ should be at /DietPi/dietpi/ but I not see any cp or mv or symlink

DietPi-RamDisk (```/DietPi```) isn't started until: https://github.com/Fourdee/DietPi/blob/master/PREP_SYSTEM_FOR_DIETPI.sh#L158-L163

So before that point, we have to call physical location ```/boot```, afterwards we call the Ramdisk location ```/DietPi```

 PREP_SYSTEM_FOR_DIETPI.sh is designed to be executed from start to finish. Everything is in the correct order :)

CGarces, 2016-10-19:
Ok I found the error after execute the process step by step in a clean image.

The service not was started correctly 

`service dietpi-service start`

Due a error on dietpi-ramdisk, chmod +x is need.

`chmod +x /boot/dietpi/dietpi-ramdisk`

Also I have fixed the dietpi-logclear

`chmod +x /boot/dietpi/dietpi-logclear`

How you copy the files from `https://github.com/Fourdee/DietPi`?

Probably you have all the files with the correct permissions?

Will update #571 latter-

CGarces, 2016-10-19:
After execute again the script from #571 the system _looks like_ is booting. I think that no image is send to HDMI.
Any ideas to test what is happening?

Fourdee, 2016-10-20:
@CGarces 

> Due a error on dietpi-ramdisk, chmod +x is need.

With fat16/32 ```/boot``` partition, this shouldn't be needed. fat16/32 doesn't support permissions. 
```
root@DietPi-XU4:~# chmod -x /boot/zImage
root@DietPi-XU4:~# ls -lha /boot
-rwxr-xr-x  1 root root 3.2M Oct 16 16:14 zImage
```
For EXT4 etc, can run ```chmod +x -R /boot``` beforehand. I'll put in the sysprep script.

> After execute again the script from #571 the system looks like is booting. I think that no image is send to HDMI.
Any ideas to test what is happening?

Not sure, are you able to boot the base ARMbian image on the device and get a screen output?

Fourdee, 2016-10-20:
@CGarces 

ARMbian build tools (if you used them), the process I use (sets FAT32 ```/boot``` partition)
```
git clone https://github.com/igorpecovnik/lib --depth 1
cp lib/compile.sh .

#Modify compile.sh for
# - en_GB
sed -i 's/en_US/en_GB/' compile.sh
# - disable dialog interface

#Must be added to top
#echo -e "PROGRESS_DISPLAY=plain" >> compile.sh
sed -i "/KERNEL_ONLY=/a PROGRESS_DISPLAY=plain" compile.sh

#Modify lib/configuration.sh
# + BOOTSIZE="64"
# bootsize exists in latest version, change value
# - packages
sed -i '/BOOTSIZE=/c\BOOTSIZE=64' lib/configuration.sh

#rootpw
sed -i '/ROOTPWD=/c\ROOTPWD="dietpi"' lib/configuration.sh

./compile.sh
```

CGarces, 2016-10-20:
The ARMBian image boots without problem

The image used was
Armbian_5.21_Amlogic-s905x_Debian_jessie_3.14.29_desktop_20161018.img.xz

At https://yadi.sk/d/srrtn6kpnsKz2/Linux/ARMBIAN

This is the structure to the device that contains my ARMBian

```
Disk /dev/sdb: 4 GB, 4046837760 bytes
255 heads, 63 sectors/track, 492 cylinders
Units = cylinders of 16065 * 512 = 8225280 bytes

   Device Boot      Start         End      Blocks   Id  System 
/dev/sdb1               1           9       72261    e  FAT16 LBA
Warning: Partition 1 does not end on cylinder boundary.                   
/dev/sdb2               9         468     3686917   83  Linux
Warning: Partition 2 does not end on cylinder boundary. 
```

CGarces, 2016-10-20:
My boot folder is empty?

```
root@amlogic-s905x:/boot# ls -l -all
total 12
drwxr-xr-x  2 root root 4096 Oct 20 00:26 .
drwxr-xr-x 22 root root 4096 Oct 19 23:16 ..
-rw-r--r--  1 root root   55 Oct 20 00:26 .verbose
```
not looks like a standard Debian system, right?
Some other info:

```
root@amlogic-s905x:/# fdisk -l
Device     Boot  Start     End Sectors  Size Id Type
/dev/sda1         2048  133119  131072   64M  e W95 FAT16 (LBA)
/dev/sda2       133120 7503872 7370753  3.5G 83 Linux
```

```
root@amlogic-s905x:/# mount
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,nosuid,nodev,noexec,relatime)
udev on /dev type devtmpfs (rw,relatime,size=10240k,nr_inodes=70695,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620)
tmpfs on /run type tmpfs (rw,nosuid,relatime,size=158516k,mode=755)
/dev/sda2 on / type ext4 (rw,noatime,nodiratime,errors=remount-ro,commit=600)
securityfs on /sys/kernel/security type securityfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,release_agent=/lib/systemd/systemd-cgroups-agent,name=systemd)
cgroup on /sys/fs/cgroup/debug type cgroup (rw,nosuid,nodev,noexec,relatime,debug)
cgroup on /sys/fs/cgroup/cpu,cpuacct type cgroup (rw,nosuid,nodev,noexec,relatime,cpu,cpuacct)
cgroup on /sys/fs/cgroup/freezer type cgroup (rw,nosuid,nodev,noexec,relatime,freezer)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=23,pgrp=1,timeout=300,minproto=5,maxproto=5,direct)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
mqueue on /dev/mqueue type mqueue (rw,relatime)
hugetlbfs on /dev/hugepages type hugetlbfs (rw,relatime)
configfs on /sys/kernel/config type configfs (rw,relatime)
fusectl on /sys/fs/fuse/connections type fusectl (rw,relatime)
tmpfs on /tmp type tmpfs (rw,nosuid,relatime)
tmpfs on /run/user/1000 type tmpfs (rw,nosuid,nodev,relatime,size=79260k,mode=700,uid=1000,gid=1000)
```

The boot configuration use is at 
https://github.com/150balbes/lib/blob/master/config/bootscripts/boot-amlogics905x.cmd
But I don't know how manage it.

Fourdee, 2016-10-22:
@CGarces 

> The /dev/sda1 not is mounted.

I'd like to check the mount paths on that image. On the android box with ARMbian image installed, can you please paste results of:
```
cat /etc/fstab
blkid
df -h
```

Fourdee, 2016-10-22:
@CGarces 

Image has 2 paritions, 1st is FAT, so should be fine.

Yep, so ```/boot``` is not mounted on the image ```/etc/fstab```:
```
/dev/root / ext4 defaults,noatime,nodiratime,commit=600,errors=remount-ro 0 1
tmpfs /tmp tmpfs defaults,nosuid 0 0
```

So you'll need to edit ```/etc/fstab``` to add the ```/boot``` entry, but we'll need to see results of the following on the installed android system with ARMbian image:
```
blkid
```

Fourdee, 2016-11-29:
No response, marking as closed.

closed 2016-11-29 by Fourdee