MichaIng/DietPi #557: Integrating ioBroker.net in Homeautomation

maxtox, 2016-10-15:
Hi,

could you please integrate ioBroker in Homeautomation...

it is quiet simple script.

depends on system we have to install node.js, pyton and buil-essentials (which is already a part of dietpi software i think):

```
Install Node.js
sudo apt-get install -y build-essential python
cd /tmp
wget https://nodejs.org/dist/v4.5.0/node-v4.5.0-linux-arm64.tar.xz
cd /usr/local
sudo tar --strip-components=1 -xvf /tmp/node-v4.5.0-linux-arm64.tar.xz
```
reference here...

http://www.iobroker.net/?page_id=5347&lang=en

After installation of node.js the command “node -v” should show the node.js version. if not, create alias to bin file:
`sudo ln -s /usr/local/bin/nodejs /usr/bin/node`

Installation of ioBroker:
```
install ioBroker
sudo mkdir /opt/iobroker
sudo chmod 777 /opt/iobroker
cd /opt/iobroker
sudo npm install iobroker --unsafe-perm
ioBroker über IP im Webbrowser aufrufen: http://IP-Address:8081
```

that's all...

Would be realy nice to see ioBroker (more then 2.000 Users) in DietPi

Fourdee, 2017-12-30:
General Cleanup:
Marking all software requests as closed. Once interest peaks, we'll reopen once work starts.

closed 2017-12-30 by Fourdee

Soulguard, 2018-03-08:
Yes, would be very nice if you integrate iobroker to Homeautomation!
Interest peaks ;)

btw. thank you for your fantastic dietpi image!