MichaIng/DietPi #558: XU4 fails to power on

Fourdee, 2016-10-15:
As reported by some XU4 users: http://dietpi.com/phpbb/viewtopic.php?f=11&t=707&p=3337#p3337

Unable to replicate on my system.

Fourdee, 2016-10-16:
Re-created the DietPi image from scratch. On the off chance we had some corruptions on the base image due to FStrim: http://dietpi.com/downloads/testing/DietPi_OdroidXU4-armv7-(Jessie).7z

biccy, 2016-10-16:
Yep, with the newly downloaded image, I can now use the poweroff command or just pull the power and the XU4 boots up fine afterwards every time. Previously it NEVER booted again once I ran poweroff or pulled the power (it only rebooted with the reboot command).

Fourdee, 2016-10-16:
@biccy 
Excellent, thanks for testing and letting us know.

I'll move this and replace the current image download.

nphil, 2016-10-17:
Good to hear, I would assume this requires a clean flash of the OS image again?

Fourdee, 2016-10-17:
@nphil 
Will need a clean install, unfortunately. Most likely the partitions and/or table on our previous image is corrupt which is causing the no boot issue.

justin0a0, 2016-10-18:
I experienced this issue this morning using the old (127) image when I hit the power button.
Used the new image and no issues with it so far issued poweroff and used the button reboots went fine.

Fourdee, 2016-10-19:
@nphil @justin0a0 @biccy 
Thanks guys for your help testing! 👍 
We can now mark this one as resolved.

closed 2016-10-19 by Fourdee