MichaIng/DietPi #560: Dietpi-Software : Wallabag ?

Bridouz, 2016-10-16:
Hi,

I think wallabag could ne a good software to have on Dietpi.

Wallabag is a self-hostable software for saving web pages ans articles to read them later (FOSS alternative to Pocket)

http://doc.wallabag.org/en/latest/user/installation.html#requirements

It's a self hostable application for saving web pages.

Bridouz, 2016-10-18:
That's how I managed to install Wallabag and run it with Lighttp : 

`cd /var/www/
wget http://wllbg.org/latest-v2-package && tar xvf latest-v2-package
mv latest-v2-package wallabag 
chown -R www-data:www-data /var/www/wallabag/var
chown -R www-data:www-data /var/www/wallabag/bin
chown -R www-data:www-data /var/www/wallabag/app/config
chown -R www-data:www-data /var/www/wallabag/vendor
chown -R www-data:www-data /var/www/wallabag/data/
`
mod_fastcgi has to be enabled in lighttpd.conf

adress to access wallabag :+1: 
`serverip/wallabag/web/app.php`

I have to learn how to redirect that adress to just servername/wallabag

mdemoss, 2017-12-17:
I think this is broken in wallabag 2.3.0. Looks like they're considering dropping sqlite support.

mdemoss, 2017-12-17:
I grabbed wallabag 2.2.3 and it works. ( Following instructions at https://www.wallabag.org/en/news/wallabag-v2 ) As far as I can tell, there aren't "shared hosting" instructions for 2.3.0.

Fourdee, 2017-12-30:
General Cleanup:
Marking all software requests as closed. Once interest peaks, we'll reopen once work starts.

closed 2017-12-30 by Fourdee

pedrom34, 2018-06-07:
Please, consider Wallabag!

Balumis, 2022-06-23:
Do we have wallabeg on DietPi?

Joulinar, 2022-06-23:
No

Joulinar, 2022-06-24:
If you are familiar with Docker try this one https://doc.wallabag.org/en/admin/installation/installation.html#installation-with-docker
This would keep your Wallabag instance separated from any DietPi software

Caporeira, 2022-06-24:
I have docker and portainer on my DietPi.
But how can i import this wallabage image on portainer ?

Joulinar, 2022-06-24:
There is no need to use Portainer. Simply have a look to their Docker Hub package. There are different descriptions depending on the database of your choice https://hub.docker.com/r/wallabag/wallabag/