MichaIng/DietPi #563: DietPi-Software | WiringOP

Fourdee, 2016-10-17:
> https://github.com/zhaolei/WiringOP
http://dietpi.com/phpbb/viewtopic.php?f=12&t=249#p3370

Fourdee, 2016-10-24:
This is alot more involved than originally thought. As mentioned by Pete on forums, GPIO assignments are likely to be different for various boards. So, the WiringOP source code will need to be modified and tested. 

I don't really have the time (and minimal GPIO experience) to take this on at the moment, so moving to v135 for now.

Fourdee, 2017-12-30:
General Cleanup:
Marking all software requests as closed. Once interest peaks, we'll reopen once work starts.

closed 2017-12-30 by Fourdee

Fmert24, 2023-09-23:
ı find github page that instaled wiringop successfully for orangepi zero lts
https://github.com/zhaolei/WiringOP/tree/h3

MichaIng, 2023-09-23:
Based on our current SBC support, a RK3588-compatible version would be relevant. But that repo/branch got its last commit in 2017.

Nowadays WiringX needs to be seen as deprecated in general, as we have generic libgpio/[gpiod](https://packages.debian.org/gpiod) now. It is not yet perfect, not yet supported by all SBCs/used kernels, behaves a little different here and there, but it is the future to go for, getting rid of all those moreless badly maintained WiringPi vendor forks.

Fmert24, 2023-09-24:
> Based on our current SBC support, a RK3588-compatible version would be relevant. But that repo/branch got its last commit in 2017.
>
> Nowadays WiringX needs to be seen as deprecated in general, as we [...]

thank you for updating me on gpio libary condition