MichaIng/DietPi #567: OPi one | WiFi not functional after dietpi-update

Fourdee, 2016-10-18:
> http://dietpi.com/phpbb/viewtopic.php?f=12&t=764&p=3409

Fourdee, 2016-10-18:
Fresh image with dietpi.txt wifi entries and country codes:
- WiFi works on first run
- dietpi-update 
- reboots
- WiFi is down.
- unplug wifi and plug back in = 🈯️ 
- reg set = 🈴 Always fails, no errors in dmesg

Fourdee, 2016-10-18:
```
[    9.732754] ADDRCONF(NETDEV_UP): wlan0: link is not ready
[    9.917065] RTL871X: set ssid [FuzonWifi] fw_state=0x00000008
[   11.931713] RTL871X: start auth
[   17.370079] RTL871X: SetHwReg8188E:(HW_VAR_CHECK_TXBUF)NOT empty(100) in 2000 ms
[   17.370120] RTL871X: SetHwReg8188E:(HW_VAR_CHECK_TXBUF)0x200=0x009f0900, 0x204=0x00990900 (0x009f0900, 0x00990900)
[   18.682179] RTL871X: rtw_set_802_11_connect(wlan0)  fw_state=0x00000008
[   18.792844] RTL871X: start auth
[   18.885135] RTL871X: auth success, start assoc
[   18.919524] RTL871X: assoc success
[   18.920174] ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   18.925247] RsvdPageNum: 8
[   19.505624] RTL871X: send eapol packet
[   19.570558] RTL871X: send eapol packet
[   19.571178] RTL871X: set pairwise key camid:4, addr:e8:de:27:4a:d0:66, kid:0, type:AES
[   19.574770] RTL871X: set group key camid:5, addr:e8:de:27:4a:d0:66, kid:1, type:AES
[   29.580028] wlan0: no IPv6 routers present

#NO IP
#physical reconnect

[   93.345963] ehci_irq: highspeed device disconnect
[   93.346137] usb 1-1: USB disconnect, device number 2
[   93.390383] RTL871X: rtw_cmd_thread: DriverStopped(0) SurpriseRemoved(1) break at line 478
[   93.391553] cfg80211: Calling CRDA for country: GB
[   93.460417] RTL871X: rtw_ndev_uninit(wlan0)
[   93.461932] cfg80211: Regulatory domain changed to country: GB
[   93.461955] cfg80211:  DFS Master region ETSI
[   93.461969] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp)
[   93.461987] cfg80211:   (2402000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm)
[   93.462003] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz), (N/A, 2000 mBm)
[   93.462019] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz), (N/A, 2000 mBm)
[   93.462037] cfg80211:   (5490000 KHz - 5710000 KHz @ 160000 KHz), (N/A, 2700 mBm)
[   93.462055] cfg80211:   (57000000 KHz - 66000000 KHz @ 2160000 KHz), (N/A, 4000 mBm)
[   93.462168] cfg80211: Calling CRDA to update world regulatory domain
[   93.482464] cfg80211: World regulatory domain updated:
[   93.482484] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp)
[   93.482497] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm)
[   93.482509] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm)
[   93.482521] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm)
[   93.482533] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz), (N/A, 2000 mBm)
[   93.482545] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz), (N/A, 2000 mBm)
[   93.482562] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm)
[   93.482574] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm)
[   93.482590] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm)
[   93.482715] cfg80211: Calling CRDA for country: GB
[   93.497934] cfg80211: Regulatory domain changed to country: GB
[   93.497954] cfg80211:  DFS Master region ETSI
[   93.497963] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp)
[   93.497976] cfg80211:   (2402000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm)
[   93.497988] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz), (N/A, 2000 mBm)
[   93.498000] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz), (N/A, 2000 mBm)
[   93.498012] cfg80211:   (5490000 KHz - 5710000 KHz @ 160000 KHz), (N/A, 2700 mBm)
[   93.498024] cfg80211:   (57000000 KHz - 66000000 KHz @ 2160000 KHz), (N/A, 4000 mBm)
[   93.680322] RTL871X: rtw_ndev_uninit(wlan1)
[   93.810330] RTL871X: rtw_dev_unload: driver not in IPS
[   95.041087] ehci_irq: highspeed device connect
[   95.320054] usb 1-1: new high-speed USB device number 3 using sunxi-ehci
[   95.477475] bFWReady == _FALSE call reset 8051...
[   95.504837] RTL871X: MAC Address from Efuse is 00:13:ef:70:12:bd
[   95.507228] RTL871X: rtw_ndev_init(wlan0)
[   95.508414] RTL871X: rtw_ndev_init(wlan1)
[   96.070904] ==> rtl8188e_iol_efuse_patch
[   96.365980] ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   96.378979] RTL871X: set ssid [FuzonWifi] fw_state=0x00000008
[   98.259809] RTL871X: start auth
[  103.630106] RTL871X: SetHwReg8188E:(HW_VAR_CHECK_TXBUF)NOT empty(100) in 2000 ms
[  103.630142] RTL871X: SetHwReg8188E:(HW_VAR_CHECK_TXBUF)0x200=0x009f0900, 0x204=0x00990900 (0x009f0900, 0x00990900)
[  104.942164] RTL871X: rtw_set_802_11_connect(wlan0)  fw_state=0x00000008
[  105.018575] RTL871X: start auth
[  105.221015] RTL871X: auth success, start assoc
[  105.335650] RTL871X: assoc success
[  105.336277] ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  105.341626] RsvdPageNum: 8
[  105.554374] RTL871X: send eapol packet
[  105.850232] RTL871X: send eapol packet
[  105.850698] RTL871X: set pairwise key camid:4, addr:e8:de:27:4a:d0:66, kid:0, type:AES
[  105.854270] RTL871X: set group key camid:5, addr:e8:de:27:4a:d0:66, kid:1, type:AES
[  115.900044] wlan0: no IPv6 routers present
```

Fourdee, 2016-10-18:
Will use latest ARMbian build and convert to DietPi then re-test. Clearly some differences in active modules. Possibly updated since we built the 1st image.

Latest ARMbian
```
root@orangepione:~# lsmod
Module                  Size  Used by
sunxi_cir               1609  0
8188eu                911607  0
bmp085                  3499  0
pcf8591                 3321  0```

DP:
```
root@DietPi:~# lsmod
Module                  Size  Used by
8723bs               1588835  0
brcmfmac               92323  0
brcmutil                2942  1 brcmfmac
8188eu                908230  0
```

Fourdee, 2016-10-19:
Ok, latest ARMbian build converted to DietPi image.

WiFi requires the user to "login" before the wifi becomes available 😆 
```
[   32.840056] wlan0: no IPv6 routers present
[  494.024305] RTL871X: send eapol packet
[  494.026261] RTL871X: set group key camid:6, addr:e8:de:27:4a:d0:66, kid:2, type:AES
```

Fourdee, 2016-10-29:
Notes:
Looks like my OPi one / PSU is failing, 3 corrupt SD cards after a few reboots with both ARMbian official and DietPi images.

EDIT: Plugging in a Microsoft USB keyboard results in kernel panic with MMC related errors. Weird.

whhsn, 2017-01-26:
Why this is marked "Not a DietPi issue"?
IS there a  way to resolve this?

Fourdee, 2017-01-26:
@whhsn 
We use ARMbian to build the image base, kernel, uboot, modules etc. So that part is essentially out of our hands. We rely on ARMbian to apply patches/fixes and hopefully resolve issues related to that.

I'll do another image build and re-test, see if its resolved.

Fourdee, 2017-01-27:
Image updated: http://dietpi.com/downloads/images/

Fourdee, 2017-01-27:
Issue still exists, same as M1: https://github.com/Fourdee/DietPi/issues/729#issuecomment-275686055

whhsn, 2017-01-27:
Can we run a fix script after the update & reboot just for the affected targets (M1 & OPiOne for now)?

Fourdee, 2017-01-28:
@whhsn 

Not sure yet, i'll run some tests, report to ARMbian and see if a workaround is even possible.

Fourdee, 2017-01-28:
M1:
```
root@DietPi:~# ifup wlan0
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan0 ; Operation not supported.
Internet Systems Consortium DHCP Client 4.3.1
Copyright 2004-2014 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/wlan0/74:da:38:26:9d:65
Sending on   LPF/wlan0/74:da:38:26:9d:65
Sending on   Socket/fallback
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 8
[   77.585881] ------------[ cut here ]------------
[   77.590035] Kernel BUG at c0044d34 [verbose debug info unavailable]
[   77.590035] sunxi oops: enable sdcard JTAG interface
[   77.590035] sunxi oops: cpu frequency: 240 MHz
[   77.590035] sunxi oops: ddr frequency: 624 MHz
[   77.590035] sunxi oops: gpu frequency: 105 MHz
[   77.590035] sunxi oops: cpu temperature: 32
[   77.590035] Internal error: Oops - BUG: 0 [#1] PREEMPT SMP ARM
[   77.590035] Modules linked in: rtl8192cu rtlwifi rtl8192c_common mac80211 bmp085 pcf8591 [last unloaded: scsi_wait_scan]
[   77.590035] CPU: 1    Tainted: G           O  (3.4.113-sun8i #4)
[   77.590035] PC is at __queue_work+0x23c/0x3d8
[   77.590035] LR is at __queue_work+0x204/0x3d8
[   77.590035] pc : [<c0044d34>]    lr : [<c0044cfc>]    psr: 00000193
[   77.590035] sp : d52e5d80  ip : d52e5d80  fp : d52e5db4
[   77.590035] r10: 0000000a  r9 : 40000113  r8 : 00000188
[   77.590035] r7 : d581d0c0  r6 : d4815438  r5 : c1006a00  r4 : c10015c0
[   77.590035] r3 : d481543c  r2 : 00000000  r1 : d581d0c0  r0 : c1006a00
[   77.590035] Flags: nzcv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
[   77.590035] Control: 10c5387d  Table: 54e9006a  DAC: 00000015
[   77.590035]
[   77.590035] PC: 0xc0044cb4:
[   77.590035] 4cb4  e5933004 e1570003 01a04005 0a00000a e1a01009 e1a00005 eb196fd8 e1a00004
[   77.590035] 4cd4  eb19719a e1a09000 ea000003 e59f01e0 eb197196 e59f41d8 e1a09000 e1a01007
[   77.590035] 4cf4  e5940010 ebfff946 e1a05000 e1a00000 ea000006 e5983004 e2433001 e5883004
[   77.590035] 4d14  e5983000 e3130002 0a000000 eb196d51 e5962004 e2863004 e1520003 0a000000
[   77.590035] 4d34  e7f001f2 e5958008 e0852108 e1a08208 e5923010 e2833001 e5823010 e595204c
[   77.590035] 4d54  e5953050 e1520003 a3888002 a285a054 aa000024 e1a00000 ea000006 e59a3004
[   77.590035] 4d74  e2433001 e58a3004 e59a3000 e3130002 0a000000 eb196d37 e595304c e2833001
[   77.590035] 4d94  e585304c e5953004 e5933000 e3130010 e2843008 15947008 01a0a003 0a000011
[   77.590035]
[   77.590035] LR: 0xc0044c7c:
[   77.590035] 4c7c  e1a00006 ebfff696 e1540000 13500000 e1a05000 0a00000e eb1971aa e1a01006
[   77.590035] 4c9c  e1a09000 e1a00005 ebffff65 e3500000 0a000004 e5903010 e5933004 e1570003
[   77.590035] 4cbc  01a04005 0a00000a e1a01009 e1a00005 eb196fd8 e1a00004 eb19719a e1a09000
[   77.590035] 4cdc  ea000003 e59f01e0 eb197196 e59f41d8 e1a09000 e1a01007 e5940010 ebfff946
[   77.590035] 4cfc  e1a05000 e1a00000 ea000006 e5983004 e2433001 e5883004 e5983000 e3130002
[   77.590035] 4d1c  0a000000 eb196d51 e5962004 e2863004 e1520003 0a000000 e7f001f2 e5958008
[   77.590035] 4d3c  e0852108 e1a08208 e5923010 e2833001 e5823010 e595204c e5953050 e1520003
[   77.590035] 4d5c  a3888002 a285a054 aa000024 e1a00000 ea000006 e59a3004 e2433001 e58a3004
[   77.590035]
[   77.590035] SP: 0xd52e5d00:
[   77.590035] 5d00  c001a0d0 c00fcc64 c0adfa54 c00522bc 00000000 d4817328 c0044d38 00000193
[   77.590035] 5d20  c000de94 00000000 d52e5db4 d52e5d38 c000dbec c000820c c1006a00 d581d0c0
[   77.590035] 5d40  00000000 d481543c c10015c0 c1006a00 d4815438 d581d0c0 00000188 40000113
[   77.590035] 5d60  0000000a d52e5db4 d52e5d80 d52e5d80 c0044cfc c0044d34 00000193 ffffffff
[   77.590035] 5d80  d52e5da4 00000001 c0484e84 d4815438 00000001 d581d0c0 00000008 00000188
[   77.590035] 5da0  bf096c40 0000000a d52e5dd4 d52e5db8 c0044f14 c0044b04 d52e4000 00000001
[   77.590035] 5dc0  d4dcab40 00000008 d52e5dec d52e5dd8 c0044f60 c0044edc d4816240 00000001
[   77.590035] 5de0  d52e5dfc d52e5df0 c0044fac c0044f30 d52e5e1c d52e5e00 bf09207c c0044f94
[   77.590035]
[   77.590035] IP: 0xd52e5d00:
[   77.590035] 5d00  c001a0d0 c00fcc64 c0adfa54 c00522bc 00000000 d4817328 c0044d38 00000193
[   77.590035] 5d20  c000de94 00000000 d52e5db4 d52e5d38 c000dbec c000820c c1006a00 d581d0c0
[   77.590035] 5d40  00000000 d481543c c10015c0 c1006a00 d4815438 d581d0c0 00000188 40000113
[   77.590035] 5d60  0000000a d52e5db4 d52e5d80 d52e5d80 c0044cfc c0044d34 00000193 ffffffff
[   77.590035] 5d80  d52e5da4 00000001 c0484e84 d4815438 00000001 d581d0c0 00000008 00000188
[... 162 lines cut ...]
[   83.122852] [<c000db40>] (__irq_svc+0x40/0x74) from [<c000f2dc>] (default_idle+0x34/0x3c)
[   83.122852] [<c000f2dc>] (default_idle+0x34/0x3c) from [<c000f694>] (cpu_idle+0xa8/0xf4)
[   83.122852] [<c000f694>] (cpu_idle+0xa8/0xf4) from [<c068f3b0>] (rest_init+0x9c/0xb8)
[   83.122852] [<c068f3b0>] (rest_init+0x9c/0xb8) from [<c096d8ac>] (start_kernel+0x2f8/0x358)
[   83.122854] CPU3: stopping
[   83.122854] [<c0016a20>] (unwind_backtrace+0x0/0xe8) from [<c0697880>] (dump_stack+0x20/0x24)
[   83.122854] [<c0697880>] (dump_stack+0x20/0x24) from [<c0014170>] (handle_IPI+0x114/0x298)
[   83.122854] [<c0014170>] (handle_IPI+0x114/0x298) from [<c0008554>] (gic_handle_irq+0x64/0x6c)
[   83.122854] [<c0008554>] (gic_handle_irq+0x64/0x6c) from [<c000db40>] (__irq_svc+0x40/0x74)
[   83.122854] Exception stack(0xd5901f68 to 0xd5901fb0)
[... cut ...]
[   83.122854] [<c000f694>] (cpu_idle+0xa8/0xf4) from [<c069477c>] (secondary_start_kernel+0x108/0x12c)
[   83.122854] [<c069477c>] (secondary_start_kernel+0x108/0x12c) from [<40693d94>] (0x40693d94)
[   83.088120] [hotplug]: cpu(1) try to kill cpu(0)
[   83.088120] [hotplug]: try to kill cpu:0 failed!
[   83.088120] [hotplug]: cpu(1) try to kill cpu(2)
[   83.088120] [hotplug]: try to kill cpu:2 failed!
[   83.088120] [hotplug]: cpu(1) try to kill cpu(3)
[   83.088120] [hotplug]: try to kill cpu:3 failed!
[   83.088120] Rebooting in 10 seconds..
```

Fourdee, 2017-02-04:
No fix.
Not wasting any more time trying to deal with ~~ARM~~ThomasKaiserBian.

We'll wait for ~~ARM~~ThomasKaiserBian to reach 4.10 (functional sound).

closed 2017-02-04 by Fourdee