MichaIng/DietPi #568: PiHole - password protect admin page

sootstone, 2016-10-18:
@pi-hole have a solution for password protecting the web admin page of pi-hole on a lightttpd [http://jacobsalmela.com/password-protect-a-lighttpd-web-server-on-a-raspberry-pi-using-mod-auth/](http://jacobsalmela.com/password-protect-a-lighttpd-web-server-on-a-raspberry-pi-using-mod-auth/)

Might be impossible, but would there be a way to add that kind of functionality into DietPi for other webservers (e.g. Apache2, Nginx)?

sootstone, 2016-10-25:
Seems like the pi-hole people are working on a way to password protect [https://discourse.pi-hole.net/t/password-protect-the-dashboard/223](https://discourse.pi-hole.net/t/password-protect-the-dashboard/223)

Fourdee, 2016-11-29:
> Seems like the pi-hole people are working on a way to password protect https://discourse.pi-hole.net/t/password-protect-the-dashboard/223

Marking as closed.

closed 2016-11-29 by Fourdee