MichaIng/DietPi #569: Use URL shortener to map dynamically ling updated

Zone22, 2016-10-18:


Fourdee, 2016-10-20:
@Zone22 

Great idea, makes sense. But, there is possible issues with updating the download links through master branch:
- When some software upgrades, the installation procedure might need to be changed, possible outcome of failed installation.
- With software where we have made a pre-defined SQL DB or config (eg: ampache), we would also need to update those, else, failed install.

So, that is why we have the testing branch, even when we update the download link eg: from 8.0 to 8.1, it has to be tested, before we roll it down to our users.

I try to keep our release cycles between 2-3 weeks. So, although there is a wait for what appears to be a simple URL link change, that gives us time to ensure the installation works for our users, before we roll it out.

Zone22, 2016-10-20:
Makes a lot of sense.

closed 2016-10-20 by Zone22