MichaIng/DietPi #570: VPN

peakyc, 2016-10-19:
Hi Guys,

Firstly well done on the project - it is easily my favourite distro for Pi atm.

Whilst I can understand the emphasis for the project seems to be ease of setup I think the VPN installation could be slighty improved with little or no increase in difficulty for the user.

Before testing DietPi is was using a script from http://www.pivpn.io/ and https://github.com/pivpn to install openvpn.

The benefits over the DietPi installation for me are (these can all be skipped by pressing enter for default):

(1) VPN port 
(2) Certificate details (I am concerned about the default values in dietpi)
(3) Key encryption strength (For users who need more security)
(4) Client DNS server
(5) Password protected certificates

Further to the pre installation options they also provide scripts that seem to fit well with dietpi principals e.g. easy uninstallation.

Also their documentation for adding/removing and revoking are easy enough for even basic users.

I feel these methods are easier and more secure than those currently provided in dietpi e.g default certificate and editing in notepad.

Just as a side not when i installed PiVPN only port 1194 was needed for a sucessful connection.

I don't know if you could either implement some changes to your current installation scripts or perhaps just implement theirs instead.

Anyway thanks again for all your hard work.

Hope this helps

peakyc, 2016-10-19:
NB: If you plan on testing this please install lsb-release first to fix the problem with DietPi not being detected properly as debian jessie.

peakyc, 2016-10-20:
Just a quick update on this. 

The only issue I have had so far when testing this on DietPi depended on when i installed PiVPN - after or before installing various other software using dietpi-software.

If you have a DNS issue after installing where by you can connect to the VPN but not resolve domain (google.com or anything else) try this:

sudo iptables -t nat -A POSTROUTING -s 10.8.0.0/24 -o eth0 -j MASQUERADE

sudo iptables-save

sudo dpkg-reconfigure iptables-persistent

This fix was from cabsandy here https://github.com/pivpn/pivpn/issues/110 and referenced here http://unix.stackexchange.com/questions/125833/why-isnt-the-iptables-persistent-service-saving-my-changes

redfast00, 2016-10-23:
@peakyc this should be fixed now. Let me know if you still have problems.

Fourdee, 2016-10-23:
@peakyc I'll take a look 👍 
If all goes well, we could add PiVPN as a new installation option, along side the existing OpenVPN one, to give the users a choice.

Fourdee, 2016-10-23:
@peakyc @redfast00 
PiVPN is excellent 👍 

Few issues:
- Choose user only displays "available", instead of root (https://github.com/pivpn/pivpn/blob/d900efe49bf7f9b06d85129393571e08c7e2dce1/auto_install/install.sh#L134-L166):
[image]
 - This also results in ```pivpnUser``` being null, results in attempt to cp ```/home//ovpns``` (https://github.com/pivpn/pivpn/blob/d900efe49bf7f9b06d85129393571e08c7e2dce1/auto_install/install.sh#L915-L916)
```
cp: cannot stat ‘/tmp/pivpnUSR’: No such file or directory
mkdir: cannot create directory ‘/home//ovpns’: File exists
```

Fourdee, 2016-10-23:
Ok done, PiVPN is awesome.

@peakyc  

Any chance you would be willing to test the installation on our testing branch?
Instructions here: https://github.com/Fourdee/DietPi/blob/master/TESTING-BRANCH.md#steps-to-use-the-dietpi-testing-branch

peakyc, 2016-10-23:
Hi @redfast00 and @Fourdee .

Thanks for taking a look at this.

Yeah sorry I forgot to say that prior to install I created a new user.

I would love to test it for you. My Pi is in use ATM so can only test in a VM until later. As soon as I can test on my pi (rpi 3) I will get back to you.

Obviously full credit to the PiVPN guys for their work on this (I thought it was awesome too). 

Glad I could help introduce you to the project.

Fourdee, 2016-10-26:
I'll mark this as closed, installation is successful.

@peakyc 
Please reopen if required.

closed 2016-10-26 by Fourdee

redfast00, 2016-10-27:
@Fourdee You no longer have to create a non-root user. If there isn't one, PiVPN now prompts you to create one. (See https://github.com/pivpn/pivpn/pull/136)

0-kaladin, 2016-10-28:
Awesome work guys!  Glad to see this spreading!