MichaIng/DietPi #576: Deactivate internal WiFi an dactivate external WiFi

mok-liee, 2016-10-24:
Hi,

i have an edimax ew-7811uac 5ghz wlan stick. I tried to deactivate the internal wifi. but i cant activate the external stick for using this for the hotspot feature.

is it possible to make a "choose device" point in the wifi menue?

another thing... if the hotspot feature is activated an configured, there is no option to deactivate wifi.

hoe u can help me again.

greetings mok

Fourdee, 2016-10-24:
@mok-liee 

Hi Mok,

Currently, DietPi assumes the user would like to use the internal WiFi features of their RPi 3, essentially, its hard coded for that. So i'll need to look into this and create a feature which allows for specifying another device, if detected by system.

> another thing... if the hotspot feature is activated an configured, there is no option to deactivate wifi.

Thanks 👍, i'll take a look.

Fourdee, 2016-11-05:
Option in dietpi.txt to disable RPi 3 onboard wifi using blacklist.

Fourdee, 2016-11-06:
Onboard WiFi can now be disabled/enabled from ```dietpi-config```.

NB:
- Requires a reboot to take effect

[image]

Fourdee, 2016-11-06:
Completed.

closed 2016-11-06 by Fourdee