MichaIng/DietPi #579: DietPi Software | Block IPv6 Ads with PiHole

sootstone, 2016-10-25:
Can the DietPi version of Pi-Hole block IPv6 adverts? It seem like all AAAA records are getting through according to the Query Log on the Pi-Hole web interface.

sootstone, 2016-10-25:
Checked against the standard Pi-Hole install and AAAA ads were blocked.

Fourdee, 2016-10-26:
@sootstone 

> Checked against the standard (Jessie Lite) Pi-Hole install and AAAA ads were blocked,

Thanks 👍 I'll take a look, see whats up.

Fourdee, 2016-10-26:
@sootstone 

gravity.sh uses:
```
piholeIPfile=/etc/pihole/piholeIP
piholeIPv6file=/etc/pihole/.useIPv6
```

But the codes completely changed in the next PiHole update.

Looks like the official installer is now required, it generates files and configs (eg: current IP address, IPv6 enabled etc) during its installation that are used by ```gravity.sh```, rather than obtained at runtime.

This is a good time as any to look at dropping our installation and using the official PiHole installer.
Alot has changed since the start, PiHole's size and code has scaled up vastly than it was originally. Many of the DietPi features (eg: detect existing nginx/apache webserver, set static IP) of the fork are now available in the official installation.

@sootstone 

Solution, remove the DietPi installed version and use their official installer:
```
dietpi-software uninstall 93
curl -L install.pi-hole.net | bash
```

closed 2016-10-26 by Fourdee