MichaIng/DietPi #583: DietPi-Software | PiHole to use official installer

Fourdee, 2016-10-26:
As title, reasons: https://github.com/Fourdee/DietPi/issues/579#issuecomment-256330893

Fourdee, 2016-10-26:
There is still minor changes we need to apply post install:
- PiHole overwrites the default ```lighttpd.conf``` (https://github.com/pi-hole/pi-hole/blob/master/advanced/lighttpd.conf.debian), limiting the webserver to directory ```/var/www/html```. With DietPi, we use ```/var/www```.
- PiHole uses the 404 error handler ```server.error-handler-404	= "pihole/index.html"```, to support Nginx/Lighttpd/Apache, we'll use the website default page ```/var/www/index.html```

closed 2016-10-26 by Fourdee

Zone22, 2016-11-03:
The the message warning that this version of diet pi fork is still present in v136 when it's not needed with the official installer.