MichaIng/DietPi #588: Diet Pi Failing to Update

rubinkazan, 2016-10-31:
Current Version: 134
Server Version: 135

When running ` dietpi-update`
The following occurs:
` DietPi-Update
─────────────────────────────────────────────────────
 Mode: Checking for DietPi updates
 Please wait...

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100     5  100     5    0     0      4      0  0:00:01  0:00:01 --:--:--     4

 [Ok] Update available
 [Info] Current Version : 134
 [Info] Server Version  : 135

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: stop
 Please wait...

 [Ok] cron
 [Ok] proftpd
 [Ok] apache2
 [Ok] lighttpd
 [Ok] mysql
 [Ok] dnsmasq
 [Ok] openvpn

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   119    0   119    0     0      7      0 --:--:--  0:00:16 --:--:--    29
  0     0    0     0    0     0      0      0 --:--:--  0:00:36 --:--:--     0curl: (6) Could not resolve host: codeload.github.com
 [Failed] Download failed, unable to run update. Please try running dietpi-update again.
`

Fourdee, 2016-10-31:
@rubinkazan 

Hi, thanks for letting us know.

I'd like to check your system setup, could you please try sending a bug report: http://dietpi.com/phpbb/viewtopic.php?f=8&t=499#p2184

Thanks.

rubinkazan, 2016-10-31:
@Fourdee 
My reference code: c76b1794-eec9-4743-8fa4-dac5308124ee-0 
However I doubt it would be of any use as I received this error when following instructions from that URL.

[Info] Uploading bug report, please wait...
--19:00:34-- `bugreport-c76b1794-eec9-4743-8fa4-dac5308124ee-0.zip'
    => ftp://dietpi-survey:xxxxx@185.101.92.145:21/bugreport-c76b1794-eec9-4743-8fa4-dac5308124ee-0.zip
Connecting to 185.101.92.145:21... connected! Error: recv() timed out. No data received
Receive-Warning: read() timed out. Read '' so far.

Logging in as dietpi-survey ... Error: recv() timed out. No data received
Receive-Warning: read() timed out. Read '' so far.
Waiting 2 seconds... Error: recv() timed out. No data received
Receive-Warning: read() timed out. Read '' so far.
Connecting to 185.101.92.145:21... connected! Error: recv() timed out. No data received
Receive-Warning: read() timed out. Read '' so far.

Logging in as dietpi-survey ... Error: recv() timed out. No data received
Receive-Warning: read() timed out. Read '' so far.
Error: recv() timed out. No data received
Receive-Warning: read() timed out. Read '' so far.
FINISHED --19:00:43--
Transmission of 1 file failed.

Fourdee, 2016-10-31:
@rubinkazan 
Thanks, yes it looks like you have no route to github, so failed.

Can you paste results of the following please:
```
route
ifconfig -a
ping -c 1 github.com
```

rubinkazan, 2016-10-31:
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         192.168.2.1     0.0.0.0         UG    0      0        0 eth0
default         192.168.2.1     0.0.0.0         UG    202    0        0 eth0
192.168.2.0     *               255.255.255.0   U     202    0        0 eth0
root@PiyoteQueen:~# ifconfig -a
eth0      Link encap:Ethernet  HWaddr b8:27:eb:51:5f:37  
          inet addr:192.168.2.20  Bcast:192.168.2.255  Mask:255.255.255.0
          inet6 addr: fe80::65b5:49d0:1fb9:d795/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:7246 errors:0 dropped:0 overruns:0 frame:0
          TX packets:6268 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:745521 (728.0 KiB)  TX bytes:1084896 (1.0 MiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:98 errors:0 dropped:0 overruns:0 frame:0
          TX packets:98 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:8317 (8.1 KiB)  TX bytes:8317 (8.1 KiB)

root@PiyoteQueen:~# ping -c 1 github.com
PING github.com (192.30.253.112) 56(84) bytes of data.

--- github.com ping statistics ---
1 packets transmitted, 0 received, 100% packet loss, time 0ms

Fourdee, 2016-11-01:
@rubinkazan 

Weird, looks like you cant access anything outside your network.

Please try:
```
ping -c 1 google.com
```

And can you paste results of (remove any WIFI KEYS):
```
cat /etc/network/interfaces
```

rubinkazan, 2016-11-01:
Result from pinging google:
PING google.com MY IP 56(84) bytes of data.
64 bytes from MY IP : icmp_seq=1 ttl=58 time=35.0 ms

--- google.com ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 35.019/35.019/35.019/0.000 ms

Result of cat
#/etc/network/interfaces
#Please use DietPi-Config to modify network settings.

# Local
auto lo
iface lo inet loopback
iface eth0 inet dhcp

# Ethernet
allow-hotplug eth0
iface eth0 inet static
address 192.168.2.20
netmask 255.255.255.0
gateway 192.168.0.1
dns-nameservers 8.8.8.8 8.8.4.4

# Wifi
allow-hotplug wlan0
iface wlan0 inet dhcp
address 192.168.0.100
netmask 255.255.255.0
gateway 192.168.0.1
wireless-essid Not_Set
wireless-key Not_Set
wireless-mode Managed
wireless-power off
wpa-ssid Not_Set
wpa-psk Not_Set
#dns-nameservers 8.8.8.8 8.8.4.4

Fourdee, 2016-11-02:
@rubinkazan 

Thanks.

> PING google.com MY IP 56(84) bytes of data.
64 bytes from MY IP : icmp_seq=1 ttl=58 time=35.0 ms

MY IP? Was that your external IP address?

Can you please try the following:
- ```dietpi-config``` > Networking Options: Adapters > Ethernet | Set ```DHCP``` and save changes
- ```dietpi-config``` > Networking Options: Adapters > WiFi | Set ```Disabled``` and save changes
- ```ping -c 1 google.com```
- ```ping -c 1 github.com```

rubinkazan, 2016-11-02:
Yup, MY IP was my external IP.
Having a slight problem at the moment when running dietpi-config, it just blinks my terminal window with a bunch of special characters (?*#) etc. This hasn't happened before btw

Fourdee, 2016-11-02:
@rubinkazan 

> Yup, MY IP was my external IP.

Shouldnt be that.

> Having a slight problem at the moment when running dietpi-config, it just blinks my terminal window with a bunch of special characters (?*#) etc. This hasn't happened before btw

Are you able to take a screenshot / image of the above?

You can reset the network settings file and revert back to eth0 using DHCP (default), with:
```
cp /boot/dietpi/conf/network_interfaces /etc/network/interfaces
/DietPi/dietpi/func/obtain_network_details
# - enable allow-hotplug eth0 after copying.
sed -i "/allow-hotplug eth/c\allow-hotplug eth$(sed -n 1p /DietPi/dietpi/.network)" /etc/network/interfaces
```

After the above, try a reboot re-run the pings from https://github.com/Fourdee/DietPi/issues/588#issuecomment-257847784

rubinkazan, 2016-11-03:
Restarting the Pi twice solved my issue of being unable to access dietpi-config
Followed instructions from #588 (comment).  I do not have an internet card so the disabling WiFi option was not available. I am already using DHCP as I am hosting a site. When running the config, it initially gave me a list of errors underneath WiFi, foolishly I copied them but overwrote them by copying something else by accident. However, upon restarting through dietpi-config (think it restarts the services (?)). 
`DietPi-Set_Hardware
─────────────────────────────────────────────────────
 Mode: wifi (disable)
 Please wait...

 [Ok] wifi disable | Completed

 - Restarting networking services...
Warning: Unit file of networking.service changed on disk, 'systemctl daemon-reload' recommended.
ifup: interface eth0 already configured`
> 

> ``

Also, upon running your commands in the comment above I am no longer able to visit github.com from the Pi. Is it possible

rubinkazan, 2016-11-03:
Did another restart and the ping's seemed to work
--- google.com ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 204.141/204.141/204.141/0.000 ms
root@PiyoteQueen:/etc/pihole# ping -c 1 github.com
PING github.com (192.30.253.113) 56(84) bytes of data.
64 bytes from 192.30.253.113: icmp_seq=1 ttl=48 time=238 ms

--- github.com ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 238.414/238.414/238.414/0.000 ms

Fourdee, 2016-11-04:
@rubinkazan 

I've updated DietPi-Update to use the dietpi.com mirror.

Please do the following and let me know if update is sucessful:
```
wget http://dietpi.com/downloads/testing/dietpi-update -O /DietPi/dietpi/dietpi-update
dietpi-update
```

rubinkazan, 2016-11-04:
root@PiyoteQueen:~# wget http://dietpi.com/downloads/testing/dietpi-update -O /DietPi/dietpi/dietpi-update
--1970-01-01 22:30:01--  http://dietpi.com/downloads/testing/dietpi-update
Resolving dietpi.com (dietpi.com)... 185.101.92.145
Connecting to dietpi.com (dietpi.com)|185.101.92.145|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 11707 (11K)
Saving to: ‘/DietPi/dietpi/dietpi-update’

/DietPi/dietpi/diet 100%[=====================>]  11.43K  7.95KB/s   in 1.4s   

1970-01-01 22:30:04 (7.95 KB/s) - ‘/DietPi/dietpi/dietpi-update’ saved [11707/11707]

Fourdee, 2016-11-04:
@rubinkazan 
Excellent,

And when you try to update?
```
dietpi-update
```

rubinkazan, 2016-11-04:
Followed by dietpi-update. 
Successfully Updated from v134-v136.
Thank you for the quick response and help!

Fourdee, 2016-11-04:
@rubinkazan 

Excellent, looks like a github specific issue. We will use dietpi.com as the default update mirror from v137 onwards :)

The update script gets replaced during the update, so make sure you copy the updated script, before updating to v137 (when its released):
```
wget http://dietpi.com/downloads/testing/dietpi-update -O /DietPi/dietpi/dietpi-update
```

closed 2016-11-04 by Fourdee