MichaIng/DietPi #592: DietPi-Software | Add tonido

Fourdee, 2016-11-04:


studioai, 2017-01-19:
if "sudo apt-get install libjpeg8 -y" causes error.

Try below

`cd /tmp
wegt http://ftp.us.debian.org/debian/pool/main/libj/libjpeg8/libjpeg8_8d-1+deb7u1_armhf.deb
dkpg -i libjpeg8_8d-1+deb7u1_armhf.deb`

in that way, I could run tonido

Fourdee, 2017-03-20:
> http://www.tonido.com/support/display/docs/Raspberry+Pi

Limited by provided binaries (no source available):
http://www.tonido.com/tonidodesktop_downloads/#desktop
- ARMv6 ??
- ARMv7
- x86_64

Fourdee, 2017-03-27:
Cannot  get sync client to work:
[image]

Fourdee, 2017-03-31:
- 🈴 ARMv6:
```
root@DietPi:/etc/tonido# export LD_LIBRARY_PATH=/etc/tonido; export TONIDODIR=/etc/tonido; ./tonidoconsole
Illegal instruction
```

Fourdee, 2017-04-01:
Tests + sync client:
- 🈯️ AMD64
- ARMv7

Fourdee, 2017-04-01:
Online doc:
http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=6476#p6476

Completed.

closed 2017-04-01 by Fourdee