MichaIng/DietPi #596: Pine64 - Font does not show cursor

midwan, 2016-11-06:
I noticed that the default font is changed during boot, on a Pine64 installation.
The replacement font is thinner (which is fine) but does not display the cursor at all, which makes it tricky to know where you are when navigating through text-based menus, editors or similar interfaces.

I couldn't find an option to restore the default font  in the dietpi-config.

Could we either:
a) leave the default font as it is (what was the reason of changing it?)
b) allow an option to switch back to it from the menu (maybe under the Display settings?)

Fourdee, 2016-11-06:
@midwan 

As far as DietPi goes, disabling the fb console cursor and font size, isn't something we've changed. I believe the missing cursor is a limitation with the framebuffer using the Longsleep image (which DietPi for Pine64 is built on).

I've also noticed this on the Odroid images, some reading here: 
http://forum.odroid.com/viewtopic.php?f=117&t=11758#p82647

Another possible solution compiling kernel with software cursor:
http://forum.odroid.com/viewtopic.php?f=117&t=11758#p83076

Fourdee, 2016-11-06:
@midwan 

> The replacement font is thinner (which is fine) 

Have you tried other fonts, do any of them provide a cursor?

midwan, 2016-11-06:
@Fourdee 
I haven't tried any other fonts, but while playing around with it I noticed something.
If you upgrade the firmware (which upgrades the kernel), then after the eventual reboot the font is changed again to something closer to the standard "default" font. But the cursor is still missing nevertheless.

I think I caught a glimpse of the boot messages mentioning a font change originally, that's why I suspected it was changed from DietPi. It was a message near the end of the boot process, since when it starts it does so with the default font, then that is changed before allowing you to login. But it looks like after the kernel upgrade the font is changed again, so maybe it's part of the kernel image.

Fourdee, 2016-11-06:
@midwan 

This workaround works, copy paste all. Blue cursor set after login:
```
cat << _EOF_ >> "$HOME"/.bashrc
infocmp > terminfo.txt
sed -i -e 's/?0c/?112c/g' -e 's/?8c/?48;0;64c/g' terminfo.txt
tic terminfo.txt
tput cnorm
_EOF_
```

k-plan, 2016-11-19:
@Fourdee 

> Blue cursor set after login (.bashrc is executed during login):

Very nice! Tested on OrangePi One (armv7l) it will work! 👍 
Where is as well a missing cursor on display ...

Will test it later on Odroid C1+ how had the same issue. :no_mouth:
https://github.com/Fourdee/DietPi/issues/260#issuecomment-219769702

k-plan, 2016-11-19:
> Will test it later on Odroid C1+ who had the same issue.

Done and it work! 👍 

Nice!

No big deal, but on Odroid C1+ cursor is white and on OrangePi One cursor is cyan.
Can live with it, is much better when no cursor.

midwan, 2017-01-22:
@Fourdee 
This took ages it seems, but I can confirm it also.

Could we have this by default on the affected systems please?

Fourdee, 2017-01-22:
@midwan 

Yep, this will be patched during v143 update, enabling this workaround by default on all pines.

Fourdee, 2017-01-22:
note to self:
- check + add to Odroids if required also

Fourdee, 2017-01-23:
Completed

closed 2017-01-23 by Fourdee

k-plan, 2017-01-23:
> note to self:
>
> - check + add to Odroids if required also

hmm... didn't see any fixes for Odroid C1/C2 or OrangePi One 

❔

meoow, 2017-02-24:
@Fourdee 
The commands above `tput cnorm` are required to run only once, and if you run `tic terminfo.txt` in root privilege, the configuration will be system-wide. `tput cnorm` is the sole command you need to run every time after you logged in. (or put it in somewhere like /etc/profile of course)

Fourdee, 2017-02-24:
@meoow 

Thanks for the info 👍