MichaIng/DietPi #597: Image for RaspberryPi's (all models) seems to be broken

Jurlind, 2016-11-08:
Hi,

I have tried to mount the iso or to open it with winrar, and I get the same result, namely that the image is broken. Any suggestions how to deal with this problem on a Windows platform? Actually, I tried win32disk however it modifies the size of the sd card, which makes the installation impossible, as there is no enough space on the disk anymore.

Cheers.

Fourdee, 2016-11-08:
@Jurlind 

Try:
- Downloading the image again: http://dietpi.com/downloads/images/DietPi_RPi-armv6-(Jessie).7z
- Extract the compressed file with 7zip (http://www.7-zip.org/download.html)
- Open win32diskimager
 - Select the extracted ```.img``` file
 - Click ```write```

DietPi automatically resizes the filesystem during 1st run. Allowing full use of your SD card capacity.

Jurlind, 2016-11-08:
Hi Fourdee,

thanks for your quick response. It seems that the installation is still failing, with error code 100. I am guessing that it is due to the size of the sd card, 1GB. I will try with a larger sd card, and will let you know, if this solves the problems I had. However, it definetly goes further in the installation by following your suggestion. Note that using 7zip, did not solve the problem of mounting the iso in Windows.

Cheers.

Fourdee, 2016-11-10:
@Jurlind 

>  due to the size of the sd card, 1GB.

Possibly, although 1GB should be fine for a default/minimal installation of DietPi RPi image.

If the error 100 occurs again, please can you take a screenshot / image / note of the onscreen information?

Fourdee, 2016-11-29:
No response, will mark as resolved.

@Jurlind 
Please reopen if required.

closed 2016-11-29 by Fourdee