MichaIng/DietPi #598: Many libraries are marked as "manual"

trajano, 2016-11-08:
`apt-mark showmanual` lists the manually installed packages.  I think it's best that this list be examined to make sure that packages that do not need to be loaded are not explicitly loaded in order to further reduce the space needed.

Fourdee, 2016-11-08:
On the current DietPi-RPi image v136, some obvious ones we can remove:
```
apt-get purge libboost-iostreams1.49.0 libboost-iostreams1.50.0 libboost-iostreams1.53.0 libboost-iostreams1.54.0 gcc-4.6-base gcc-4.7-base gcc-4.8-base libsigc++-1.2-5c2
```

trajano, 2016-11-08:
Here is my query

````
apt-mark showmanual | egrep '^lib' | grep -v -- "-bin" | grep -v -- "-progs"
apt-mark showmanual | grep -- "-dev"
````

trajano, 2016-11-08:
You should probably update https://docs.google.com/spreadsheets/d/1mDHGZC-H6tU6_O8kuLTG8d4A8Nt7lV1Q7MXTR_6qw30/edit#gid=0 when you finish unmarking these packages :+1:

MichaIng, 2017-10-29:
@Fourdee 
Superseded by https://github.com/Fourdee/DietPi/issues/1146

As said there, would be nice if we `apt-mark manual` ALL packages before installing the required once. I just tried it with the virtualbox image and it worked quite well. I am just not sure if all usual necessary firmware is installed by [firmware-linux-nonfree](https://packages.debian.org/de/jessie/firmware-linux-nonfree).

Fourdee, 2017-10-30:
@MichaIng

Thanks, i'll mark this as closed.

Firmware packages on all images are currently:
```
firmware-linux-nonfree firmware-realtek firmware-ralink firmware-brcm80211 firmware-atheros
```

> As said there, would be nice if we apt-mark manual ALL packages before installing the required once

Yep, we could probably do this on image prep script, eg:
```
apt-mark manual $(dpkg --get-selections | awk {'print $1'})
```

closed 2017-10-30 by Fourdee