MichaIng/DietPi #599: "build image" instructions

trajano, 2016-11-08:
I read that in the README.md that this is just a bunch of shell scripts, but I can't find the specific script used to build the images.  Perhaps a Wiki page with a build guide would help people to get started.  Specifically I wanted to try if I can run it through Travis CI.

Fourdee, 2016-11-09:
@trajano 

Image prep is currently a manual process, please see here: https://github.com/Fourdee/DietPi/blob/master/PREP_SYSTEM_FOR_DIETPI.sh

xanview, 2017-07-10:
Thank you for that, I'm looking at the instructions and it says:

```
#Resize 2nd parition to mininum size +50MB
```

Could you please provide instructions how you do this? :) - also, how do you then create the image file after doing this?

It isn't clear how you start and end the process. Do you start with a Debian version provided by the manufacturer of the board and then run these commands? - or do you start with something like debootstrap? -- Could you please add some comments for the beginning and completing steps.

Fourdee, 2017-07-12:
@xanview 

Mount the image under loopback, then run ```gparted``` (gui) to resize:

Pre-reqs
```
apt-get install gparted zerofree -y
```

```
	modprobe loop
	losetup -f
	losetup /dev/loop2 /path/to/image
	partprobe /dev/loop2
	gparted /dev/loop2
	sync

	zerofree -v /dev/loop2p1
	zerofree -v /dev/loop2p2

	losetup -d /dev/loop2

```

xanview, 2017-07-13:
Thank you for that, that's very useful :) - quick question, which images do you start out with? - For raspbian, I'm guessing the stock one, but what about for x86_64?

Fourdee, 2017-07-13:
@xanview 
- RPi = Raspbian Lite
- VM/x86_64 = Standard Debian net install
- OPi/BPi/NanoPi = Image built with ARMbian build tools
- Tinkerboard = Official ASUS image

Then we strip and optimize the image for DietPi:
https://github.com/Fourdee/DietPi/blob/master/PREP_SYSTEM_FOR_DIETPI.sh

MichaIng, 2018-01-09:
Should be merged into: https://github.com/Fourdee/DietPi/issues/1285

Will close it and reference from other issue with recommended information to add to script/DietPi readme.

closed 2018-01-09 by MichaIng