MichaIng/DietPi #600: DietPi-Software | Installing pi-hole changes IPv6 address.

sootstone, 2016-11-09:
This has no adverse effects as far as I can tell, but is just strange. Before installing pi-hole I have one global static IPv6 address ('A'), it is same regardless of how many times I reinstall. When I install pi-hole and choose to block IPv6 adverts, the installation says it will use the same address ('A') to block adverts. DietPi then reboots and its IPv6 address has changed to something else ('B'), this address is different on each (pi-hole) reinstall, but is still static. The only downside to this is if you have DNS records pointing to your Pi, you have to change them if you do reinstall.

Fourdee, 2016-11-11:
@sootstone 

Sounds like PiHole may be setting the IPv6 static IP incorrectly? DietPi only changes IP address in ```dietpi-config``` and is end user driven.

I dont have a IPv6 capable network to confirm. So, you'd need to check ```/etc/network/interfaces``` for any changes, before and after PiHole has been installed (and before DietPi reboots system).

sootstone, 2016-11-19:
I'll run a few tests, but otherwise feel free to close this.

Fourdee, 2016-11-28:
Will mark as closed. @sootstone After running your tests, please reopen if required.

closed 2016-11-28 by Fourdee