MichaIng/DietPi #601: [Discuss] Semantic Versioning in DietPi.

bhaveshgohel, 2016-11-10:
Hello @Fourdee ,
Today when i ssh to my RaspberryPi i see a message like  "Run now to update DietPi (from V132 to V136)."
And i wonder will this update are minor or major update to current version. Will upgrading breaks some things? 

[image]

I agree i can see the changelog file and guess the change. but this issue can easily solved by [Semantic Versioning](http://semver.org/).

We can implement it from today like V136 to V1.3.6! (just a suggestion)

Pl discuss. :smiley: 

-Bhavesh Gohel

Fourdee, 2016-11-12:
@bhaveshgohel 

Excellent idea 👍

bhaveshgohel, 2017-07-29:
Hello Dan ( @Fourdee ),
Just curious to know if any second though on this implementation?

Fourdee, 2017-07-29:
@bhaveshgohel 

> Will upgrading breaks some things?

Generally speaking, our updates "should not" break current installations. We run many tests our end, to ensure they are stable before releasing to the public. However, issues may occur, but we do try to resolve them with urgency.

I like the idea of Semantic Versioning, but we would need to completely rewrite the current update system to support it. It would also very likely mean a complete image/update reset, rendering the current installations up-patch-able with the new system.

Its not something we want to do currently, but i'll add this to the milestone for image resets and take a look then.

Fourdee, 2017-12-30:
We'll do this starting from DietPi v6.0

Fourdee, 2018-01-02:
Completed.
[image]

closed 2018-01-02 by Fourdee