MichaIng/DietPi #605: reduce mysqld usage

trajano, 2016-11-12:
This goes by the assumption that the usage of MySQL would be a little less than a normal server, we can reduce the amount of memory being used by limiting some resources.  In my case, I added the following file ` /etc/mysql/conf.d/reduce_resources.cnf` which contains

<pre>
[mysqld]
key_buffer=8M
max_connections=30
query_cache_size= 8M
query_cache_limit= 512K
thread_stack= 128K
</pre>

More info in http://opensourcehacker.com/2011/03/31/reducing-mysql-memory-usage-on-ubuntu-debian-linux/

Fourdee, 2016-11-15:
@trajano 
Done, will also be applied as patch during v137 update on client.

closed 2016-11-15 by Fourdee

MarcosPer, 2016-11-20:
I don't like this feature, i wasted 1 hour because my stored procedures doesn't work with this.

Fourdee, 2016-11-21:
@MarcosPer 

> i wasted 1 hour because of all my stored procedures doesn't work with this

Sorry to hear that. 
Are these issues with SQL queries, or with a peice of software (eg: NextCloud)? Also, how big is your current database size (if known)?

trajano, 2016-11-21:
Perhaps we can make this configurable, something like a2enconf