MichaIng/DietPi #606: Use F2FS

trajano, 2016-11-14:
Is it possible to set it up so the boot disk is formatted as F2FS rather than ext4?

https://en.wikipedia.org/wiki/F2FS

I am checking this out to see if I can just convert my existing one.
http://whitehorseplanet.org/gate/topics/documentation/public/howto_ext4_to_f2fs_root_partition_raspi.html

Fourdee, 2016-11-14:
@trajano 

Not tried it personally, but tutorial looks good. 
You may run into issues if F2FS doesn't support permissions and symlinks. DietPi (and the majority of software in dietpi-software) needs those to function correctly.

trajano, 2016-11-14:
Just asked stackexchange before I go on now ^_^

http://unix.stackexchange.com/questions/323155/does-f2fs-support-permissions-and-symlinks/323156#323156

trajano, 2016-11-15:
Using f2fs on my box right now. No noticeable change except for a periodic f2fs GC.

trajano, 2016-12-05:
This is probably best done on `dietpi.txt` perhaps we should have an option to specify what file system we want there.

trajano, 2017-02-17:
I would hold off on using F2FS due to #101 power fluctuations can kill a raspberry pi installation especially for a writable file system.  The forcefsck recovery does not fix the file system errors as I would expect.

Fourdee, 2017-06-17:
Marking as closed. Please reopen if required.

trajano, 2017-06-17:
Yup we have to hold off on this until we get get a proper Fsck to work #765

Cjkeenan, 2020-09-19:
What is the current state of F2FS and dietpi? I am upgrading to a Pi 4 with a USB SSD and was hoping to use a more modern file system.

MichaIng, 2020-09-20:
Yes the guide will still work. Before copying the rootfs to a backup location, I'd mount it as R/O:
```
dietpi-services stop
mount -o remount,ro /
```
That will often fail with "device is busy" in which case:
- Assure the swap file on this fs is off: `swapoff /var/swap`
- Stop/kill non-required processes manually: `htop`
- Last report: In `/etc/fstab` change rootfs `rw` flag with `ro` and `reboot`.

In case it's okay to still boot from SD card but only have the rootfs on USB SSD you can also use `dietpi-drive_manager` to move it to an external drive which supports formatting it as F2FS and Btrfs already.

Btw as I was never really experimenting with it: What is the practical benefit of F2FS and Btrfs over ext4? Probably we can create a testing image with the one or the other to allow more wide-spread testing an identifying possible issues with our or 3rd party tools expecting a different one.

Cjkeenan, 2020-09-20:
I was under the impression that it was simply a filesystem designed around flash/nand memory, and as such has optimizations for them such as reducing unnecessary writes given that flash has a limited amount of those. I do not know the specifics but I read that for any flash memory in some cases it improves performance, but in most cases it improves longevity of the drive.

MichaIng, 2020-09-20:
Some benchmark on Linux 5.0: https://www.phoronix.com/scan.php?page=article&item=linux-50-filesystems&num=1
It's on SSD but we would expect similar comparisons on SD cards. Yes I think it makes sense to provide an F2FS testing image. Probably we can add an option to change the rootfs type to our image creation script: https://github.com/MichaIng/DietPi/blob/master/.meta/dietpi-imager
But this can only work with direct cmdline access to change `rootfstype` of course and support by kernel/firmware (RPi).

Cjkeenan, 2020-09-20:
OK, sounds good, is there anything you need from me besides a tester? Also do you have any idea how long it would take to modify the image creation script? My current Rpi install has a corrupted gcc compiler so it is in a static state right now without the ability for me to make any changes, so I was hoping to fix it, but I am willing to wait if you think it is relatively easy to get a test image.

MichaIng, 2020-09-20:
Probably I can do that today (not rewrite DietPi-Imager but creating an RPi F2FS image manually), but have two other tasks in the pipeline.

Cjkeenan, 2020-09-20:
Yeah no biggie and not going to hold you to that, just wanted to get a sense was all. Love the work you and I appreciate you.

MichaIng, 2020-09-20:
Image for testing up (I didn't find time to test it myself): https://dietpi.com/downloads/images/testing/DietPi_RPi-ARMv6-Buster_F2FS.7z

There is some more work to do to add full support, as well to our build tools:
- We use `resize2fs -M` to shrink the images root file system to the minimal possible size. There is `resize.f2fs` but this currently only supports increasing the size, not shrinking it. So the only way to create an image is either creating an ext4 image first, shrink it and do conversion, or create a new partitioning with a moreless guessed min fs size so that all files fit. "Apparent" size + file count times block size should be failsafe? Testing required...
- Then `f2fs-tools` needs to be installed for `resize.f2fs`, `fsck.f2fs` etc. This cannot be done nicely during DietPi-Imager step but is better done during DietPi-PREP already.
- First boot expansion script requires this support as well: https://github.com/MichaIng/DietPi/blob/dev/rootfs/var/lib/dietpi/services/fs_partition_resize.sh#L53 (I hacked `resize.f2fs` into the image above)

I'll step by step add rootfs type detection (inkl. Btrfs) and using the correct tools to DietPi-FS_partition_resize, DietPi-PREP and DietPi-Imager first and then we need to think about a good concept where and how to add type selection and create or transform the partitions without leaving plenty of free space.

Cjkeenan, 2020-09-20:
Sounds amazing, anything special I have to do or just flash via Raspberry Pi Imager? Also any specific tests do you want run or just give day to day feedback?

MichaIng, 2020-09-20:
Just flash it with any method you usually do, `dd` or Rufus or Etcher or RPi Imager (which I never used so far, but plan to suggest/commit 7z support to it 🙂). For now I simply hope it boots and automated file system expansion on first boot works fine.

If you have some benchmark result with ext4, would be interesting to see possible differences with F2FS. Else, as you anyway wanted to setup a new production system, simply use it as normal and report if you face any issue. Since `ext4` is so widely used as rootfs default, some 3rd party software might expect it or behave unexpected, however from Linux-side what I read it should be pretty stable and supported.

I hope I find some time next week to do own tests and as well try it on my NanoPi NEO3 which should moreless be a proof of concept to add support for all (Linux 5.X) Armbian-based images (FriendlyARM + PINE64).

Joulinar, 2020-09-20:
@MichaIng 
doesn't looks like fs get extended on first boot. Initial setup failed with

```
[FAILED] DietPi-Software | Free space check: path=/ | available=86 MiB | required=500 MiB
[FAILED] DietPi-Software | Install aborted due to insufficient free space
```

MichaIng, 2020-09-20:
Does running `resize.f2fs /dev/mmcblk0p2` manually work?
What does the log say?
```
cat /var/tmp/dietpi/logs/fs_partition_resize.log
```

Joulinar, 2020-09-20:
```
root@DietPi:~# cat /var/tmp/dietpi/logs/fs_partition_resize.log
Removed /etc/systemd/system/local-fs.target.wants/dietpi-fs_partition_resize.service.
Disk /dev/mmcblk0: 29.7 GiB, 31914983424 bytes, 62333952 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x907af7d0

Old situation:

Device         Boot  Start     End Sectors   Size Id Type
/dev/mmcblk0p1        8192  532479  524288   256M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      532480 2274095 1741616 850.4M 83 Linux

/dev/mmcblk0p2:
New situation:
Disklabel type: dos
Disk identifier: 0x907af7d0

Device         Boot  Start      End  Sectors  Size Id Type
/dev/mmcblk0p1        8192   532479   524288  256M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      532480 62333951 61801472 29.5G 83 Linux

The partition table has been altered.
Info: Mounted device!
        Error: Not available on mounted device!
root@DietPi:~#
```

MichaIng, 2020-09-20:
Ahh, dammit, it cannot be used on a mounted device? Hmm, how to achieve that... Probably on a read-only mounted rootfs at least?
```
mount -o remount,ro /
resize.f2fs /dev/mmcblk0p2
mount -o remount,rw /
```
To do that reliably on boot, AFAIK our service needs to run earlier, not After= but Before= rootfs remount: https://github.com/MichaIng/DietPi/blob/dev/rootfs/etc/systemd/system/dietpi-fs_partition_resize.service#L6
But then the service cannot disable itself: https://github.com/MichaIng/DietPi/blob/dev/rootfs/var/lib/dietpi/services/fs_partition_resize.sh#L4
Puhh, needs some more throughts...

Joulinar, 2020-09-20:
not working that easy.

```
root@DietPi:/boot# mount -o remount,ro /
mount: /: mount point is busy.
root@DietPi:/boot#
```

MichaIng, 2020-09-20:
Yes that is the reason why it needs to be done very early. I'm not sure how to reliably determine the process that "blocks" the fs, sometimes `lsof` shows not a single write-opened file and it still does not work or only after playing a bid around suddenly without being able to identify any certain step...

Last resort would be:
Add `ro` on first boot (or even better on all f2fs images right from the start) to /etc/fstab. When the service runs, it expands the file system, alters fstab to have rootfs read-write and immediately reboots. On reboot, if the service finds rootfs-rw, it only disables itself and boot continues.

Joulinar, 2020-09-20:
ok I relay did the manual way while mounting the SD card on a different system and expanding the fs. First Boot now completed successful. Will need to run some test on it. Boot times seems to be higher. Mounting the device takes quite some time. To be verified.

```
root@DietPi3:~# systemd-analyze blame
         12.870s dev-mmcblk0p2.device
         11.945s systemd-fsck-root.service
```

Cjkeenan, 2020-09-20:
I noticed the same thing when I was experimenting F2FS on Raspios, I think it has something to do with fsck running every boot and that taking extra time with F2FS.

Cjkeenan, 2020-09-20:
> ok I relay did the manual way while mounting the SD card on a different system and expanding the fs. First Boot now completed successful. Need to run some test on it. Boot times seems to be higher. Mounting the device takes quite some time. To be verified.
> 
> ```
> root@DietPi3:~# systemd-analyze blame
>          12.870s dev-mmcblk0p2.device
>          11.945s systemd-fsck-root.service
> ```
> 
> ```
> root@DietPi3:~# lsblk -o name,fstype,label,size,ro,type,mountpoint
> NAME        FSTYPE LABEL  SIZE RO TYPE MOUNTPOINT
> mmcblk0                  29.7G  0 disk
> ├─mmcblk0p1 vfat   boot   256M  0 part /boot
> └─mmcblk0p2 f2fs         29.5G  0 part /
> root@DietPi3:~#
> ```

So how exactly did you accomplish this? Because outside of this resizing issue the build needs to be tested so if I can help with that I want to.

Cjkeenan, 2020-09-21:
So I got basically everything up an running except for a couple quirks, and idk if it is with dietpi in general or with f2fs:
1. In Node-RED, I use node-red-contrib-alexa-remote2 to interface with my alexa devices, but it needs to authorize with Amazon's server and save that cookie, but for whatever reason no matter the path I give it, it does not want to actually write that cookie to a file, I am theorizing due to lack of permissions but I do not know for sure. As a result, it require reauthentication every time a single change is made, which is quite annoying. Again idk if this is a dietpi overall issue or F2FS issue.
2. DietPi-Backup does not seem to support F2FS and as a result does not generate a backup image unless another path is provided onto another drive. Does anyone know a way of uploading the image backups from DietPi-Backup to a cloud storage such as Google Drive or OneDrive?
3. Node-RED commands that come with the install such as start, stop, restart, and status do not seem to work, stating that `dietpi-services: command not found' and if I copy and paste the exact line of code from that command, it works fine. Those commands are located in `/usr/bin/` but even when I `cd` there, and run the code it works fine so idk what is going on. 

**Command**: 
```
dietpi-services start node-red
journalctl -f -n 0 -u node-red -o cat
```

Finally in terms of performance, I got a 16% gain over ext4 in the benchmark I ran with the same SD card.

MichaIng, 2020-09-21:
Many thanks for testing.

Within the Node-RED environment you need to call DietPi scripts with full paths. The aliases are only created for interactive bash shells. And actually it does not make much sense to use `dietpi-services` which is just an error handled wrapper with nice output, at least for single service handling not more. For Node-RED I'd simply use:
```
systemctl start node-red
systemctl stop node-red
```
Also take care that those commands require root permissions, so you need to run those with `sudo` from within Node-RED.

While `sudo` can be used to grant the user write permissions to everywhere, for locations where it needs to write regularly, I'd grant it native UNIX permissions to those files/directories, or at best write to the services home directory where possible: `/mnt/dietpi_userdata/node-red`
Where does this cookie needs or is aimed to be created and which other service/process/user has/needs write access there?

> DietPi-Backup does not seem to support F2FS

Good find 👍. No reason to not support it, easy to fix: https://github.com/MichaIng/DietPi/commit/b6d7fb6447fcec4d23fca465c31bdf3f6f22ce2f
Changelog: https://github.com/MichaIng/DietPi/commit/6fb8aa99f85a9bb1794c0c585bf7ae498bd27925

Cjkeenan, 2020-09-21:
> Within the Node-RED environment you need to call DietPi scripts with full paths, e.g. `/boot/dietpi/dietpi-services`. The aliases are only created for interactive bash shells. And actually it does not make much sense to use `dietpi-services` which is just an error handled wrapper with nice output, at least for single service handling not more. For Node-RED I'd simply use:
> 
> ```
> systemctl start node-red
> systemctl stop node-red
> ```
> 
> Also take care that those commands require root permissions, so you need to run those with `sudo` from within Node-RED.

Thank you so much, got these commands working now. I ended up using the full services path since I did not want to deal with `sudo` permissions.

> While `sudo` can be used to grant the user write permissions, for file I/O I'd grant it native UNIX permissions to those files/directories, or at best write to the services/users home directory where possible: `/mnt/dietpi_userdata/node-red`
> Where does this cookie needs or is aimed to be created and which other service/process/user has/needs write access there?

This is the File Path I am trying to write: `/mnt/dietpi_userdata/node-red/alexaAuth/BH_authFile`. Am I misunderstanding something or for is that last entry being considered a directory that does not exists so it is freaking out?

This is the error I get in NR: `"EACCES: permission denied, open '/mnt/dietpi_userdata/node-red/alexaAuth/BH_authFile'"`

Glad to be of help, is that backup fix already pushed an available via DietPi-Update?

Joulinar, 2020-09-21:
> Glad to be of help, is that backup fix already pushed an available via DietPi-Update?

nope, not yet available on any update. Will be released on upcoming version 6.33. However you could switch to development branch if you really need it. Or you do the update manual on `dietpi-backup` script.

Cjkeenan, 2020-09-21:
No, no direct need right now, just curious was all. Thank you guys so much for your help.

MichaIng, 2020-09-21:
I hope it is easy enough to add it manually to `/boot/dietpi/dietpi-backup`?

About the permissions error, can you try:
```
chown -R nodered:nodered /mnt/dietpi_userdata/node-red
```

Cjkeenan, 2020-09-21:
> About the permissions error, can you try:
> 
> ```
> chown -R nodered:nodered /mnt/dietpi_userdata/node-red
> ```

Perfect, it worked, is there a way to do this by default? OMG I cannot tell you how happy I am that this fixed it, I have tearing my hair out trying new directories, trying new files, even went to the node's github to see if it was an issue there.

MichaIng, 2020-09-21:
Actually this is done automatically on install: https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-software#L8327
Did you create files manually there or used the node-red CLI command with a different user than "nodered"?

Cjkeenan, 2020-09-21:
No I installed from DietPi-Software and I primarily use this through the Web Interface, so I am not sure if another user is being used there.

MichaIng, 2020-09-21:
Hmm strange, I cannot imagine a reason then. However, keep an eye on it if there are permission issues again, probably we did overlook something.

Cjkeenan, 2020-09-21:
I will say, and I do not know which exactly, but if there are logs, I can retrieve them for you. But a couple of programs, including pihole, pivpn, and node-red, failed on their initial install and then after I retried via the prompt, it installed fine, but that was the only oddity I remember.

MichaIng, 2021-03-11:
Partly done by adding support for F2FS and Btrfs partitions to the code via: #4164
Now we need to create images to test it thoroughly.

Micha-Btz, 2021-09-06:
Are there images? I would like to test images with btrfs for /. I really like the snapshot possibility.

For now my raspi 3B+ runs on ssd with ext4 / and btrfs for home. Sadly I don't get a working btrfs for boot.
This tutorial seems not to work anymore. https://hackmd.io/@seaQueue/HJOFQphWW?type=view#Creating-an-Initramfs

MichaIng, 2021-09-06:
First imaging task is to migrate everything to Bullseye. Then I'll see if I find time again to create images with alternative root filesystem types.

I'd skip that subvolume part first, to keep it simply. This can be done later as well. Also `elevator=deadline` should not be added to cmdline anymore, as it is deprecated, both, the deadline scheduler (replaced with deadline-mq) as well as the elevator option itself.

Micha-Btz, 2021-09-07:
@MichaIng since the buster image for the raspis is not more available, can you point me to the latest used cmdline.txt and config.txt?
Thanks

Joulinar, 2021-09-07:
The image is still available on our server https://dietpi.com/downloads/images/DietPi_RPi-ARMv6-Buster.7z

Micha-Btz, 2021-09-07:
yeah, but I need ARMv7 Buster and there is only bullseye

MichaIng, 2021-09-07:
There never was an ARMv7 Buster image. Note that the ARMv6 images is compatible with all RPi models.

Current `cmdline.txt`: https://github.com/MichaIng/DietPi/blob/ad19d27/PREP_SYSTEM_FOR_DIETPI.sh#L560
Current `config.txt:` https://github.com/MichaIng/DietPi/blob/e662281/config.txt

Joulinar, 2021-09-07:
The link above I posted our Buster image for Raspberry Pi

rekcodocker, 2023-02-04:
This issue warped into a discussion on BTRFS.  Fine that this works, but I am really interested in F2FS as it is designed to extend the life of SD cards. 

So, still the request: Can DietPi be provided out of the box with an F2FS image? 
I am definitely willing to test this.

MichaIng, 2023-02-04:
We had a working RPi F2FS images available for testing a while ago. Looks like I missed to link it here. Has become easier than ever to generate one.

There are a few `e2fsck` calls in our build script, but for NanoPi R5S/R6S and Allo GUI images only, otherwise it should work: https://github.com/MichaIng/DietPi/blob/dev/.build/images/dietpi-build
Let me trigger a build and we see how it goes.

MichaIng, 2023-02-04:
Image ready: https://dietpi.com/downloads/images/testing/
Let me know if you need an ARMv6 (Raspberry Pi 1/Zero) or ARMv7 (Raspberry Pi 2 v1.1) image.

rekcodocker, 2023-02-05:
That is absolutely fantastic, thank you so much!

Found an issue with the size. I install it on a 16GB sd card. 
I tried to install docker-compose and it failed because there was not enough disk space.

Tried 'resize filesystem' using dietpi-drivemanager. 
It reboots.  

**Issue**: size has not changed. 

Tried to alter the filesystem on my laptop.
I found that the partition is sized at 14.7GB.  But the f2fs filesystem reports 890M (!)

I thought that maybe the partition is resized, but the filesystem still needs to 'grow' into the new partition size.
Tried 'resize.f2fs' on my laptop.
Did not work, this was the output: 

```
sudo resize.f2fs /dev/sdb2 
Info: [/dev/sdb2] Disk Model: STORAGE DEVICE  
Info: Segments per section = 1
Info: Sections per zone = 1
Info: sector size = 512
Info: total sectors = 30845952 (15061 MB)
Info: MKFS version
  "Linux version 5.15.0-1031-azure (buildd@lcy02-amd64-010) (gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #38-Ubuntu SMP Mon Jan 9 12:49:59 UTC 2023"
Info: FSCK version
  from "Linux version 5.15.0-1031-azure (buildd@lcy02-amd64-010) (gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #38-Ubuntu SMP Mon Jan 9 12:49:59 UTC 2023"
    to "Linux version 5.15.0-58-generic (buildd@lcy02-amd64-101) (gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #64-Ubuntu SMP Thu Jan 5 11:43:13 UTC 2023"
Info: superblock features = 0 : 
Info: superblock encrypt level = 0, salt = 00000000000000000000000000000000
Info: total FS sectors = 1826816 (892 MB)
Info: CKPT version = 223718b2
Info: Duplicate valid checkpoint to mirror position 1024 -> 512
Info: Write valid nat_bits in checkpoint
        Error: Device size is not sufficient for F2FS volume, more segment needed =12534

```

rekcodocker, 2023-02-05:
Update: 
Tried to flash the card and without booting, resize the f2fs partition.   All on my laptop.

First tried with KDE partition editor.
It resizes the partition but not the filesystem.
Same error: The partition grows to 14.7GB.  But the filesystem does not resize.
The filesystem lives on a 14GB partition but still thinks it is 890MB in size.

Tried a cleanly flashed SD card again.
Tried the resize.f2fs from the commandline.
No error but no resize either: It reports it is 890MB in size (which is correct as I have not enlarged the partition).

I am at a loss; I don't know how to resize this filesystem.

MichaIng, 2023-02-05:
Does it not resize automatically on first boot? The code for that is in place.

After resizing the partition, you need to inform the kernel about it:
```sh
partprobe /dev/mmcblk0
partx -u /dev/mmcblk0
```
Two methods, just to be sure.

rekcodocker, 2023-02-05:
It should have... but it didn't.  The partition was resized but the filesystem was not.
I also could not resize the filesystem on my laptop. See previous post for the output of the resize.f2fs command.

Workaround for now:  
I rsynced all the files from the filesystem, deleted the fs, created a new f2fs and rsynced all the files back to it. 

It is up and running on f2fs.
Have installed docker-compose and nodered, and I will continue testing.

rekcodocker, 2023-02-05:
Ok, tried again.
Clean SD card, 16GB.

After initial login, it updates and installs minimal software. 
Then I get to the command prompt.

I see this:
**df -h** -> indicates the root filesystem op /dev/root is **890MB** in size, 139M available which is 85% used.

**parted**, show partition information:  **The partition is 15.8GB** and the filesystem is f2fs.

Start dietpi-drivemanager.
Ran filesystem repair (requires reboot) -> no change.

Start dietpi-drivemanager.
Ran filesystem resize (requires reboot) -> no change.
no change. 

Rebooted again. -> no change.

partition = 15.8GB, sounds resized.
Filesystem is 890MB, is not resized.

It is not a failure to re-read the partition info.  It persists after reboot.

Did you read where I got the same numbers when I inserted the SD card on my laptop? 
And I also tried 'resize.f2fs'  on my laptop?  It did not work. It did not grow the filesystem into the resized partition. 
If it didn't work on my laptop, maybe the same error happens in dietpi. 

It is not re-reading the partition table. 
The resize of the partition works.
The resize of the FS does not work.

It is not caused re-reading the partition info. If it was, I could insert the SD card in my laptop and see a filesystem of 15GB.  Instead I see 890MB.

MichaIng, 2023-02-05:
`fsck` cannot resize a filesystem, that is expected. So `resize.f2fs` failed. Can you show the output of:
```sh
cat /var/tmp/dietpi/logs/fs_partition_resize.log
resize.f2fs "$G_ROOTFS_DEV"
mount -o remount,ro /
resize.f2fs "$G_ROOTFS_DEV"
mount -o remount,rw /
```

rekcodocker, 2023-02-05:
I can't run these on the pi I think?  I can't resize an active partition.

But I got this output on my laptop:

sudo resize.f2fs /dev/sdb2 
Info: [/dev/sdb2] Disk Model: STORAGE DEVICE  
Info: Segments per section = 1
Info: Sections per zone = 1
Info: sector size = 512
Info: total sectors = 30845952 (15061 MB)
Info: MKFS version
  "Linux version 5.15.0-1031-azure (buildd@lcy02-amd64-010) (gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #38-Ubuntu SMP Mon Jan 9 12:49:59 UTC 2023"
Info: FSCK version
  from "Linux version 5.15.0-1031-azure (buildd@lcy02-amd64-010) (gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #38-Ubuntu SMP Mon Jan 9 12:49:59 UTC 2023"
    to "Linux version 5.15.0-58-generic (buildd@lcy02-amd64-101) (gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #64-Ubuntu SMP Thu Jan 5 11:43:13 UTC 2023"
Info: superblock features = 0 : 
Info: superblock encrypt level = 0, salt = 00000000000000000000000000000000
Info: total FS sectors = 1826816 (892 MB)
Info: CKPT version = 223718b2
Info: Duplicate valid checkpoint to mirror position 1024 -> 512
Info: Write valid nat_bits in checkpoint
        Error: Device size is not sufficient for F2FS volume, more segment needed =12534

MichaIng, 2023-02-05:
Um: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1002034
Looks like solved in Bookworm but was never solved for Bullseye 😞.

I'll generate a Bullseye image.

rekcodocker, 2023-02-05:
Nicely found, well done!

In de meantime, my manually resized image is running happily right now.

Impressed that you could so quickly generate a working F2FS image, truely impressed.  I take it you have this automated because it was so quick.
I have several Raspberries running on an F2Fs image which I had to convert from Ext4 first - tedious process.

Is it an option to include this in the builds and also the download options? (with the tag 'testing' attached?)

Is it an option (and this is a long standing wish of mine, I was never able to create it myself) to have zstd-compression enabled on F2FS to improve performance of the SD card?

MichaIng, 2023-02-05:
> I take it you have this automated because it was so quick.

Yes exactly. We do now have build scripts, trigger and run them via GitHub Actions. General support for F2FS, as said, as been added already a while ago. Not sure why I forgot to mention it here 😅.

> Is it an option to include this in the builds and also the download options?

Yes I can add this to the overall build list we trigger after a DietPi release. We can add them to the download page as well. But not sure how to deal with the F2FS bug. There are no backports and sadly the package from Bookworm depends on glibc 2.34, hence cannot be installed on Bullseye. That after flashing, expansion needs to be done from a remote system with sufficiently recent `f2fs-tools` doesn't sound so great. We could however provide Bookworm images only. As F2FS is not well tested, having it combined with Debian testing should be fine 😄? Bookworm has reached first freeze stages and will be released this August, so it is mostly fine, runs in production on my home server and `dietpi.com` server as well.

> Is it an option (and this is a long standing wish of mine, I was never able to create it myself) to have zstd-compression enabled on F2FS to improve performance of the SD card?

This is a mount option, I guess? Does it automatically compress existing data when set, or only newly written data? I generally wanted to add custom mount option support to `dietpi-drive_manager`, but there is a larger task needed for this script, to not overwrite the whole `/etc/fstab` but only the individual entries that you edit. But

> to have zstd-compression enabled on F2FS to improve performance of the SD card?

This would reduce performance on the SD card, not improve it, since compression/decompression needs to be done on every write/read action, or do I misunderstand something?

rekcodocker, 2023-02-05:
If you add it, it should definitely come with a warning label and a 'how to work around the issue' note.  It is really unusable as people will run out of diskspace almost immediately and they will start filing bugs.

However, this may help.
Te bug seems to have occurred in 2021 since the mentions of that bug are on different forums by that date. 

Here is an interesting remark: https://bugs.archlinux.org/task/71801 says that they found the error occurs in f2fs-tools 1.14.0-2, and downgrading to 1.14.0-1 would not cause the error.
If that works, you can fix the version of the package on 1.14.0-1 for the time being, and return to the latest when the issue is fixed.

**Compression**: It compresses only new data. So you have to do this when you create the image to reap the full benefits.  
I have tried this since f2fstools 1.14 came out supporting on the fly compressopn. My problem was that I could not get it to boot. 
If you are interested to look into it, I took clues from here:  https://wiki.archlinux.org/title/F2FS . 
It states that you need to use a specific option when creating the filesystem as well as when you mount it. 
Apparently at creation time you can state 
-O extra_attr,compression,compress_algorithm=zstd
and when you mount it:
-o compress_algorithm=zstd:6,compress_chksum

Note the inconsistencies here: When creating you use a capital 'O', the options are preeded by 'extra_attr' and they include 'compress' as well as 'compress_algorithm'
The mount options are slightly different.

MichaIng, 2023-02-05:
I saw the Arch bug report. However note that `1.14.0-2` on Arch isn't equal to `1.14.0-2` on Debian. These are build suffixes. F2FS was on version `1.14.0`, and Arch as well as Debian released a second package based on the same source, but with changes to build flags, distribution/package files or such. And I guess it was fixed upstream with v1.15.0 and Arch backported the fix into their 1.14.0, adding another packaging suffix increment. So as long as Debian doesn't do the same, we won't see this fix in Bullseye.

`-O`/`-o` is btw expected. These are two different things:
- `-O` is the common option for `mkfs` to set filesystem capabilities. Here you add the capability to do filesystem compression. Interesting is btw the `compress_algorithm=zstd`, which isn't mentioned on the man page: https://manpages.debian.org/mkfs.f2fs
- `-o` is the `mount` option for setting mount options, in this case to make actual use of the filesystem capability. So you can choose the algorithm there, the compression level (`6` is the default for zstd), and checksums. I guess the letter could be skipped, increasing performance but increasing the risk for undetected filesystem corruption as a result of a faulty compression.

It takes another iteration to enable these features for our images, respectively for `dietpi-drive_manager` in the first place.

[19 more replies]