MichaIng/DietPi #610: Installing mate

MCosta2016, 2016-11-16:
installing mate or lxde i get this error, that in the vm machine doesn't appear.
I can resolve it ?

Many thanks, Marco

[image]

Fourdee, 2016-11-16:
@MCosta2016 

Unable to replicate here, so lets try changing the apt mirror:
- run ```dietpi-config```
- select ```Network Options: NAS/Misc```
- select ```Apt Mirror```
- select ``` ftp://ftp.fu-berlin.de/linux```

Lets clean, update the apt cache and repair any error-ed package installs:
```
apt-get clean
apt-get update
apt-get install -f
```

Now please try to install MATE via ```dietpi-software``` and let us know outcome.

MCosta2016, 2016-11-16:
ok, good, but ...
[image]
[image]

MCosta2016, 2016-11-16:
i think that is another problem ... sigh!

MCosta2016, 2016-11-16:
infact i try to ping 8.8.8..8 and it is reacheable.... i don' t understand why the virtual machine go on without problem.... sigh!

Fourdee, 2016-11-16:
@MCosta2016 

> infact i try to ping 8.8.8..8 and it is reacheable.... i don' t understand why the virtual machine go on without problem.... sigh!

Lets take a look at your network settings, please paste contents of the following (or screenshot). Hide any WiFi keys before hand:
```
cat /etc/network/interfaces
```

Fourdee, 2016-11-29:
No response, will mark as closed.

@MCosta2016 
Please reopen if required.

closed 2016-11-29 by Fourdee