MichaIng/DietPi #611: DietPi-Software | PiVPN

Fourdee, 2016-11-19:
Issues reported by users:
- PiVPN fails to install
- PiVPN not showing up as installed after install: http://dietpi.com/phpbb/viewtopic.php?f=11&t=865

Fourdee, 2016-11-19:
> PiVPN not showing up as installed after install: 

Most likely cause is selecting ```Yes``` to reboot after PiVPN installer is completed.

k-plan, 2016-11-19:
@Fourdee 

> Most likely cause is selecting `Yes` to reboot after PiVPN installer is completed. This prevents DietPi-Software from completing the remaining installation code, which, also sets installed states at the end.

💯  👍 

Do some testing as well in the last days, but miss to write it up, with same result. Sorry.

If you choose:

[image]

at the last install step and let `dietpi-software` do the build-in auto-reboot after finishing installation, all went well.

```
root@OPi-One:~#  dietpi-software list | grep '[[:space:]]=2'
id 3 | =2 | mc: midnight commander, powerful file manager |
id 12 | =2 | iperf: internet protocol bandwidth measuring tool |
id 13 | =2 | mtr-tiny: full screen ncurses traceroute tool |
id 15 | =2 | tcpdump: command-line network traffic analyzer |
id 17 | =2 | git client: git clone etc |
id 102 | =2 | rsyslog: system logging | 
id 103 | =2 | dietpi-ramlog: minimal, optimized logging |
id 104 | =2 | dropbear: ssh server | 
id 117 | =2 | pivpn: openvpn installer & management tool | +rsyslog | http://dietpi.com/phpbb/viewtopic.php?http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=3469#p3469

```

But it is not your fault, it's a problem using default installer script. 

I think, we can not do anything to prevent this, except to write a message before running installation, like you do with `pihole` installation for setting a static ip-address.
BTW: message/dialog for setting a static ip-address before starting installation will be likewise a good idea for `pivpn`.  😉 

> OpenVPN repo that PiVPN adds, does not support ARM64 binaries

Not only for ARM64, repo does only support Architectures: i386 amd64

http://swupdate.openvpn.net/apt/dists/jessie/

Same as last openvpn builds: http://swupdate.openvpn.net/apt/dists/jessie-20161103104827002230280/main/

But here redfast00 had to take a look in his installer script:
https://github.com/pivpn/pivpn/blob/mas ... ll.sh#L433

Only thing user can do is to delete after installation:

`rm /etc/apt/sources.list.d/swupdate.openvpn.net.list `

Fourdee, 2016-11-29:
@k-plan 

> rm /etc/apt/sources.list.d/swupdate.openvpn.net.list

This is now removed during ```dietpi-software``` uninstall.

I've also disabled PiVPN from rebooting system. Should prevent a incomplete DietPi-Software install with our users.

Marking as completed.

closed 2016-11-29 by Fourdee

k-plan, 2016-12-04:
@Fourdee 

> I've also disabled PiVPN from rebooting system. Should prevent a incomplete DietPi-Software install with our users.

Tested now with v138 and it will work. Well done, nice job. 👍 

> rm /etc/apt/sources.list.d/swupdate.openvpn.net.list
> This is now removed during `dietpi-software` uninstall.

hmm, it will be a good idea (better), to do this direct after installation on ARM devices, before `dietpi-software` will force a auto reboot. Because now you will get error messages on every `apt-get update` run.

> W: Failed to fetch http://swupdate.openvpn.net/apt/dists/jessie/Release Unable to find expected entry 'main/binary-armhf/Packages' in Release file (Wrong sources.list entry or malformed file)

But it's a failure of PiVPN installer, not from DietPi. Only a "_work around_" to get off this annoying error messages.