MichaIng/DietPi #612: DietPi-Software | CAVA (console spectrum)

Fourdee, 2016-11-19:
As title: https://github.com/karlstav/cava#cava-mp

Implementing with:
- MPD FIFIO
- ```dietpi-justboom```

k-plan, 2016-11-19:
Huuh ...  as well with 3.2 / 3.5 LCD Touchscreen Shield Support in DietPi ?

____________

_PS: - after the gaming trip (retropi, OpenTyrian, Cuberite, MineOS, AmiBerry, DXX-Rebirth) you are now on the audio trip (JustBoom)?_  😄

Fourdee, 2016-11-19:
@k-plan 

> as well with 3.2 / 3.5 LCD Touchscreen Shield Support in DietPi ?

😉 yep, looks great on both :D

k-plan, 2016-11-19:
@Fourdee 

> looks great on both

wow, great news! 😃 

Have you please any "preview" picture of a working device? 
Would be very nice to see how it looks.

Is support for the 3 or 4 display keypads possible (MPD or CAVA) ?

http://odroid.com/dokuwiki/doku.php?id=en:c1_tft_keypads

http://odroid.com/dokuwiki/doku.php?id=en:3.5inch_lcd_shield#using_keypad

Fourdee, 2016-11-19:
@k-plan 

> Is support for the 3 or 4 display keypads possible (MPD or CAVA) ?

I'll take a look at the Odroid LCD 3.5 first. Next/previous track and volume seems good combo for the keys?

> Have you please any "preview" picture of a working device? 
Would be very nice to see how it looks.

Yep, i'll do a video when its nearly done 👍. 
Having a few issues with CAVA performance on i2c, appears to be related to when the majority of bars are filled, temporally drastically lowers framerates.
Note to self: try a larger font.

k-plan, 2016-11-19:
@Fourdee 

> Yep, i'll do a video when its nearly done

Wow ... a video. Thanks, very nice. 😃 

> Next/previous track and volume seems good combo for the keys?

Hmm .. for 3.2 '' LCD (WaveShare) witch 3 keypads I think:

“Up” => Next

“Down” => Previous

“Enter” => Stop

3.5 '' LCD (Hardkernel) witch 4 keypads I think:

“Up” => Volume + (or: Next)

“Down” => Volume - (or: Previous)

“SPACE” => Previous (or: Stop)

“Enter” => Next (or: Repeat/Play)

(volume is easier to adjust direct at speaker)

Fourdee, 2016-11-20:
@k-plan 

"Bash'd it", Odroid C2 + Odroid LCD 3.5:

```
cat << _EOF_ > /etc/odroid_lcd_mpd_keys.sh
#!/bin/bash

VALUE=0
TICK_RATE=0.06
KEY_RELEASE_DELAY=0.3

aSW_MIN=(
    0
    505
    670
    760
)

aSW_MAX=(
    15
    525
    690
    780
)

aSW_COMMAND=(
    'next'
    'prev'
    'play' #+ pause
    'stop'
)

#Main

while true
do

    VALUE=\$(cat /sys/class/saradc/ch0)

    for ((i=0; i<\${#aSW_MIN[@]}; i++))
    do

        if (( \$VALUE <= \${aSW_MAX[\$i]} && \$VALUE >= \${aSW_MIN[\$i]} )); then

            # - Play/pause
            if [ "\${aSW_COMMAND[\$i]}" = "play" ] || [ "\${aSW_COMMAND[\$i]}" = "pause" ]; then

                aSW_COMMAND[\$i]='play'
                if (( \$(mpc | grep -ci -m1 'playing') )); then

                    aSW_COMMAND[\$i]='pause'

                fi

            fi

            # - exec
            mpc \${aSW_COMMAND[\$i]} &> /dev/tty1
            sleep \$KEY_RELEASE_DELAY
            break

        fi

    done

    sleep \$TICK_RATE

done

unset aSW_MIN
unset aSW_MAX
unset aSW_COMMAND

_EOF_

chmod +x /etc/odroid_lcd_mpd_keys.sh

cat << _EOF_ > /etc/systemd/system/odroid_lcd_mpd_keys.service
[Unit]
Description=odroid_lcd_mpd_keys
After=sound.target

[Service]
Type=forking

ExecStart=/bin/bash -c '/etc/odroid_lcd_mpd_keys.sh &'
#ExecStop=/usr/bin/killall -w odroid_lcd_mpd_keys.sh

[Install]
WantedBy=default.target
_EOF_
systemctl daemon-reload

systemctl restart odroid_lcd_mpd_keys.service
```

k-plan, 2016-11-20:
@Fourdee 

looks great, nice work.  But I can't test it. 😢 

It's seems to be time going out to buy a DAC HAT or a sound card for some of my devices.
Any suggestion?

Fourdee, 2016-11-21:
@k-plan 
I've got a spare HiFi Shield Plus for C1/C2, will send it over in next few days.

For Raspberry Pi, JustBoom DAC HAT (https://www.justboom.co/product/justboom-dac-hat/) or DAC Zero (https://www.justboom.co/product/justboom-dac-zero-phat/) are both excellent.

Fourdee, 2016-11-21:
The low framerate (10-15) on the Odroid LCD shield causes massive overheads when cava is running, even at matching framerates:
[image]

> http://www.hardkernel.com/main/products/prdt_info.php?g_code=G147435282441
- The frame rate is around 10~15FPS due to slow GPIO toggling speed.

Fourdee, 2016-11-24:
Online Doc: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=3928#p3928

@k-plan 
> PS: - after the gaming trip (retropi, OpenTyrian, Cuberite, MineOS, AmiBerry, DXX-Rebirth) you are now on the audio trip (JustBoom)? 

Lol, yep 👍

Fourdee, 2016-11-29:
Completed.

closed 2016-11-29 by Fourdee