MichaIng/DietPi #613: Error installing NextCloud

joaofl, 2016-11-22:
While installing NextCloud, i got an error that seems to be related to dependency on php

The error is:
sudo: php: command not found
cat: /var/www/nextcloud/config/config.php: No such file or directory
sed: can't read /var/www/nextcloud/config/config.php: No such file or directory
/DietPi/dietpi/dietpi-software: line 5884: php: command not found
/DietPi/dietpi/dietpi-software: line 5884: / 1024 / 1024: syntax error: operand expected (error token is "/ 1024 / 1024")

Follows a part of the log:

`Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following extra packages will be installed:
  libass5 libavc1394-0 libavcodec57 libavdevice57 libavfilter6 libavformat57 libavresample3
  libavutil55 libbluray1 libbs2b0 libcaca0 libcdio-cdda1 libcdio-paranoia1 libchromaprint1
  libdc1394-22 libenca0 libfftw3-double3 libflite1 libglu1-mesa libgme0 libiec61883-0
  libmodplug1 libopenal-data libopenal1 libopencv-core2.4 libopencv-imgproc2.4 libopenjp2-7
  libpgm-5.1-0 libpostproc54 libraw1394-11 librubberband2 libshine3 libsnappy1 libsodium13
  libssh-gcrypt-4 libswresample2 libswscale4 libtwolame0 libva-drm1 libvdpau1 libwavpack1
  libx265-87 libzmq3 libzvbi-common libzvbi0
Suggested packages:
  ffmpeg-doc libbluray-bdj libfftw3-bin libfftw3-dev libportaudio2 libraw1394-doc vdpau-driver
Recommended packages:
  libaacs0
The following NEW packages will be installed:
  ffmpeg libass5 libavc1394-0 libavcodec57 libavdevice57 libavfilter6 libavformat57
  libavresample3 libavutil55 libbluray1 libbs2b0 libcaca0 libcdio-cdda1 libcdio-paranoia1
  libchromaprint1 libdc1394-22 libenca0 libfftw3-double3 libflite1 libglu1-mesa libgme0
  libiec61883-0 libmodplug1 libopenal-data libopenal1 libopencv-core2.4 libopencv-imgproc2.4
  libopenjp2-7 libpgm-5.1-0 libpostproc54 libraw1394-11 librubberband2 libshine3 libsnappy1
  libsodium13 libssh-gcrypt-4 libswresample2 libswscale4 libtwolame0 libva-drm1 libvdpau1
  libwavpack1 libx265-87 libzmq3 libzvbi-common libzvbi0
0 upgraded, 46 newly installed, 0 to remove and 0 not upgraded.
Need to get 26,0 MB of archives.
After this operation, 48,1 MB of additional disk space will be used.
Get:1 http://ftp.debian.org/debian/ jessie/main libbluray1 armhf 1:0.6.2-1 [106 kB]
Get:2 http://security.debian.org/ jessie/updates/main libopenjp2-7 armhf 2.1.0-2+deb8u1 [105 kB]
Get:3 http://ftp.debian.org/debian/ jessie/main libcaca0 armhf 0.99.beta19-2 [336 kB]
Get:4 http://ftp.debian.org/debian/ jessie/main libva-drm1 armhf 1.4.1-1 [12,4 kB]
Get:5 http://ftp.debian.org/debian/ jessie/main libvdpau1 armhf 0.8-3+deb8u2 [31,5 kB]
Get:6 http://ftp.debian.org/debian/ jessie-backports/main libavutil55 armhf 7:3.2-2~bpo8+2 [223 kB]
Get:7 http://ftp.debian.org/debian/ jessie/main libshine3 armhf 3.1.0-2.1 [23,7 kB]
Get:8 http://ftp.debian.org/debian/ jessie/main libsnappy1 armhf 1.1.2-3 [39,3 kB]
Get:9 http://ftp.debian.org/debian/ jessie-backports/main libswresample2 armhf 7:3.2-2~bpo8+2 [79,0 kB]
Get:10 http://ftp.debian.org/debian/ jessie/main libtwolame0 armhf 0.3.13-1.1 [47,7 kB]
Get:11 http://ftp.debian.org/debian/ jessie/main libwavpack1 armhf 4.70.0-1 [70,0 kB]
Get:12 http://ftp.debian.org/debian/ jessie-backports/main libx265-87 armhf 2.0-4~bpo8+1 [475 kB]
Get:13 http://ftp.debian.org/debian/ jessie/main libzvbi-common all 0.2.35-3 [61,3 kB]
Get:14 http://ftp.debian.org/debian/ jessie/main libzvbi0 armhf 0.2.35-3 [241 kB]
Get:15 http://ftp.debian.org/debian/ jessie-backports/main libavcodec57 armhf 7:3.2-2~bpo8+2 [4109 kB]
Get:16 http://ftp.debian.org/debian/ jessie-backports/main libchromaprint1 armhf 1.3.2-2~bpo8+1 [26,3 kB]
Get:17 http://ftp.debian.org/debian/ jessie/main libraw1394-11 armhf 2.1.0-3 [41,9 kB]
Get:18 http://ftp.debian.org/debian/ jessie/main libdc1394-22 armhf 2.2.3-1 [96,6 kB]
Get:19 http://ftp.debian.org/debian/ jessie/main libenca0 armhf 1.16-1 [54,9 kB]
Get:20 http://ftp.debian.org/debian/ jessie/main libfftw3-double3 armhf 3.3.4-2 [442 kB]
Get:21 http://ftp.debian.org/debian/ jessie/main libglu1-mesa armhf 9.0.0-2 [129 kB]
Get:22 http://ftp.debian.org/debian/ jessie/main libiec61883-0 armhf 1.2.0-0.2 [28,4 kB]
Get:23 http://ftp.debian.org/debian/ jessie/main libopenal-data all 1:1.15.1-5 [11,5 kB]
Get:24 http://ftp.debian.org/debian/ jessie/main libopenal1 armhf 1:1.15.1-5 [153 kB]
Get:25 http://ftp.debian.org/debian/ jessie/main libopencv-core2.4 armhf 2.4.9.1+dfsg-1+deb8u1 [653 kB]
Get:26 http://ftp.debian.org/debian/ jessie/main libopencv-imgproc2.4 armhf 2.4.9.1+dfsg-1+deb8u1 [549 kB]
Get:27 http://ftp.debian.org/debian/ jessie/main librubberband2 armhf 1.8.1-6 [76,2 kB]
Get:28 http://ftp.debian.org/debian/ jessie/main libsodium13 armhf 1.0.0-1 [149 kB]
Get:29 http://ftp.debian.org/debian/ jessie/main libssh-gcrypt-4 armhf 0.6.3-4+deb8u2 [107 kB]
Get:30 http://ftp.debian.org/debian/ jessie/main libpgm-5.1-0 armhf 5.1.118-1~dfsg-1 [154 kB]
Get:31 http://ftp.debian.org/debian/ jessie/main libzmq3 armhf 4.0.5+dfsg-2+deb8u1 [421 kB]
Get:32 http://ftp.debian.org/debian/ jessie/main libass5 armhf 0.10.2-3 [52,7 kB]
Get:33 http://ftp.debian.org/debian/ jessie/main libavc1394-0 armhf 0.5.4-2 [21,2 kB]
[... 331 lines cut ...]

 [Ok] cron
 [Ok] transmission-daemon
 [Ok] proftpd
 [Ok] mysql
 [Ok] noip2
 [Ok] avahi-daemon
 [Ok] syncthing

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Optimize and configure software
 Please wait...

 [Info] Applying DietPi optimizations and configurations for oDroid XU3/4 (armv7l)
 [Info] Please wait...

 #               processing | D i e t  P i             #

sudo: php: command not found
cat: /var/www/nextcloud/config/config.php: No such file or directory
sed: can't read /var/www/nextcloud/config/config.php: No such file or directory
/DietPi/dietpi/dietpi-software: line 5884: php: command not found
/DietPi/dietpi/dietpi-software: line 5884: / 1024 / 1024: syntax error: operand expected (error token is "/ 1024 / 1024")
`

Fourdee, 2016-11-24:
@joaofl 

Thanks for the report.

Need to get a bit more info on installed packages, can you paste results of following commands please:
```
dpkg -l | awk '{print $1,$2}'
cat /DietPi/dietpi/.installed
```

joaofl, 2016-11-24:
@Fourdee 

Follows:

dpkg -l | awk '{print $1,$2}'
`Desired=Unknown/Install/Remove/Purge/Hold 
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required
||/ Name
+++-====================================-=================================-============-========================================================================================================== 
ii acl
ii adduser
ii ajenti
ii alsa-base
ii alsa-utils
ii apt
ii apt-show-versions
ii apt-utils
ii aspell
ii aspell-en
ii autotools-dev
ii avahi-daemon
ii base-files
ii base-passwd
ii bash
ii bash-completion
ii bc
ii bind9-host
ii binutils
ii bsdmainutils
ii bsdutils
ii bzip2
ii ca-certificates
ii ca-certificates-java
ii console-common
ii console-data
ii console-setup
ii console-setup-linux
ii coreutils
ii cpio
ii cpp
ii cpp-4.9
ii crda
ii cron
ii curl
ii dash
ii dbus
ii dbus-x11
ii dc
ii dconf-gsettings-backend:armhf
ii dconf-service
ii deb-multimedia-keyring
ii debconf
ii debconf-i18n
ii debconf-utils
ii debian-archive-keyring
ii debianutils
ii dialog
ii dictionaries-common
ii diffutils
ii dmidecode
ii dmsetup
ii dosfstools
ii dphys-swapfile
ii dpkg
ii e2fslibs:armhf
ii e2fsprogs
ii emacsen-common
ii ethtool
ii fake-hwclock
[... 709 lines cut ...]
aSOFTWARE_INSTALL_STATE[111]=0
aSOFTWARE_INSTALL_STATE[112]=0
aSOFTWARE_INSTALL_STATE[113]=0
aSOFTWARE_INSTALL_STATE[114]=0
aSOFTWARE_INSTALL_STATE[115]=0
aSOFTWARE_INSTALL_STATE[116]=0
aSOFTWARE_INSTALL_STATE[117]=0
aSOFTWARE_INSTALL_STATE[118]=0

#DietPi Choice System: SSH Server
INDEX_SSHSERVER_CURRENT=-2
INDEX_SSHSERVER_TARGET=-2

#DietPi Choice System: File Server
INDEX_FILESERVER_CURRENT=-1
INDEX_FILESERVER_TARGET=-1

#DietPi Choice System: Logging
INDEX_LOGGING_CURRENT=-1
INDEX_LOGGING_TARGET=-1

#DietPi Preference System: Webserver base
INDEX_WEBSERVER_CURRENT=-2
INDEX_WEBSERVER_TARGET=-2
`

Fourdee, 2016-11-24:
Notes to self:
dpkg, no webserver installed. lighttpd [84] claims installed.

@joaofl 
For some reason, lighttpd packages are missing from dpkg. Although, DietPi has registered it as previously installed.

#### Please try reinstalling LLMP stack with:
This will reinstall lighttpd packages.
```
dietpi-software reinstall 80
```

Once completed, you can re-attempt nextcloud installation with:
```
dietpi-software install 114
```

joaofl, 2016-11-24:
Thanks @Fourdee
Just tried what you said. By reinstalling  80, I get [Info] 80: 

```
LLMP is not currently installed
 [Info] The program must be installed, before reinstall can be used
 [Ok] No changes applied for: LLMP
```

Then, I install 114. It deflates the zip file successfully, but it gives the following error in the end:

```
sudo: php: command not found
cat: /var/www/nextcloud/config/config.php: No such file or directory
sed: can't read /var/www/nextcloud/config/config.php: No such file or directory
/DietPi/dietpi/dietpi-software: line 5884: php: command not found
/DietPi/dietpi/dietpi-software: line 5884: / 1024 / 1024: syntax error: operand expected (error token is "/ 1024 / 1024")
```

joaofl, 2016-11-24:
I was thinking:

LLMP uses sqlite.. maybe I need the LLMP setup instead (with mysql, which is installed by nexcloud automatically)

Fourdee, 2016-11-24:
@joaofl 

> I was thinking:
LLMP uses sqlite.. maybe I need the LLMP setup instead (with mysql, which is installed by nexcloud automatically)

Ah yes, so we can install LLMP with:
```
dietpi-software install 80
```

joaofl, 2016-11-24:
@Fourdee  Done!

Had to remove LLMP and install LLMP. Installed 114 and it works. Thanks for the support. It was probably some old messy configuration of mine, for some reason I can't recall.

closed 2016-11-24 by joaofl

Fourdee, 2016-11-24:
@joaofl 

No worries, great to hear its resolved 👍