MichaIng/DietPi #616: DietPi | New website

Fourdee, 2016-11-25:
As title

Fourdee, 2016-11-25:
Preview:
[image]
[image]

bhaveshgohel, 2016-11-26:
Awesome @Fourdee :+1: 
Can we have the Link for Preview? :smiley:

WolfganP, 2016-11-26:
Nice design. Radioactive background :-)

Fourdee, 2016-11-26:
@bhaveshgohel 

Hoping to get it finished today, will post when its up :)

Fourdee, 2016-11-26:
@bhaveshgohel @WolfganP 
Nearly done, you can preview it here: http://dietpi.com/dietpi.com_v2/

Welcome any feedback and testing if you guys have any :)

WolfganP, 2016-11-26:
Nice one. I notice a minor misspelling at the top buttons and text below: "Comminuty"

Fourdee, 2016-11-27:
Ok, new site is live (ctrl+f5) http://dietpi.com

Fourdee, 2016-11-30:
- 🈴 Update dietpi desktop icons with new logo

k-plan, 2016-12-01:
@Fourdee 

fresh and  colourful looking new website.  👍 

... and a fitting new forum style! 😃 

Great job.

> Update dietpi icons with new logo

Thinking, you have leave one in old style: https://github.com/Fourdee/DietPi#dietpi

Fourdee, 2016-12-01:
@k-plan 

> Thinking, you have leave one in old style here: https://github.com/Fourdee/DietPi#dietpi

Thanking you 👍. The one that got away 😄 lol

bhaveshgohel, 2016-12-03:
@Fourdee Awesome!

Just one suggestion to decrease the load time of site:
### Compress all Images using [mozjpeg](https://imageoptim.com/mozjpeg ) :
Image  - 311KB [Original]
[image]

Image - 35.8KB! [compressed]
[image]

Fourdee, 2016-12-03:
@bhaveshgohel 

Good spot and thanks for link 👍. I'll reopen and take another pass on whole site.

Fourdee, 2016-12-04:
@bhaveshgohel 
Down from 3.5MB: http://tools.pingdom.com/fpt/bXGWGd/http://webcache.googleusercontent.com/search?q=cache:oFg0pV5BXHYJ:dietpi.com/+&cd=1&hl=en&ct=clnk&gl=uk

To 1.1MB: https://tools.pingdom.com/#!/cAAHDS/dietpi.com

Thanks @bhaveshgohel, awesome 👍

closed 2016-12-04 by Fourdee