MichaIng/DietPi #621: DietPi-Config | Add support for Odroid Remote

Fourdee, 2016-12-02:


Fourdee, 2016-12-06:
draft code/testing status:
- 🈯️ C2

Fourdee, 2016-12-08:
Now, where to put this in DietPi-Config...

Fourdee, 2016-12-09:
Display Options at the moment. DietPi-Config main menu needs a reshuffle at some point:
[image]
[image]

closed 2016-12-09 by Fourdee

k-plan, 2016-12-10:
@Fourdee 

> Now, where to put this in DietPi-Config...

[image]

hmm .. other option is to put it under `DietPi - Advanced Options`

[image]

It is rather a input option like a keyboard. But I don't like to put it under `DietPi - Language and Regional Options ` 👎 

> DietPi-Config main menu needs a rethink and reshuffle at some point:

 Yes, DietPi is raise up like the wind in the last year. So many new devices and supported hardware in the past months. 😃

Fourdee, 2016-12-11:
@k-plan 

> It is rather a input option like a keyboard. But I don't like to put it under DietPi - Language and Regional Options 👎

Yep, I think we could do with a new menu option called?:
- Peripheral Options
 - Remote controls
 - LCD Addons
 - Bluetooth (maybe fits below)

Networking Options NAS/MISC, could change to?
- Connectivity Options
 - Bluetooth

Or, we could do a?:
- Hardware Options
 - Display Options
 - Audio Options
 - Network Options

So many options and possibilities, will make a new GiT ticket for this: https://github.com/Fourdee/DietPi/issues/639

k-plan, 2016-12-12:
@Fourdee 

puh ... it's really difficult. 

I don't know, where your journey will end up ...

For the moment:  `dietpi-config`

---------------

1 - System Options
 * Locale 
 * Change Hostname
 * User Data Location
 * Timezone
 * Time sync mode
 * NTP Server
 * Swapfile size 
 * Log System 
 * Apt Mirror
 * Change Root Password 
 * Update firmware
 

2 - Performance Options 
 * Overclocking 
 * Change CPU Governor
 * CPU Throttle Up
 * Ondemand Sample Rate
 * Ondemand Down Factor
 * CPU Max Freq Limit
 * ARM Initial Turbo
 * ARM Temp Limit
 * SD-Card Overclocking
 
                
3 - Display Options
 * Change Resolution 
 * GPU/RAM Memory Split 
 * HDMI Boost
 * Overscan
 * LCD Panel addon
 
                    
4 - Audio Options 
  * Soundcard 
 
                     
5 - Peripheral Options
 * Keyboard
 * Serial console
 * Remote controls
 * Max USB current
 * I2c state
 * I2c frequency
 * RPi Camera
 * RPI Camera led 
                   
                 
6 - Network Options
 * Ethernet 
 * WiFi
 * Bluetooth 
 * IPv6 
 * Proxy
 * Test
 
     
7 - Connectivity Options 
 * SSH Server
 * File Server
 * Samba Client
 * FTP Client
 * NFS Client

8 - Advanced Options 
 * ???
 
  
9 - AutoStart Options 
 * Console:   Manual 
 * Console:   Automatic
 * Kodi
 * Desktops
 * DietPi-Cloudshell
 * CAVA 
 * RetroPie
 * Uae4ARM: (Fast boot)
 * Uae4ARM
 * DXX-Rebith
 * OpenTyrian
 

10 - Tools  
 * Benchmarks
 * Stress Test

----

Better in `dietpi-software` :
* NoIp
                 

------------

But it's not fix or final.
Had to sleep some night and rethink this.   

Have a look and give an opinion.