MichaIng/DietPi #624: DietPi-Software | RealVNC (for direct graphics capture)

Fourdee, 2016-12-04:


Fourdee, 2016-12-04:
> Installing Raspberry Pi license (VNC Personal)
Please note that this is only valid for non-commercial use on Raspberry Pi devices.

RPi only 👎

Fourdee, 2016-12-04:
online doc: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=4149#p4149

Fourdee, 2016-12-04:
Completed.

closed 2016-12-04 by Fourdee

opticyclic, 2016-12-08:
I did this manually and when I typed `vncserver` I got the error:

    Error: execvp(xauth): No such file or directory (2) [System]

I think you also need to include `apt-get install xauth`

Fourdee, 2016-12-08:
@opticyclic 
xauth is installed with XserverXorg: https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software#L4986

XserverXorg gets triggered for all software that requires a desktop.

Lets check your install file, whats results of:
```
dietpi-software list | grep '[[:space:]]=2'
```