MichaIng/DietPi #625: XU4 Kodi | No sound

Fourdee, 2016-12-04:


Fourdee, 2016-12-06:
Requested Bug report to check system config files.

Fourdee, 2016-12-12:
No response, moving out of milestone.

Fourdee, 2016-12-17:
No response. Marking as closed.

p3p1, 2017-04-25:
I don't know if it's possible open again this ticket, but I have this problem related to audio over HDMI using kodi installed through `dietpi-software`. Can you investigate?

Fourdee, 2017-04-25:
@p3p1 

Thanks for the report.

I'd like to check your system settings/installed packages. Please can you send us a bug report?
http://dietpi.com/phpbb/viewtopic.php?f=8&t=499

p3p1, 2017-04-25:
I will send you the report in couple of hours.

p3p1, 2017-04-25:
Bug report send. Reference code: 3b7de7c5-6117-4842-8212-1a36256df7f9-0

Fourdee, 2017-05-03:
@p3p1 

Apologies, will put this in milestone and take a look.

Fourdee, 2017-05-04:
New password prompts during install. Seems they broke automated password entries:
```
root@DietPi:~# pihole -a -p dietpi
Enter New Password (Blank for no password):
```

Fourdee, 2017-05-04:
#### Tests/notes:
- 🈯️ Fresh install of XU4 v145 image + Kodi install only + autostart kodi = sound ok
- Check bug report logs/settings

Fourdee, 2017-05-04:
@p3p1 

Most likley a pulse audio setting/issue, but lets test basic sound. Please run the following (not from desktop):
```
speaker-test -c2
```

Press CTRL+C to stop. Let me know if you hear sound on HDMI.

p3p1, 2017-05-07:
Apologies for late response. No sound unfortunately on HDMI. I think it's not a hardware problem: HDMI works on other device and TV is at maximum volume.

Fourdee, 2017-05-07:
@p3p1 

> I think it's not a hardware problem: HDMI works on other device and TV is at maximum volume.

Its very bizarre, there is an issue with no sound in XFCE4, however, Kodi replicates sound fine each test for me. We've also had no other reports of this, so you could be right.

To verify DietPi is not the cause, you could try a fresh installation, only installing Kodi, then test sound again.

Failing that, possible hardware issue (try another XU4 image to verify), or, incompatibility with TV, or, faulty HDMI cable.

p3p1, 2017-05-30:
I can confirm that kodi works perferctly under LXDE, in XFCE I had a problem about the audio but I think it's connected to fact that the XFCE is started with root user. I still have a problem but I can't understand the reason: when kodi is launched from LXDE works well, but any other service (ssh,home assistant,nginx) is stucked.  I will try with a fresh install in the weekend.

Fourdee, 2018-02-22:
Outdated, marking as closed.

closed 2018-02-22 by Fourdee