MichaIng/DietPi #633: Can udhcpc be used as the DHCP client?

trajano, 2016-12-08:
Just wondering, is it possibel to use udhcpc as the default DHCP client?  It seems to have a smaller footprint than  isc-dhcp-client

Fourdee, 2016-12-08:
@trajano 

Seems ```udhcpc``` requires ```busybox``` which actually makes it bigger overall?:
```
root@DietPi:~# apt-get install udhcpc
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following extra packages will be installed:
  busybox
The following NEW packages will be installed:
  busybox udhcpc
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 389 kB of archives.
After this operation, 662 kB of additional disk space will be used.
Do you want to continue? [Y/n] n
Abort.

root@DietPi:~# apt-get purge isc-dhcp-client
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages will be REMOVED:
  isc-dhcp-client*
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
After this operation, 401 kB disk space will be freed.
Do you want to continue? [Y/n] n
```

trajano, 2016-12-08:
Cool and I presume we're not using busybox already.

closed 2016-12-08 by trajano