MichaIng/DietPi #635: Image | OrangePi Zero

Fourdee, 2016-12-08:


Fourdee, 2016-12-08:
```dhd``` wifi module hangs system.

Fourdee, 2016-12-09:
Image created by @aliosa27 : https://github.com/Fourdee/DietPi/pull/631

Fourdee, 2016-12-09:
@aliosa27
Sound working fine. I'll add to website and release your image 👍

Fourdee, 2016-12-09:
Still having issues with wifi module not being enabled during boot.

Workaround (tested in AP mode):
```
cat << _EOF_ > /etc/systemd/system/xradio_wlan.service
[Unit]
Description=xradio_wlan
After=network.target network-online.target

[Service]
Type=oneshot
RemainAfterExit=yes

ExecStartPre=/bin/bash -c 'modprobe xradio_wlan'
ExecStartPre=/bin/bash -c 'iwconfig wlan0 power on'
ExecStart=/bin/bash -c 'ifup wlan0'

[Install]
WantedBy=multi-user.target
_EOF_
systemctl enable xradio_wlan.service
systemctl daemon-reload
```

Fourdee, 2016-12-09:
Online doc: http://dietpi.com/phpbb/viewtopic.php?f=8&t=985&p=4262#p4262

Fourdee, 2016-12-09:
@aliosa27

Great work on this, your image is now live http://dietpi.com/download. Will mark this as closed 👍

closed 2016-12-09 by Fourdee

kutysam, 2017-02-23:
root@DietPi:~# systemctl status  systemd-modules-load.service

   Active: failed (Result: exit-code) since Thu 2017-02-23 20:00:17 SGT; 11min ago
     Docs: man:systemd-modules-load.service(8)

Feb 23 20:00:16 DietPi systemd-modules-load[160]: Module 'fuse' is builtin
Feb 23 20:00:17 DietPi systemd-modules-load[160]: Failed to insert 'xradio_wlan': Connection timed out

I don't think this 'fix' is working very well =/

Fourdee, 2017-02-23:
@kutysam 

Many thanks 👍 
Yep, the current ```xradio_wlan``` driver is somewhat "flaky". Although, its been a while since we last built the base image using ARMbian build tools. 
If I can get my OPi Zero to work (I think its completely dead, no serial, will test another adapter), i'll rebuild and test latest image/driver, hopefully its been "improved" since the last.

> Or if you don't care about the power consumption, do this from your orangepi itself.
sudo ping -s 1 -i 0.2 <orangePiWifiIp>
Basically it is a loopback test but targetted at the wifi ip. It works!

Very interesting way to keep powersaving at bay 👍

aliosa27, 2017-02-23:
I'll rebuild the base image over the next day or so. My wifi stays active so I never notice it go down.. pesky wifi

kutysam, 2017-02-23:
i discovered this 'ping' thing when i was using motion https://motion-project.github.io/

I was wondering, why did my ping become 1-2ms when i was streaming... and soon i realize, mjpeg streams really fast and multiple times within 1 second. So yeap.

Please do put in the 10second sleep function! Otherwise, it may not connect to wifi when the system comes online.

Fourdee, 2017-02-23:
@kutysam 

> Please do put in the 10second sleep function! Otherwise, it may not connect to wifi when the system comes online.

Done 👍 , i've updated the workaround service. 
Although, just bear in mind, as this is a oneshot service, it will delay boot by an additional 10 seconds. This is due to SystemD waiting for oneshot's to complete, before continuing to the next service.

> I'll rebuild the base image over the next day or so.

@aliosa27 

If you could do this, would be really grateful 👍

kutysam, 2017-06-24:
Hi! Just wondering, did you put 
sudo ping -s 1 -i 0.2 

I realize the power consumption is 0.35ma instead of 0.21ma.

Do you have the commit to github on this. Thanks!