MichaIng/DietPi #637: [Question] Does DietPi Supports Orange Pi Plus2?

bhaveshgohel, 2016-12-09:
Hello @Fourdee ,

Does DietPi Supports Orange Pi Plus2? or future plans for it?

P.S. : I didn't find it on http://dietpi.com/#download

Fourdee, 2016-12-09:
@bhaveshgohel 

From what I can read, the Plus and Plus 2 use the same .fex (https://github.com/igorpecovnik/lib/blob/master/config/fex/orangepiplus.fex#L10). Which basically means, the OrangePi Plus image should work with the OrangePi Plus 2 board. If you try it, let us know results 😄

bhaveshgohel, 2016-12-09:
@Fourdee Hahaha! I'm planing to purchase the board, if and only if it support the awesomeness of DietPi. :wink:  

Let it open for others to reply. :smiley: 
Thanks.

aliosa27, 2016-12-09:
It works on my plus 2. Although on boot I can add a small change to detect the different model and change the id so it displays that it is in fact a plus 2

Fourdee, 2016-12-14:
@aliosa27 

I think we could share the HW_MODEL and modify description:
eg: 
- Device: OrangePi Plus / Plus 2

I'll update the website,forums.

Fourdee, 2016-12-14:
Ok done. OrangePi Plus / Plus 2 share HW_MODEL and image.
Will mark as closed.

closed 2016-12-14 by Fourdee