MichaIng/DietPi #639: DietPi-Config | Rethink, reshuffle menu options

Fourdee, 2016-12-11:
>

k-plan, 2016-12-12:
@Fourdee 

puh ... it's really difficult. 

I don't know, where your journey will end up ...

Desktops? Audio Player? Multimedia? Gaming? Web-Services? NAS? IOT? Home Server? More embedded Tools and Programs? Networking? 

For the moment:  `dietpi-config`

---------------

1 - System Options
 * Locale 
 * Change Hostname
 * User Data Location
 * Timezone
 * Time sync mode
 * NTP Server
 * Swapfile size 
 * Log System 
 * Apt Mirror
 * Change Root Password 
 * Update firmware
 

2 - Performance Options 
 * Overclocking 
 * Change CPU Governor
 * CPU Throttle Up
 * Ondemand Sample Rate
 * Ondemand Down Factor
 * CPU Max Freq Limit
 * ARM Initial Turbo
 * ARM Temp Limit
 * SD-Card Overclocking
 
                
3 - Display Options
 * Change Resolution 
 * GPU/RAM Memory Split 
 * HDMI Boost
 * Overscan
 * LCD Panel addon
 
                    
4 - Audio Options 
  * Soundcard 
 
                     
5 - Peripheral Options
 * Keyboard
 * Serial console
 * Remote controls
 * Max USB current
 * I2c state
 * I2c frequency
 * RPi Camera
 * RPI Camera led 
                   
                 
6 - Network Options
 * Ethernet 
 * WiFi
 * Bluetooth 
 * IPv6 
 * Proxy
 * Test
 
     
7 - Connectivity Options 
 * SSH Server
 * File Server
 * Samba Client
 * FTP Client
 * NFS Client

8 - Advanced Options 
 * ???
 
  
9 - AutoStart Options 
 * Console:   Manual 
 * Console:   Automatic
 * Kodi
 * Desktops
 * DietPi-Cloudshell
 * CAVA 
 * RetroPie
 * Uae4ARM: (Fast boot)
 * Uae4ARM
 * DXX-Rebith
 * OpenTyrian
 

10 - Tools  
 * Benchmarks
 * Stress Test

----

Better in `dietpi-software` :
* NoIp
                 

------------

But it's not fix or final.
Had to sleep some night and rethink this.   

Have a look and give an opinion.

Fourdee, 2016-12-13:
@k-plan 
Excellent work, I really like this layout 👍 

A few things pop out that we could maybe move:
- LCD Panel addon > 5 - Peripheral Options ?
- Bluetooth > 7 - Connectivity Options?

k-plan, 2016-12-13:
@Fourdee 

> A few things pop out that we could maybe move:
> * LCD Panel addon > 5 - Peripheral Options ?
> * Bluetooth > 7 - Connectivity Options?

Yes, for sure, you can do all this. 
But let me explain my consideration.

* LCD Panel addon
- yes, it is a peripheral device. It is something like a monitor/TV, especially if you use a Deskop on your device. 
But if you will switch it on or off, you have to change your display resolution, if you use a external monitor. So you don't have to navigate to a different category

* Bluetooth
- yes, it is for connectivity, just like USB, serial, WiFi, ethernet. And yes it is peripheral device, if you don't have a RPI3 for example.
For me, it is technical a [PAN Network](https://www.techopedia.com/2/29090/networks/lanwanman-an-overview-of-network-types)  , so I have put it under network.

But it is possible, that is miss bearing of [connectivity](http://dict.leo.org/ende/index_de.html#/search=Connectivity).
Example: Bluetooth is for me more something like Media
layers ( Layer 1-3 in [OSI](https://en.wikipedia.org/wiki/OSI_model#Description_of_OSI_layers) )
Connectivity is more Host layer ( Layer 4-7 in [OSI](https://en.wikipedia.org/wiki/OSI_model#Examples) ) so I have put in here services like SSH, FTP, NFS, AFP, Samba ...

But this are only my view of things. 
Don't know if it is self-explanatory and most user can/will follow this/my logic. 😟 
Better you follow your own logic, because you have to implement, enlarge and groom it.

k-plan, 2016-12-13:
@Fourdee 

we can have a look, how other system sort and handle this, for example:

[openmediavault](http://www.openmediavault.org/screenshots.html)

http://www.openmediavault.org/wp-content/uploads/2014/07/altnavpanel.png

or [Synology Diskstation](https://www.synology.com/en-us/knowledgebase/DSM/video)

[DSM 6.1 Beta Online Demo](https://www.synology.com/en-us/dsm/live_demo)

MichaIng, 2018-04-06:
@Fourdee @k-plan 
Yey another great construction site 😃.

I would follow @k-plan's arguments about LCD Panel and Bluetooth:
- LCD Panel is also some sort of display, otherwise displays are peripheral as well, where the "Display" category also contains some settings that could fit into "Performance". Finally as there are some more settings related to device visual output that need to be configured together, it makes sense to put these into a separate category, as done above.
- Bluetooth also from my sight fits better into "Network", but it's more a feeling than knowledge based 😉. "Network" so far contains settings, that look more hardware related to me, where "Connectivity" looks more like software related aspects. Don't know it this separation somehow goes along with "media layer" and "host layer" 😆.
- Some of the "**" features are great and I was also thinking already about the one or the other, e.g. DietPi-Survey switch and Git-Branch choose. Especially the sd card overclocking worked really great on my RPi2, pushing felt performance quite a lot, but I didn't make benchmarks, it just feels more responsive (Nextcloud navigation as well) in general. But I did not yet open an issue about it, as I read that using with non U1+ sd cards it can lead to data corruption which occurs even silently, so that you just recognize it, when it's already too late. I am a bid scared to offer this to our users, when it seems so dangerous, or what you think?

About SSH, file server etc. in general. I am not yet sure if this is better placed within config or software. Maybe it makes sense to leave everything (especially those choices/preferences) that can be installed via dietpi-software there, so not have the same options in both scripts? Perhaps we could add a "Software config" category to dietpi-config, which contains subcategories for e.g. Samba etc. based on what is actually installed? This could be extended, e.g. by allowing some settings for webserver, database, Nextcloud/ownCloud or what we can think of in the future. Also dietpi-letsencrypt could be possibly integrated. But in this case it makes sense to leave those software settings within separate files/scripts to call from main dietpi-config in case.

Just had mood to add my thoughts, don't take this as a push at this topic, as usual we have more great ideas and opened feature requests as we can handle 😛.