MichaIng/DietPi #640: NanoPi Neo AIR | No WiFi

Fourdee, 2016-12-11:
>

Fourdee, 2016-12-11:
Seems to work fine for a while:
```
root@DietPi:~# iwconfig
lo        no wireless extensions.

tunl0     no wireless extensions.

wlan0     IEEE 802.11  ESSID:"FuzonWifi"
          Mode:Managed  Frequency:2.447 GHz  Access Point: E8:DE:27:4A:D0:66
          Bit Rate=26 Mb/s   Tx-Power:32 dBm
          Retry min limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Managementmode:All packets received
          Link Quality=5/5  Signal level=-57 dBm  Noise level=-91 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:1  Invalid misc:0   Missed beacon:0
```

Then connection hangs with:
```
[  374.585777] Loading kernel module for a network device with CAP_SYS_MODULE (deprecated).  Use CAP_NET_ADMIN and alias netdev-dhd instead.
[  381.693527] Loading kernel module for a network device with CAP_SYS_MODULE (deprecated).  Use CAP_NET_ADMIN and alias netdev-dhd instead.
[  381.721118] Loading kernel module for a network device with CAP_SYS_MODULE (deprecated).  Use CAP_NET_ADMIN and alias netdev-dhd instead.
```

aliosa27, 2016-12-12:
Taking a look into this.

nydal91, 2016-12-14:
Does wifi work now?

Fourdee, 2016-12-16:
I actually think this is an issue with the ```dhd``` module that ARMbian uses. Although wifi works, the signal strength is noticeably lower than other onboard WiFi devices. Even with half-decent WiFi antenna: http://www.dx.com/p/wi-fi-antenna-ipx-female-to-sma-ky-adapter-connection-cable-w-1-13-cable-black-white-25cm-191117
```iw reg set GB``` also has no effect.

From what I remember, on the official FriendlyARM image (which our users are claiming their WiFi works on), they use ```brcm```.

nydal91, 2016-12-18:
So its best to use a wifi dongle for now?

JaggerMD, 2016-12-18:
On my Neo Air only wlan1 was recognized.
Changing wlan0/1 in /etc/udev/rules.d/70-persistent-net.rules did the trick.
From this point wifi is starting fine.

Fourdee, 2016-12-24:
User claims WiFi works fine on ARMbian image.

> http://dietpi.com/phpbb/viewtopic.php?f=9&t=886&start=10#p4330

note to self:
- re-test and check hciattach? service.
- Try using ```bcm....``` module and their hciattach? script on FriendlyARM image. Which works flawlessly.

Fourdee, 2016-12-26:
Been testing with ARMbian 4.9 kernel. Firmware is needed, but looking good so far:
https://github.com/igorpecovnik/lib/commit/8f4e8e25e425169c90d12c67d21e5b84f5cdb231#commitcomment-20297066

Fourdee, 2016-12-26:
WiFi works flawlessly.

Issue with cpu govs
```
cat: /sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq: No such file or directory
```

thomaspfeiffer-git, 2016-12-28:
Hi Fourdee,
what did you do to get wifi working?
(Will there be a new image available soon?)

Thanks and best regards
Thomas

Fourdee, 2016-12-28:
@thomaspfeiffer-git 
So it looks like a new image will be needed to fix WiFi at the moment (using the 4.9 kernel). When we release that image depends on if we can resolve the issues with cpu govs, and, no other issues crop up: https://github.com/Fourdee/DietPi/issues/640#issuecomment-269232618

nydal91, 2016-12-30:
get the wifi to work with nmtui. thanks for a stable build! the ubuntu core was unstable when I was using it on nano pi air.

Fourdee, 2016-12-30:
@thomaspfeiffer-git @nydal91 

With thanks to @zador-blood-stained from ARMbian (https://github.com/igorpecovnik/lib/issues/592#issuecomment-269524792), 4.9 kernel image is now ready for testing:
http://dietpi.com/downloads/testing/DietPi_NanoPiNEOAir-armv7-(Jessie).7z

Personally tested WiFi (and system in general), seems much more stable. If others can test image above and let us know results, i'd be grateful.

thomaspfeiffer-git, 2016-12-30:
Great, wifi works fine now, thanks very much! :-)

Fourdee, 2016-12-30:
@thomaspfeiffer-git 
Excellent 👍 

I'll set this to live image.

Fourdee, 2016-12-30:
Completed.

closed 2016-12-30 by Fourdee