MichaIng/DietPi #642: Libresonic + FFmepg (libfdk_acc)

Dalto1, 2016-12-11:
I installed libresonic from source on my dietpi-server. I think this could be usefull to add into the software bank. I works great. Anyways, i got a little problem when im trying to read FLAC songs. libfdk_acc seems to be needed. And im not able to configurate ffmpeg for it. i read issues in https://github.com/Fourdee/DietPi/issues/84. Dont know if it is related. Could you help me to configurate it? do you compil ffmpeg in the installation of it? Thanks,

Gabriel

Fourdee, 2016-12-12:
@Dalto1 

Hi, which device are you running DietPi on, and, did you install FFMPEG from ```dietpi-software``` ?

You can check installed with (green ```=2``` if installed):
```
dietpi-software list | grep 'ffmpeg:'
```

For ARMv6 (RPi 1/zero), we pre-compiled debs which installs:
- ffmpeg_2.7.2
- fdk-aac_0.1.4

For ARMv7 RPi, we use the ```deb-multimedia.org``` repo to install FFMEG. ```libfdk_acc``` may be available, try running this to find the package:
```
apt-cache search libfdk
```

Fourdee, 2016-12-12:
@Dalto1 

> libfdk_acc seems to be needed. And im not able to configurate ffmpeg for it.

Lets check installed packages, please paste results:
```
dpkg -l | grep libfdk
dpkg -l | grep ffmpeg
```

Dalto1, 2016-12-12:
I am on pine64
apt-cache search libfdk* | grep 'runtime'      ---> show nothing
dpkg -l | grep libfdk
dpkg -l | grep ffmpeg

--->
ii  libfdk-aac-dev:arm64                  0.1.4-2                           arm64        Fraunhofer FDK AAC Codec Library - development files
ii  libfdk-aac1:arm64                     0.1.4-2                           arm64        Fraunhofer FDK AAC Codec Library - runtime files
ii  ffmpeg                                7:3.2-2~bpo8+2                    arm64        Tools for transcoding, streaming and playing of multimedia files

i installed the .deb from https://packages.debian.org/search?keywords=libfdk-aac-dev
in the testing branch

Fourdee, 2016-12-12:
@Dalto1 

> libfdk-aac1

Looks installed fine.

Lets check libflac is installed:
```
dpkg -l | grep libflac
```

Question:
- Did you install FFMPEG from ```dietpi-software```?

Dalto1, 2016-12-12:
ii  libflac8:arm64                        1.3.0-3                           arm64        Free Lossless Audio Codec - runtime C library

and yes i installed ffmpeg from dietpi-software

Fourdee, 2016-12-12:
@Dalto1 
Thanks.
I'll take a look at libresonic, add it to DietPi-Software and test flac, should have a solution for you within a few days.

Fourdee, 2016-12-12:
Git Ticket: https://github.com/Fourdee/DietPi/issues/644

Dalto1, 2016-12-12:
Thanks you master! :D

closed 2016-12-12 by Dalto1

Fourdee, 2016-12-13:
@Dalto1 
If you installed FFMPEG from dietpi-software, this should do it:
```
ln -fs /usr/bin/ffmpeg /var/libresonic/transcode
```

[image]

https://github.com/Fourdee/DietPi/issues/644

Dalto1, 2016-12-13:
This is working like a charm, thanks.

Fourdee, 2016-12-14:
@Dalto1 

Excellent 👍