MichaIng/DietPi #644: DietPi-Software | Libresonic

Fourdee, 2016-12-12:
> https://github.com/Libresonic/libresonic

Fourdee, 2016-12-13:
Tomcat: https://www.linode.com/docs/websites/frameworks/apache-tomcat-on-ubuntu-16-04

Fourdee, 2016-12-13:
Tomcat:
When using Oracle JDK:
```
root@DietPi:~# systemctl status tomcat8.service -l
● tomcat8.service - LSB: Start Tomcat.
   Loaded: loaded (/etc/init.d/tomcat8)
   Active: failed (Result: exit-code) since Tue 2016-12-13 14:51:03 GMT; 10s ago

Dec 13 14:51:03 DietPi systemd[1]: Starting LSB: Start Tomcat....
Dec 13 14:51:03 DietPi tomcat8[4151]: no JDK or JRE found - please set JAVA_HOME ... failed!
Dec 13 14:51:03 DietPi systemd[1]: tomcat8.service: control process exited, code=exited status=1
Dec 13 14:51:03 DietPi systemd[1]: Failed to start LSB: Start Tomcat..
Dec 13 14:51:03 DietPi systemd[1]: Unit tomcat8.service entered failed state.
```

Fourdee, 2016-12-13:
```
#wget https://jenkins.zifnab.net/job/libresonic/lastStableBuild/artifact/libresonic-booter/target/libresonic-booter-jar-with-dependencies.jar -O package.jar
#rm package.jar

wget https://jenkins.zifnab.net/job/libresonic/lastStableBuild/artifact/libresonic-main/target/libresonic.war -O /var/lib/tomcat8/webapps/libresonic.war

mkdir /var/libresonic
chown tomcat8:tomcat8 /var/libresonic/

```

Fourdee, 2016-12-13:
Hmm, fresh test installs on ARMv7/64:
> Notice: Transcoders does not appear to be installed.
Click Help button for more information.

Fourdee, 2016-12-14:
Well, i'am stumped. Trans-coding no longer works: 
- I've either some something really stupid (wouldn't be the 1st time)
- Trans-coding just doesn't work with 'Last successful build of Libresonic'

[image]

```
root@DietPi:~# ls -lha /var/libresonic/transcode
lrwxrwxrwx 1 tomcat8 tomcat8 15 Dec 14 17:56 /var/libresonic/transcode -> /usr/bin/ffmpeg

root@DietPi:~# /usr/bin/ffmpeg
ffmpeg version 3.2-2~bpo8+2 Copyright (c) 2000-2016 the FFmpeg developers
  built with gcc 4.9.2 (Debian/Linaro 4.9.2-10)

#Lets try permissions...
root@DietPi:~# chown tomcat8:tomcat8 /usr/bin/ffmpeg^C
root@DietPi:~# ls -lha /usr/bin/ffmpeg
-rwxr-xr-x 1 tomcat8 tomcat8 247K Nov 11 15:39 /usr/bin/ffmpeg

```

Fourdee, 2016-12-21:
Libresonic now failing to start with: https://jenkins.zifnab.net/job/libresonic/lastStableBuild/artifact/libresonic-main/target/libresonic.war
```
INFO: Deploying web application archive /var/lib/tomcat8/webapps/libresonic.war
[2016-12-21 12:51:58,498] ERROR org.springframework.web.context.ContextLoader - Context initialization failed
org.springframework.beans.factory.parsing.BeanDefinitionParsingException: Configuration problem: Unable to locate Spring NamespaceHandler for XML schema namespace [http://www.springframework.org/schema/security]
Offending resource: ServletContext resource [/WEB-INF/applicationContext-security.xml]
```

Fourdee, 2016-12-21:
Dropping this from ```dietpi-software``` as its too unstable, unreliable. 
Recommend SubSonic as an alternative (pretty much exactly the same program): http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=30#p213

I'll convert the current installation to ```tomcat8```

Fourdee, 2016-12-21:
Completed.

closed 2016-12-21 by Fourdee

Dalto1, 2016-12-23:
Dont know if this can help you, but i followed this tutorial to achieve the installation of libresonic http://dsync.blogspot.ca/2016/10/tutorial-complete-tutorial-on.html
:) thanks, Gab