MichaIng/DietPi #646: Softwares

Dalto1, 2016-12-13:
Not sure if this could help, but there is links to install serveral softwares that could interest someone : 

Couchpotato : http://www.htpcguides.com/install-couchpotato-raspberry-pi-raspbian/
Sonarr : http://www.htpcguides.com/install-sonarr-raspberry-pi-mono-310/
Headphones : http://www.htpcguides.com/install-headphones-raspberry-pi-usenet-music/
Mylar : http://www.htpcguides.com/install-mylar-raspberry-pi-usenet-torrent-comics/
Lazylibrarian : http://www.htpcguides.com/install-lazylibrarian-raspberry-pi-for-usenet-torrent-ebooks/
Jackett : http://www.htpcguides.com/install-jackett-raspberry-pi-for-custom-torrents-in-sonarr/
Htpc Manager : http://www.htpcguides.com/install-htpc-manager-ubuntu-linux/
TVheadend: https://tvheadend.org/projects/tvheadend/wiki/Building

Thanks

Dalto1, 2017-01-01:
I installed the last ffmpeg for armhf with this repo
deb http://ftp.debian.org/debian jessie-backports main

and i installed tvheadend with this repo
deb https://dl.bintray.com/djbenson/deb wheezy unstable

key for djbenson :
sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 379CE192D401AB61

dont forget to install this
sudo apt-get install apt-transport-https

the-researcher, 2017-01-12:
Just to clarify/update this thread:

Couchpotato will be getting some competition:

Sonarr (which specializes in organizing TV viewing) has been forked to create a new project: Radarr.

Radarr has the intention of competing with Couchpotato for organizing movies. It may be beneficial to also consider Radarr for integration as an install package.

https://github.com/Radarr/Radarr

Dalto1, 2017-01-12:
There is also Watcher if you wanna check. 

https://github.com/nosmokingbandit/watcher

closed 2017-06-16 by Dalto1