MichaIng/DietPi #647: DietPi-Software | Update Owncloud to v11

LexiconCode, 2016-12-14:
There's quite a few changes this time around from increased security and scalability optimizations.

Fourdee, 2016-12-14:
@LexiconCode 

Thanks, we'll get this updated for v140 👍

Fourdee, 2016-12-14:
```
Nextcloud is not installed - only a limited number of commands are available
Error while trying to create admin user: Failed to connect to the database: An exception occured in driver: SQLSTATE[42000] [1049] Unknown database 'nextcloud'
 ->
```
Looks like we now need to create the MySQL DB.

Fourdee, 2016-12-14:
Tests:
- 🈯️ ARM64
- ARMv7

[image]

Fourdee, 2016-12-14:
Completed. V11 for new installations only.

closed 2016-12-14 by Fourdee