MichaIng/DietPi #650: Odroid C2 - GPIO support

wreyniers, 2016-12-16:
I have a PIR sensor I would love to use on DietPi. I took a swing at installing wiringPi as they now support the Odroid C2 but I ended up in a dependency loop.  Essentially my goal is to turn off my TV over CEC if it doesn't detect motion for 15 minutes, then turn it back on when someone walks in the room. 

Any chance we could have this for our C2?

Here's the wiringPi Github repo:
https://github.com/hardkernel/wiringPi

Fourdee, 2016-12-17:
- 🈯️ C2
```
root@DietPi:~# gpio -v
gpio version: 2.33
Copyright (c) 2012-2014 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Hardkernel ODROID Details:
  Type: ODROID-C2, Revision: 1, Memory: 2048MB, Maker: Hardkernel
```

Fourdee, 2016-12-17:
Completed.

closed 2016-12-17 by Fourdee

wreyniers, 2016-12-17:
Wow. What a quick turn around! Do I just have to run apt-get upgrade to get
this on my dietpi?

Fourdee, 2016-12-18:
It will be available in dietpi-software when v140 is released. Hopefully before Xmas [😊]

Once v140 is released, you can update with:
dietpi-update

wreyniers, 2016-12-22:
Awesome! Thanks, you da best!

TominJoseph18, 2017-01-30:
Hi.

I want to send interrupts through my gpio pins. I am using the wiringPi Library for this purpose. When a code to wait for interrupt is run ,the ethernet connection to my laptop breaks and it says broken pipe error.

Could someone please help me to execute a command X in C when an interrupt event occurs which also prevents any kind of switch debouncing and transient voltages.

Thank you.