MichaIng/DietPi #651: Plex install fails on Odroid C2

maximm07, 2016-12-17:
Installed fresh image v.139 on Odroid C2. Plex installation afterwards fails with error;

 Apt-get has failed:                                                │
                                                 │  - Apt string = binutils:armhf plexmediaserver-installer:armhf     │
                                                 │  - Error code = 100                                                │
                                                 │  - Software Title = Plex Media Server: web interface media         │
                                                 │ streaming server                                                   │
                                                 │                                                                    │
                                                 │ Please report this issue to DietPi so it can be investigated.      │
                                                 │                                                                    │
                                                 │ DietPi-Software will now exit.

Fourdee, 2016-12-17:
@maximm07 
Thanks for the report, looks like dep issues, will take a look.

Fourdee, 2016-12-17:
```
root@DietPi:~# apt-get install binutils:armhf plexmediaserver-installer:armhf  libc6:armhf zlib1g:armhf libc6

The following packages have unmet dependencies:
 libc6 : Breaks: libc6:armhf (!= 2.19-18+deb8u7) but 2.19-18+deb8u6 is to be installed
 libc6:armhf : Breaks: libc6 (!= 2.19-18+deb8u6) but 2.19-18+deb8u7 is to be installed
E: Unable to correct problems, you have held broken packages.
```

Fourdee, 2016-12-17:
Ok, lets try a workaround/fix:
```
wget http://ftp.us.debian.org/debian/pool/main/g/glibc/libc6_2.19-18+deb8u6_arm64.deb -O package.deb;dpkg -i package.deb
```

```
root@DietPi:~# apt-get install plexmediaserver-installer:armhf libc6:armhf
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following extra packages will be installed:
  gcc-4.9-base:armhf libgcc1:armhf
Suggested packages:
  glibc-doc:armhf locales:armhf
The following NEW packages will be installed:
  gcc-4.9-base:armhf libc6:armhf libgcc1:armhf plexmediaserver-installer:armhf
0 upgraded, 4 newly installed, 0 to remove and 0 not upgraded.
Need to get 4,167 kB of archives.
After this operation, 8,764 kB of additional disk space will be used.
```

Fourdee, 2016-12-17:
@maximm07 
Please run the following commands to resolve:
Workaround/fix, :
```
wget http://ftp.us.debian.org/debian/pool/main/g/glibc/libc6_2.19-18+deb8u6_arm64.deb -O package.deb
dpkg -i package.deb
rm package.deb

#Reinstall plex
dietpi-software install 42
```

Fourdee, 2016-12-17:
Looks like a common issue with the Debian ARM64 repo at the moment. ```libc6``` is ```u7``` but most packages still depend on specific version ```u6```:
```
Depends: libc6 (= 2.19-18+deb8u6) but 2.19-18+deb8u7 is to be installed
```
http://dietpi.com/phpbb/viewtopic.php?f=11&t=1000#p4325

maximm07, 2016-12-17:
working fine now...tnx.

Fourdee, 2016-12-17:
Resolved by: https://github.com/Fourdee/DietPi/issues/651

closed 2016-12-17 by Fourdee