MichaIng/DietPi #653: Debian ARM64 repo | libc6 version requirments breaking most installations

Fourdee, 2016-12-17:
As per https://github.com/Fourdee/DietPi/issues/651#issuecomment-267763640

Workaround:
```
wget http://ftp.us.debian.org/debian/pool/main/g/glibc/libc6_2.19-18+deb8u6_arm64.deb -O package.deb
dpkg -i package.deb
rm package.deb
```

Fourdee, 2016-12-17:
Only seems to effect C2, Pine A64 install u6.
```
root@DietPi:~# apt-get install libc6 --reinstall
Reading package lists... Done
Building dependency tree
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 0 not upgraded.
Need to get 4,114 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://ftp.debian.org/debian/ jessie/main libc6 arm64 2.19-18+deb8u6 [4,114 kB]
Fetched 4,114 kB in 1s (2,917 kB/s)
Preconfiguring packages ...
(Reading database ... 15665 files and directories currently installed.)
Preparing to unpack .../libc6_2.19-18+deb8u6_arm64.deb ...
Unpacking libc6:arm64 (2.19-18+deb8u6) over (2.19-18+deb8u6) ...
Setting up libc6:arm64 (2.19-18+deb8u6) ...
Processing triggers for libc-bin (2.19-18+deb8u6) ...
```

sources.list

Fourdee, 2016-12-17:
```
root@DietPi:~# rm /etc/apt/sources.list.d/meveric-*
root@DietPi:~# apt-get clean
root@DietPi:~# apt-get update

root@DietPi:~# apt-get install libc6 --reinstall
Reading package lists... Done
Building dependency tree
Reading state information... Done
Reinstallation of libc6 is not possible, it cannot be downloaded.
0 upgraded, 0 newly installed, 0 to remove and 4 not upgraded.
```

Fourdee, 2016-12-17:
Libc6 is ```u6``` on C2 image, so only existing installations may be effected.

scargill, 2016-12-17:
Hi

I just popped in at an opportune moment.

Armed with the success of DietPi on a VM, I decided to load it onto the Odroid C2.

No GIT
Then when I loaded that... I noted no MAKE

So I ran apt-get-install build-essentials... some packages could not be installed...

It was not until I ran  UPDATE....

W: Failed to fetch http://repo.mosquitto.org/debian/dists/jessie/InRelease  Unable to find expected entry 'main/binary-arm64/Packages' in Release file (Wrong sources.list entry or malformed file)

W: Failed to fetch http://download.webmin.com/download/repository/dists/sarge/Release  Unable to find expected entry 'contrib/binary-arm64/Packages' in Release file (Wrong sources.list entry or malformed file)

E: Some index files failed to download. They have been ignored, or old ones used instead.

So - by the look ot it - something up with the SOURCES (which - looking at the file LOOK sensible).

What to do next???? Help?

Fourdee, 2016-12-18:
@scargill 
Hi Peter,

> So I ran apt-get-install build-essentials... some packages could not be installed...

Once you have run the following commands here (https://github.com/Fourdee/DietPi/issues/653#issue-196227253), ```build-essential``` should install, if not, please let us know the output (paste it here).

#### Mosquito:
This is due to the repo http://repo.mosquitto.org/debian/dists/jessie/InRelease not currently providing packages for ARM64 (Odroid C2/Pine).

We can add support for 32bit ARM binaries/packages on the C2 and install ```Mosquito``` with:
```
echo -e "deb [arch=armhf] http://repo.mosquitto.org/debian jessie main" > /etc/apt/sources.list.d/mosquitto-jessie.list

dpkg --add-architecture armhf

apt-get update

apt-get install mosquitto
```

#### Webmin:
I just tried our installation which works correctly. We actually pull in the ```.deb``` rather than add the repo (http://www.webmin.com/download/deb/webmin-current.deb).
You can install Webmin with ```dietpi-software```, but make sure you remove the webmin repo you added from ```/etc/apt/sources.list.d/```

Install Webmin:
```dietpi-software install 115```

scargill, 2016-12-18:
Thanks for that - as it happens I struggled through... I do have a question over WIRINGPI - I eventually found and installed your version - and turned GPIO 1 on and off - EXCELLENT - if only the other guys would do this with their boards instead of one having to struggle with guessing which pin is which.

But could you answer this for me please.. would be very helpful..

Which pin numbers are the two PWM pins referred to as 19 and 33 in your docs because reading "gpio readall" I'm not seeing them.  Also do the two analog inputs work and does this version of WiringPi support I2c etc?

Fourdee, 2016-12-18:
@scargill 

> Which pin numbers are the two PWM pins referred to as 19 and 33 in your docs because reading "gpio readall" I'm not seeing them. Also do the two analog inputs work and does this version of WiringPi support I2c etc?

Unfortunately, i'am the wrong person to ask for GPIO stuff. Unless its a HAT, I have minimal experience in this field.

The Odroid C2 version of WiringPi we use is a branch by hardkernel, which lacks any additional information on Odroid implimentation: https://github.com/hardkernel/wiringPi

I'd highly recommend posting on the Odroid forums, someone with more experience and knowledge in that area, should be able to answer your questions:
http://forum.odroid.com/viewforum.php?f=139

Fourdee, 2016-12-20:
Marking this as resolved on v140 release.

closed 2016-12-20 by Fourdee