MichaIng/DietPi #654: RPI1 B+ Qbittorent software install fails

imakiro, 2016-12-18:
hello!

It seems that a dependency issue as happened

Mode: Install software
 Please wait...

 qBitTorrent: bittorrent server with web interface (c++)
 [Info] Please wait...

 #               processing | D i e t  P i             #

Reading package lists... Done
Building dependency tree
Reading state information... Done
You might want to run 'apt-get -f install' to correct these:
The following packages have unmet dependencies:
 qbittorrent-nox : Depends: libboost-system1.55.0 but it is not going to be installed
                   Depends: libqt4-network (>= 4:4.5.3) but it is not going to be installed
                   Depends: libqtcore4 (>= 4:4.8.0) but it is not going to be installed
                   Depends: libtorrent-rasterbar7 (>= 0.16.17) but it is not going to be installed
 rpimonitor : Depends: librrds-perl but it is not going to be installed
              Depends: libhttp-daemon-perl (>= 6.0.0) but it is not going to be installed or
                       libwww-perl (< 6.0.0) but it is not going to be installed
              Depends: libjson-perl but it is not going to be installed
              Depends: libipc-sharelite-perl but it is not going to be installed
              Depends: libfile-which-perl but it is not going to be installed
              Depends: aptitude but it is not going to be installed
E: Unmet dependencies. Try 'apt-get -f install' with no packages (or specify a solution).

                   ┌─────────────────────────────┤ Error ├──────────────────────────────┐
                   │                                                                    │
                   │ Apt-get has failed:                                                │
                   │  - Apt string = qbittorrent-nox                                    │
                   │  - Error code = 100                                                │
                   │  - Software Title = qBitTorrent: bittorrent server with web        │
                   │ interface (c++)                                                    │
                   │                                                                    │
                   │ Please report this issue to DietPi so it can be investigated.      │
                   │                                                                    │
                   │ DietPi-Software will now exit.                                     │
                   │                                                                    │
                   │                                                                    │
                   │                               <Ok>                                 │
                   │                                                                    │
                   └────────────────────────────────────────────────────────────────────┘

Fourdee, 2016-12-18:
@imakiro 
Thanks for the report 👍 , i'll take a look

Fourdee, 2016-12-18:
@imakiro 
Looks limited to your system, test install on RPi zero (ARMv6) fine.

Have you tried running?:
```
apt-get install -f
```
If not, please run it now and paste results.

Should that also fail, can you please send us a bug report, i'll check installed packages and apt sources:
http://dietpi.com/phpbb/viewtopic.php?f=8&t=499#p2184

Fourdee, 2016-12-21:
@imakiro 
I was unable to replicate this issue on fresh install of RPi Zero, so will mark as closed. Thanks for letting us know 👍

closed 2016-12-21 by Fourdee