MichaIng/DietPi #659: Allow setting MAC in dietpi.txt and dietpi-config

maweki, 2016-12-23:
I have some cheap (micro-)usb-to-ethernet interfaces from china that I was going to use with my raspberryPi Zeros that do not have their own ethernet interface. But these devices are very cheap and the drivers aren't the best so their MAC always gets populated with "00:e0:4c:53:44:58" (I don't actually care for the exact reason).

With one of those devices in my network, everything was fine. With a second (and third) it now isn't anymore. I know that I can edit `/etc/network/interfaces` but I do believe that gets overwritten by dietpi (which is fine).

So it would be really great if I could use the initial config (with `dietpi.txt`) to set a new MAC that is unique in my network.

MichaIng, 2018-01-12:
I found some guide on how to do it:
http://www.aboutlinux.info/2005/09/how-to-change-mac-address-of-your.html
```
ifconfig eth0 down
ifconfig eth0 hw ether 00:80:48:BA:d1:30
ifconfig eth0 up
```
With interface name `eth0` and MAC address `00:80:48:BA:d1:30`

But as this will be reset on reboot, we need to store & recover the info on `post-down` & `pre-up`.
Either this can be done within `/etc/network/interfaces` or by using the related subfolders to place a script: `/etc/network/if-down.d/` etc.
https://wiki.ubuntuusers.de/interfaces/

I would prefer the latter, as it is easier to just place files and remove them, when setting is disabled. I am not sure if this works just the same for WiFi (wlan0) and with `wpasupplicant`?

Fourdee, 2018-02-22:
Marking as closed due to no progress or interest from our community.

closed 2018-02-22 by Fourdee

maweki, 2018-02-24:
Can we reopen this? I still want to implement this. This would be very good in dietpi.txt because otherwise you have to boot dietpi (but you have no working network), remove the SD card, change the interfaces file, plug the card in again and boot.

And you play the same game every time you change the network settings via dietpi-config.

MichaIng, 2018-02-24:
Try:
```
cat << _EOF_ > /etc/network/if-pre-up.d/dietpi-change-mac
ifconfig eth0 hw ether 00:80:48:BA:d1:30
```
respectively use `wlan0` inside.
If this easy method works fine, for eth0 and wlan0, then I will implement it, first as "experimental" feature to DietPi.

CollinChaffin, 2022-05-13:
It's a shame this was closed, this random changing MAC address I'm amazed more aren't complaining this really is a pain on my DHCP server. I see mention of editing interfaces file etc, but that's not on the boot config.txt. Is there ANY way to set the ethernet MAC for the 1st boot/setup?  I only have had this issue with dietpi, haven't noticed it with other OS'.  Thanks!

MichaIng, 2022-05-14:
Which SBC are you using? There is only a small number with still dynamic MAC address, and (since you talk about the RPi-only `config.txt`) the RPi has a static MAC address for it's onboard Ethernet and WiFi in case.

mlanner, 2023-10-28:
I have the same issue on an Orange Pi 5 board. Every time I reboot the MAC address changes. Never seen this before. The suggested solution above, to add the `/etc/network/if-pre-up.d/dietpi-change-mac` doesn't seem to work for me.

MichaIng, 2023-10-28:
Try to use the modern `iproute2` toolset:
```sh
cat << '_EOF_' > /etc/network/if-pre-up.d/dietpi-change-mac
#!/bin/dash
ip l set dev eth0 address 00:80:48:BA:d1:30
_EOF_
```