MichaIng/DietPi #660: Static IP Bug

yunussandikci, 2016-12-23:
Hello, Firstly thank you for the this best operation system for raspberry pi 3.
The problem is about static ip. When you write v136(latest downloadable) image and make changes in dietpi.txt (wifi=1 static ip = 192.168.1.100.. bla bla) the raspberry pi 3 doesnt take the static ip on "FIRST" boot, its just getting an ip from dhcp. And then when you find this ip from you router and connect it via ssh its directs you to dietpi-update and make you device version to v140 and then it restarting itself. And your raspberry takes the static ip adress that you writed at dietpi.txt. Sorry For My Bad English.
Summary)
1)Write İmage change dietpi.txt for static ip
2)RPi3 get random ip from router
3)RPi3 makes update and restart
4)RPi3 gets static ip you writed in dietpi.txt

Fourdee, 2016-12-26:
@yunussandikci 

Thanks for the report, i'll take a look

Fourdee, 2016-12-27:
Looks like ```dhclient``` is still active after we apply 1st run network changes, overriding static IP:
[image]

Fourdee, 2016-12-27:
Tested here, works fine.

@yunussandikci 

Any chance you would be willing to also test and confirm working?
- Write image
- Setup ```dietpi.txt``` as previously
- Download https://raw.githubusercontent.com/Fourdee/DietPi/testing/dietpi/boot and save it to ```/boot/dietpi/boot```
- Power on and test

yunussandikci, 2016-12-29:
I made it, and it works well now. Thanks :)

Fourdee, 2016-12-30:
@yunussandikci 
Excellent 👍 

This will only take effect on DietPi v140 images or higher, as the ```boot``` script does not update until after ```dietpi-update```. Until then, if you do any reinstalls, make sure to copy the updated ```boot``` file, before booting system.

Fourdee, 2016-12-31:
I'll mark this as closed. Created a ticket to update the RPi image, so this fix takes effect on new installs:  https://github.com/Fourdee/DietPi/issues/676

closed 2016-12-31 by Fourdee