MichaIng/DietPi #661: Wifi Hotspot

Dalto1, 2016-12-24:
I got a pine64 with a Wifi hotspot not working. Im able to connect to the hotspot, but i got no internet on it. Uninstall-install not working.

Gab

Fourdee, 2016-12-27:
@Dalto1 

Which WiFi device are you using on the Pine, onboard addon or USB?

Dalto1, 2016-12-27:
Onboard addon

Fourdee, 2017-01-03:
Unable to replicate on fresh install of Pine A64 and hotspot installation using addon board. Appears to be working fine.

@Dalto1 
Possibly limited to your system/installation, we'll need to do some debugging.
Apologies for the delay in response. Could you please send us a bug report as we need some system information:
http://dietpi.com/phpbb/viewtopic.php?f=8&t=499#p2184

Could you also please paste results of the following commands:
```
systemctl status hostapd.service -l
cat /etc/hostapd/hostapd.conf
iwconfig wlan0
iwconfig wlan1
lsmod | grep brcmfmac
```

Dalto1, 2017-01-03:
**Reference CODE: c4764f40-537b-4590-96f2-920d3258c521-1**

**systemctl status hostapd.service -l**
â hostapd.service - LSB: Advanced IEEE 802.11 management daemon
   Loaded: loaded (/etc/init.d/hostapd)
   Active: active (running) since Mon 2017-01-02 23:37:45 EST; 14h ago
  Process: 1689 ExecStart=/etc/init.d/hostapd start (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/hostapd.service
           ââ1692 /usr/sbin/hostapd -B -P /run/hostapd.pid /etc/hostapd/hostapd.conf

Jan 02 23:37:45 BobPine2-I-1 hostapd[1689]: Starting advanced IEEE 802.11 management: hostapd.
Jan 02 23:37:45 BobPine2-I-1 systemd[1]: Started LSB: Advanced IEEE 802.11 management daemon.

**cat /etc/hostapd/hostapd.conf**
interface=wlan1
driver=nl80211
ssid=test
hw_mode=g
channel=9
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
wpa_passphrase=test
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP

**iwconfig wlan0**
wlan0     No such device

**iwconfig wlan1**
wlan1     IEEE 802.11bg  ESSID:"test"  Nickname:"<WIFI@REALTEK>"
          Mode:Master  Frequency:2.452 GHz  Access Point: 36:C3:D2:C2:70:19
          Bit Rate:54 Mb/s   Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=0/100  Signal level=-95 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

lsmod | grep brcmfmac
Nothing show up

**cat /etc/dhcp/dhcpd.conf**
ddns-update-style none;
default-lease-time 600;
max-lease-time 7200;
authoritative;
log-facility local7;

subnet 192.168.42.0 netmask 255.255.255.0 {
        range 192.168.42.10 192.168.42.50;
        option broadcast-address 192.168.42.255;
        option routers 192.168.42.1;
        option domain-name "local";

aliosa27, 2017-01-04:
You will also need to add an iptables masquerade rule and enable forwarding.

echo 1 > /proc/sys/net/ipv4/ip_forward
iptables -t nat -A POSTROUTING -o eth0 -j MASQUERADE

Fourdee, 2017-01-04:
@aliosa27 
Yep, this should be applied during install: 
https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software#L7927-L7937

@Dalto1 
Could you try a reinstall of the HotSpot, might resolve it:
```
dietpi-software uninstall 60
reboot
dietpi-software install 60
reboot
```

Dalto1, 2017-01-04:
Its not working. I strangly saw that ipv6 was activated so i uninstalled it from dietpi-config. But i cant get it to work, even with install/reinstall/uninstall and aplying the commands that @aliosa27 sent.

Dalto1, 2017-01-04:
btw, my settings are the same after uninstall/install, so i assume that my settings are saved somewere, is there any command to purge them?

Fourdee, 2017-01-05:
@Dalto1 

> btw, my settings are the same after uninstall/install, so i assume that my settings are saved somewere, is there any command to purge them?

This is what gets uninstalled, covers everything excluding ```wifi_hotspot_ssid=```, which is pulled from dietpi.txt:
https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software#L9422-L9463

> cat /etc/dhcp/dhcpd.conf

Can we confirm that the DNS server entry below exists ```cat /etc/dhcp/dhcpd.conf``` (was missing from your paste)
```
option domain-name-servers 8.8.8.8, 8.8.4.4;
```

Also, which devices have you tested with the HotSpot and no connection to internet (eg: android, win PC etc)?

If you have a spare SD card, please try a fresh installation of DietPi, install HotSpot and see if problems persist.

Dalto1, 2017-01-05:
`cat /etc/dhcp/dhcpd.conf
ddns-update-style none;
default-lease-time 600;
max-lease-time 7200;
authoritative;
log-facility local7;

subnet 192.168.42.0 netmask 255.255.255.0 {
        range 192.168.42.10 192.168.42.50;
        option broadcast-address 192.168.42.255;
        option routers 192.168.42.1;
        option domain-name "local";
        option domain-name-servers 8.8.8.8, 8.8.4.4;
}
`

Dalto1, 2017-01-05:
Cant connect with android, ill test on pc when i got time and try a new sd too.

Dalto1, 2017-01-05:
maybe my iptables are mess up?

Fourdee, 2017-01-06:
@Dalto1 

> maybe my iptables are mess up?

Possibly, although we flush IP tables during uninstall of HotSpot, so it should be same as my test install.

Either way, lets check your IPtables on the current installation, and i'll compare against a test install:
```
iptables-save > /root/iptables
cat /root/iptables
```

Dalto1, 2017-01-16:
Ill reinstall dietpi on it when i got more time, but thanks anyways !

closed 2017-01-25 by Dalto1